u-blox/ubxlib #9: Module configuration "extended errors"

eeFLis, 2021-03-25:
Hi
Shouldn't be AT+CMEE=1 (numeric) instead of AT+CMEE=2 (verbose) in the module configuration?
Otherwise uAtClientDeviceErrorGet() returns a struct where member "code" cannot be assigned correctly and always has the value 0. On the other hand, verbose mode is of course more helpful for debugging.

RobMeades, 2021-03-25:
Hi there.  Yes, I've wondered about this.  It's a choice really.  I've tended to go for the verbose rather than the numeric but that's only useful if you have the AT prints switched on.  My guess, so far, has been that everyone will have the AT prints switched on while they're developing, hence the verbose is useful, and once they've finished developing it had better darned well work so who cares what the error value is, only zero or non-zero really matters, little choice to be made based on the grey in-between.

What do you think?

RobMeades, 2021-03-27:
Actually, we might as well just put in a configuration flag so that if you would like to obtain numeric errors you can do so.  Note that no `ubxlib` cellular code will depend on numeric errors, hence this flag can be defined if you wish, or not, as the case may be.

closed 2021-03-27 by RobMeades

eeFLis, 2021-03-27:
hello sorry for the late reply. That's great thank you. 
I have created this issue because the functions specifies to return an error code which not work if CMEE=2. We stumbled across this when we were using the lib. I saw that the ubxlib is not affected, but wanted to know what you think about it. THX