u-blox/ubxlib #12: Feature voting and roadmap

hkro, 2021-05-11:
### Bring in and up-vote your favorite features!

- Follow this [LINK](https://github.com/u-blox/ubxlib/issues?q=is%3Aopen+label%3Afeature+sort%3Areactions-%2B1-desc) where all the features are sorted by popularity
- Open the feature you'd like to be implemented first
- Add a 👍 reaction to the feature description

---

Here in this issue:
- Feel free to ask questions and discuss any features and their priority in this issue
- Contributors will post updates and outlook for upcoming ubxlib releases

hkro, 2021-05-18:
The GNSS API is the most wanted feature based on 👍-reactions up to now (May 2021). Therefore we'll ship that in the next release. ✅ Fingers crossed, let's get ready for some signals... 🛰️🌎📍

syedqaisar, 2024-05-23:
hello.
I am using SARA-R422 with STM32L4 for the cell with tcp communication in which I will use different API's like network, socket, cell.
The issue is that the Port API is not suitable for this MCU so can you help me to draw the roadmap how to use it. because theses all API's are interconnected to each and i want to use only cellular functionality.

HKPhysicist, 2024-10-04:
May I suggest you to provide ublox cellular drivers for PIC 8 Bit MCU or PIC 16 Bit MCU?

Yours faithfully.

manemannen, 2024-11-11:
Project is discontinued

closed 2024-11-11 by manemannen