u-blox/ubxlib #20: Unable Establish ST-LINK between STM32CubeIDE and u-blox C030-R510M8S

gwchin14, 2021-05-12:
Hi, Rob,

I have completed the ubxlib build (issue: https://github.com/u-blox/ubxlib/issues/3), but unable to run it on C030-R510M8S application development board. I have installed the USB ST-Link driver (https://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-utilities/stsw-link009.html) but did not help. I believe you have the same board, how did configure/install the ST-LINK? Please assists, thanks. 

I have provided multiple screenshots for your reference:
[image]
[image]
[image]
[image]
[image]

Regards,
Chin

RobMeades, 2021-05-12:
First thing to check is the jumper on the board: the jumper must be on the side nearest the module.

RobMeades, 2021-05-12:
And which of the four USB ports are you connecting to?  You should connect with the USB port which is on the top side of the board, mounted horizontally (on the same side as the Ethernet connector).

gwchin14, 2021-05-12:
[image]

Hi, Rob,

Attached image shows how I connect the board, which I believe it is correct, based on your description.

gwchin14, 2021-05-12:
[image]
Now I am getting the error as shown in the above image.

RobMeades, 2021-05-12:
The jumper position looks correct and I think that's the right USB port, though it's a bit difficult to tell from the picture.  You should not have to install anything other than the STM32Cube IDE which installs everything it needs.  @philwareublox went through this procedure earlier in the week with the same board.  I'm not sure I've ever seen an "ST-Link DLL error" unfortunately.

Let's go back to basics; when you plug the USB cable into the board and your PC does a mapped drive appear on your PC and a COM port also appear?

gwchin14, 2021-05-12:
Let's go back to basics; when you plug the USB cable into the board and your PC, does a mapped drive appear on your PC and a COM port also appear?
Answer: There isn't any COM port which appears, this is why I think there is something wrong. If I use C030-R412M board, a "STMicroelectronics STLink Virtual COM Port" will appear, but not for the case when I plug in a C030-R510M8S to the USB port of my laptop. Instead of COM Port, the STM32 appears as a Universal Serial Bus devices with the name of "STM32 STLink":
[image]

RobMeades, 2021-05-12:
Does a mapped drive appear on your PC?

gwchin14, 2021-05-12:
No

RobMeades, 2021-05-12:
The board is recognised as ST-Link, so it is connected, but since it doesn't do any of the things that it is meant to do it might be that the ST-Link tool or whatever has installed an entirely different image on the debug chip of the C030 board itself.  You need to somehow get it back to normal.  Next thing you're going to ask me is "how...?" but I've very little idea as I've never done this myself and my board is not in the same state as yours so I can't try anything out for you.  Let me think...

RobMeades, 2021-05-12:
How about this:

- uninstall the thing you installed above with the link ending `stsw-link009.html`,
- install instead this tool:  [https://www.st.com/en/development-tools/stsw-link004.html](https://www.st.com/en/development-tools/stsw-link004.html),
- run it with the board plugged in and see if it offers to update the debug image on the board.

gwchin14, 2021-05-13:
No luck after installing stsw-link004, ST-LINK DLL Error still occurs. How about make a factory reset on STM32 using hardware button? Is this possible?
[image]

RobMeades, 2021-05-13:
It is the debug chip (STM32F103CBT6) that you need to reset somehow.  On the C027 boards there used to be pads to do this but I can't find equivalent pads on the C030 board.  I will ask internally how to do it but won't get an answer until next week now as it is public holidays in most of Europe.

RobMeades, 2021-08-23:
I believe this has been resolved now (it was a matter of finding the right stuff to load onto the debug MCU in the right way), closing.

closed 2021-08-23 by RobMeades