u-blox/ubxlib #23: PSM and eDRX

eeFLis, 2021-06-28:
Integrate power saving features into the API.

RobMeades, 2021-09-10:
@eeFLis: how important is the feature for you, timescale-wise?  I ask because it is now next on the list but there are other, shorter-duration-yet-lower-priority things [e.g. a recent request to see if it is possible to support Arduino-ESP32] that we could look at first.

eeFLis, 2021-09-10:
@RobMeades we have made a quick and dirty implementation so that we can work with PSM and eDRX. But we would like to see it fully integrated into ubxlib so that all power saving features like IDLE mode also can be used. This will be especially important when the SARA-R510S-01B with NB-IoT is released. But back to your question. If it can be implemented in Q4 that would be great.

RobMeades, 2021-09-10:
Thanks for the swift response and background information, appreciate it.  Agree that this is a high priority feature and, as I'm sure you've found out, it is not always easy to accommodate (especially when, in the `ubxlib` case, it needs to work across our different modules).  I've not got a SARA-R510S-01B myself yet [even though the code for it is being written 30 metres away :-)] but, yes, being able to power save properly is kinda essential to the NBIoT use-case.

We will aim to deliver this feature for you in Q4.

eeFLis, 2021-11-08:
@RobMeades Thanks for the update.
We do not use a GNSS chip so a preview branch would be helpful.
I hope you still keep the AT+CPSMS implementation on the ToDo list, since as you said it saves much more power.

RobMeades, 2021-11-08:
Thanks for responding.  I determined today that what I thought was something odd on the cellular module side seems actually to be a really subtle bug somewhere in my code so I need to spend more time hunting it down before I inflict it upon you even in a preview branch.

`AT+CPSMS` is most definitely not off the menu, it is what I was originally aiming for, just that unfortunately `AT+UPSV` has to be addressed first or `AT+CPSMS` won't actually do anything.

I'll post here when I have at least got `AT+UPSV` working and we can decide what to do next.

adpjhu, 2021-12-18:
@RobMeades I would like to echo the importance and value of this feature. It could be argued that most users of this chip selected it because it is low power, and software support for those low power modes could be seen as a basic requirement for a driver. This is certainly true for my use case and I am not using the GNSS chipset or Arduino.

RobMeades, 2021-12-19:
Understood.  In order to help me find the rather subtle issue in the `AT+UPSV` implementation, which must be completed before `AT+CPSMS`, I diverted to create a version of `ubxlib` that will run on a PC, attached to a SARA-R5 module via USB.  I finished that on Friday so I'm hoping the problem will now reveal itself.  I will update here when I know more.

RobMeades, 2021-12-22:
I have now pushed the `AT+UPSV` part of power-saving, at least the initial implementation of it, here (see commit a9711017ff867d4d9d78f54b4291917542cf8bad).  I say the first part because, as it stands, for SARA-R5 you cannot use the CTS/RTS flow control lines while `AT+UPSV` is enabled (see the commit message for the details of why).  It is possible to fix that but I will do it as a separate commit.

Next, `AT+CPSMS`...

RobMeades, 2022-01-23:
A quick update on this:  there's a bit of a chain of things going on:

- there is a subtle issue in the rather old SARA-R5 modules (00B version) we use on our test farm which means that we don't always get the `OK` response to an `AT+GPS=0` command (i.e. switching GNSS off) if power saving is enabled; internal logging of the module shows it being sent but we don't always see it on the physical UART, must be something to do with the power saving sequence,
- the above needs to work when we run tests of the existing Cell Locate feature (since the tests have to switch between having the user using the GNSS chip and giving it up to be used by Cell Locate by issuing `AT+GPS=0`),
- hence, in order to run our tests with power saving on, we need to switch all the SARA-R5 modules on our test farm to the 01B version,
- we now have the bits needed to do that switch,
- with that done we can commit the change which enables `AT+UPSV` to be used when the UART flow control lines are connected between the MCU and the module; this change was completed a few weeks back, is tested locally and ready to go but couldn't be committed because then the existing Cell Locate tests would start failing.

Hopefully we will to get through all of the above next week.  _Then_ `AT+CPSMS`.

eeFLis, 2022-01-26:
Hi Rob.
Thanks for the update.
Good to hear that things are moving forward. 
especially since the 01B version supports NB-IoT and power saving will be even more important.

RobMeades, 2022-02-09:
The changes that allows `AT+UPSV` power saving to work correctly when the UART flow control lines are connected with a SARA-R5 module are now pushed here, see commit b57b7ec3cf3f32e3e4410661f16c50e40fae199a.

RobMeades, 2022-03-17:
Commit 9261d9c61a7f1873b09faf56cfbcc9aa11f864cf represents the implementation of 3GPP power saving and E-DRX support for SARA-R5/SARA-R5 modules.  Note that to support 3GPP power saving operation you will need to make sure that `PWR_ON` and `VINT` are both connected from the module to your MCU. Also note that three new `port` APIs were required as part of the implementation:  `uPortEnterCritical()`/`uPortExitCritical()` and `uPortTaskGetHandle()`; if you have your own `port` you will need to implement these APIs.

The best place to start in terms of finding out about the power saving APIs is probably here:

https://github.com/u-blox/ubxlib/tree/master/example/cell/power_saving

...while the APIs themselves can be found in [u_cell_pwr.h](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_pwr.h).

And, lastly, power saving is a complicated topic!  It has taken all of this year to make this work satisfactorily and, of course, there may still be holes.  Please open an issues here if/when you find the holes :-).

closed 2022-03-17 by RobMeades