u-blox/ubxlib #25: get no connection to the network

alexmaron81, 2021-08-03:
I have the SARA-R412M 02B module, more precisely "LTE IoT 6 click" in connection with the ESP32 MCU. In the beginning everything worked great. Later I used a new SIM card, which was recognized in the system but did not get a connection to the network.
Then I exchanged the SIM card again, initially the card worked again, but after several attempts to dial in, I can no longer get a connection with this card either.
[failed.txt](https://github.com/u-blox/ubxlib/files/6926678/failed.txt)
[successful.txt](https://github.com/u-blox/ubxlib/files/6926679/successful.txt)

Does anyone have an idea why this might be?

RobMeades, 2021-08-03:
Hi there.  This seems very odd.  The AT logs show nothing strange: the success case is clear enough: you give the APN `iot.1nce.net` which the module is told to use, and it connects to a network which identifies itself as "262 01 1nce.net" in 4 seconds.  Interestingly you are registered as roaming on that network and I think this is a result of the type of SIM you are using, "901 40" on the start of the IMSI of your SIM meaning you are on a global roaming SIM that associates itself with no particular country (see [here](https://news.itu.int/one-world-one-global-sim/)).

So that's fine.  Then in the second log, as you say, everything is identical and yet there is no registration. Can I ask how long you waited?  It might be worth letting it time out (I think the timeout is 240 seconds) just to make sure it really never gets coverage.

Since this is not really a `ubxlib` issue as far as I can tell, I will consult with our network/module experts internally to see if they have any suggestions as to what we might try.

Do you know if the network/SIM you are using is expected to be an LTE or a GPRS network? If an LTE network we can maybe try the [lte_cfg](https://github.com/u-blox/ubxlib/tree/master/example/cell/lte_cfg) build (or incorporating bits of that into your build) to see if the somewhat complex LTE network configuration is causing problems somehow.

alexmaron81, 2021-08-03:
Thank you for your prompt reply!
I set the timeout to 400 and it was also unsuccessful. Before that, the system had dialed into the network very quickly. I have another pycom fipy development system. There I put the band 80, as prescribed by 1nce, and it works right away, even with all sim cards I have from 1nce.
1nce supports 2G, 3G, 4G, LTE-M and NB_IoT in Germany. If I understand correctly, the system connects via NB-IoT.
I will try to run the lte_cfg build.

philwareublox, 2021-08-04:
> *** This is an EXTERNAL email. It was sent from outside of u-blox. ***

The failure shows the CEREG response as "0" - which means the module is not trying to scan for networks.
I see the AT+CFUN=1, so I would expect it should be scanning for networks.... Just that it is not.

Generally the module should be always scanning for networks after +CFUN=1, but there are some "radio policy" things which might kick in that have stopped this happening. From the AT log I can't say if this is happening or not.

AT+CFUN=1
...
...
AT+CEREG?
+CEREG: 0,2
OK

We should see this if it is scanning for networks.

Phil.

RobMeades, 2021-08-04:
@philwareublox: thanks Phil, and indeed, as the success log shows, the prints include an indication that 0 means "search":

```
0: Search
0: Search
0: Search
```

@alexmaron81: while it is a somewhat dangerous thing to fiddle with, if you are **sure** that setting band 80 on an NB network is a requirement, you could try adding something like this to the start of your code, after you've powered the cellular module on:

```
uint64_t nbBandMask1;
uint64_t nbBandMask2;
#define WANTED_BAND_UPPER_SET 80

if (uCellCfgGetBandMask(networkHandle, U_CELL_NET_RAT_NB1,
                        &nbBandMask1, &nbBandMask2) == 0) {
    printf("Band mask for NB1 is 0x%08x%08x %08x%08x.\n",
            (uint32_t) (nbBandMask2>> 32), (uint32_t) nbBandMask2,
            (uint32_t) (nbBandMask1>> 32), (uint32_t) nbBandMask1);
    if (nbBandMask2 & (1 << (WANTED_BAND_UPPER_SET - 64)) == 0) {
        printf("Band 80 is NOT included, trying to include it...\n");
        nbBandMask2 |= 1ULL << (WANTED_BAND_UPPER_SET - 64);  
        if (uCellCfgSetBandMask(networkHandle, U_CELL_NET_RAT_NB1,
                                nbBandMask1, nbBandMask2) != 0) {
             printf("Unable set band mask for NB1 to 0x%08x%08x %08x%08x.\n",
             (uint32_t) (nbBandMask2>> 32), (uint32_t) nbBandMask2,
             (uint32_t) (nbBandMask1>> 32), (uint32_t) nbBandMask1);
        } else {
            printf("Band 80 is now included.\n");
        }
    } else {
        printf("Band 80 is included.\n");
    }
} else {
    printf("Unable to read band mask for NB1.\n");
}

if (uCellPwrRebootIsRequired(networkHandle)) {
    printf("Re-booting module to apply band mask change...\n");
    uCellPwrReboot(networkHandle, NULL);
}
```

I haven't compiled this so it may contain syntax errors but you get what I mean.

alexmaron81, 2021-08-04:
Thanks for your help!
I'm about to try out the codes. but what i don't understand is that everything worked at the beginning and the device was able to dial into the network? After I inserted another SIM card from the same supplier "1nce" into the same device, it no longer wants to work.

RobMeades, 2021-08-04:
Yes, I can't explain that.

Of course, we cannot determine the situation between the supplier of the virtual network operator that supplied your SIM card and the physical network the module sees most strongly in your area.  I have seen situations in the past when, for instance, I use the wrong APN, registration will work at first and on later attempts will fail, like the network simply accepts that the SIM card is valid at first and gets on with things but then later, once it has checked, determines that it does not have an agreement with that virtual operator and so sends a reject, resulting in that network being added to the "do not try list" on the SIM to prevent any future attempts to bother it.

I had a look at the "1nce" website but couldn't determine which physical network operators in Germany it has roaming agreements with: do you know?  If we could determine that we could maybe try forcing the module to register only on that network by providing an MCC/MNC to `uCellNetRegister()` (and maybe doing a scan beforehand with the `uCellNetScanGetFirst()`/`uCellNetScanGetNext()`/`uCellNetScanGetLast()` pattern to ensure that is is there).

RobMeades, 2021-08-04:
OK, I can't find "1nce" in that spreadsheet but it seems to be 2G only and I guess you wouldn't necessarily expect Deutsche Telecom to list the incoming-roaming partners they support.  Your module _appears_ to have registered on Deutsche Telecom in the success case so we could force it to do so again.  As an experiment, you could try doing a network scan (after your `uNetworkAdd()`), just so that you know what the radio environment looks like:

```
char buffer[32];
char mccMnc[U_CELL_NET_MCC_MNC_LENGTH_BYTES];

// Do a network scan, just as an experiment
printf("Doing a network scan...\n");
for (int32_t x = uCellNetScanGetFirst(networkHandle, buffer, sizeof(buffer), mccMnc, NULL);
     x >= 0;
     x = uCellNetScanGetNext(networkHandle, buffer, sizeof(buffer), mccMnc, NULL)) {
     printf("%d: network: \"%s\", MCC/MNC %s.\n", buffer, mccMnc);
}
```

Then, assuming you are currently calling `uNetworkUp()` to connect to the network, to do manual network selection it is probably simplest to modify [u_network_private_cell.c](https://github.com/u-blox/ubxlib/blob/master/common/network/src/u_network_private_cell.c#L262) to fill in the `mccMnc` parameter which is currently `NULL` (because it normally only expects to perform automatic network selection), something like this:

```
// Force registration onto the MCC/MNC of Deutsche Telecom (260/01)
char mccMnc[U_CELL_NET_MCC_MNC_LENGTH_BYTES];
snprintf(mccMnc, sizeof(mccMnc), "%03d%02d", 260, 1);
errorCode = uCellNetConnect(handle, mccMnc,
                            pConfiguration->pApn,
                            NULL, NULL,
                            keepGoingCallback);
```

You will need to `#include "stdio"` at the top of `u_network_private_cell.c` for the above to compile.  Again, I've not compiled the above code myself, it may contain syntax errors, this is just to give you an idea.

alexmaron81, 2021-08-05:
I once played around with the manual settings for mcc mnc and afterwards I tried dialing in again automatically. Now the mcc mnc 262 01 is also found by Deutsche Telekom. see file.
Could it be that the module gets stuck somehow, so that I have to reset it if I don't shut it down properly? Or is the module reset after a restart?
[partly successful.txt](https://github.com/u-blox/ubxlib/files/6942143/partly.successful.txt)

philwareublox, 2021-08-06:
Hi,

Only a CS Domain registration is happening - no LTE.
+CREG: 5

+CGREG and +CEREG are reporting "0" or "4"

Because there is only a circuit switched connection (CS) +CGACT is not supported.

Regards,
Phil.

weinholtendian, 2021-08-06:
In my experience with the R412M, if NB-IoT is enabled in AT+URAT then that adds an extra 10-12 minutes before the module has network service. I believe this is caused by a deep scan performed when the module can't find the SIM card's home network due to roaming (but don't quote me on that).

I have used AT+URAT=7,9 to disable NB-IoT in one product I'm working on. Then the R412M normally gets service within seconds, although the first time takes about 30 seconds.

It may be that you initially left the module powered on without interruptions long enough for it to finish a deep scan and then you continued using it for a while. Then something changed in the network or the module and the module opted to do another deep scan (e.g. AT+COPS=0,0 will trigger this).

philwareublox, 2021-08-06:
This is a good point.

When you have all three technologies enabled with +URAT: 7,8,9 – and there is no NB-IoT network to connect to, this can take a long time.
The reason why NB-IoT can take many minutes is because of the technology used to provide the deep coverage. This simply requires more time to scan.

Selecting the correct technologies for +URAT should speed up the connection when developing with known networks.

RobMeades, 2021-08-06:
To select just the GPRS and CAT-M1 RATs, removing the NB1 RAT, like in the [lte_cfg](https://github.com/u-blox/ubxlib/tree/master/example/cell/lte_cfg) example, you would put something like this at the start of your code:

```
uCellNetRat_t rat[3];

// Remove NB1 from the RAT list if it is present
for (size_t x = 0; x < 3; x++) {
    // Get the RAT at rank x
    rat[x] = uCellCfgGetRat(networkHandle, x);
    if (rat[x] >= 0) {
        printf("RAT[%d] is %d.\n", x, rat[x]);
        if (rat[x] == U_CELL_NET_RAT_NB1) {
            print("Found NB1 in the list, removing the RAT!\n");
            if (uCellCfgSetRatRank(networkHandle,U_CELL_NET_RAT_UNKNOWN_OR_NOT_USED, x) != 0) {
                printf("### Unable to remove NB1!\n");
            }
        }
    }
}
// Reboot the module if required
if (uCellPwrRebootIsRequired(networkHandle)) {
    printf("Re-booting the module to apply RAT change...\n");
    uCellPwrReboot(networkHandle, NULL);
}
```

alexmaron81, 2021-08-07:
Hi, everyone. 
I finally managed to establish the connection with the other sim card.
RobMeades thanks for the tip to put the mccmnc directly in the u_network_private_cell.c.
It should also be noted that the mcc of Germany is 262.

how would you advise me to develop an iot device that should work globally? Of course, I will configure the device beforehand that comes to a certain country. But it can always happen that one or the other device changes country later.

RobMeades, 2021-08-07:
Good stuff!  @philwareublox is probably the best one to advise on your final question.

philwareublox, 2021-08-09:
For global roaming products I can only suggest that the product has some feedback or instructions to the user to note that when Cat-M1, NB-1 and 2G are all enabled and the FIRST connection may take a long - especially when many Bands are also enabled.

Unfortunately there are no URCs which return the current scanning Band, or technology it is trying at the time.

Regards,
Phil.

RobMeades, 2021-08-23:
I'm going to close this new: Alex, if you feel that there is more to do feel free to re-open it.

closed 2021-08-23 by RobMeades

philwareublox, 2021-08-31:
Unfortunately the only way to configure the device for "global roaming" is to enable all the Radio Access Technologies and Bands. This means when a module is first turned on it might take many 10's of minutes to find a network to attach to.
The application should consider the order of RATs to search, I.e. Cat-M1 first, 2G second and then NB-IoT - as NB-IoT takes to the longest to scan through.

Using the +UMNOPROF command for our R4xx and R5xx modules you can set a MNO Profile to "Global". This enables all the bands. I believe also that the 'default' setting of +UMNOPROF is "90: Global" too on the R5xx modules.

alexmaron81, 2023-07-19:
Hello everyone,
I haven't worked on the project for a long time.
Now I wanted to establish a connection again and get "unable to register with the network"

AT
AT
AT
AT
AT
[00]U_CELL_PWR: powering on.
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

M0.10.00,A.02.14

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K3

OK
AT+UPSV=4

OK
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?

+CEDRXS:

OK
AT+UMNOPROF?

+UMNOPROF: 0

OK
AT+CFUN=4

OK
Opened device with return code 0.
Bringing up the network...
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

901405102306300

OK
U_CELL_NET: user-specified APN is "iot.1nce.net".
AT+CGDCONT=1,"IP","iot.1nce.net"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
AT+CFUN=1

+CME ERROR: operation not supported
U_CELL_NET: unable to register with the network, is APN "iot.1nce.net" correct and is an antenna connected?
AT+CFUN=4

OK
U_CELL_NET: connection attempt stopped after 1 second(s).
Unable to bring up the network!
Done.