u-blox/ubxlib #27: Direct link mode support

rquilez, 2021-08-23:
I have had in mind a cellular API for a long time and it has been a real pleasure to find this library. Thank you very much for your amazing work.

Are you planning to include the "direct link" mode in the library?

In order to obtain the maximum throughput of the modem,  I was thinking myself in migrating from AT commands to write/read  to "direct link" mode. It's a pity  because I would also like to migrate to your lib implementation. Is there a reason it is not implemented?

RobMeades, 2021-08-23:
Thanks!  I hope you find it useful.

Ref. direct link mode, I've not looked into it but I guess there's no reason for it not to be supported other than it just hasn't made it to the list yet.  @hkro manages what order we do things in but I'll have a quick look to see how much effort it would be.

RobMeades, 2021-08-23:
Had a very quick look and I guess direct link mode would be supported something like this:

- Add an API `uCellSockDirectLinkStart(int32_t cellHandle, int32_t sockHandle)`: this would put the whole cellular interface into direct link mode with the given socket.  The AT parser would stop receiving characters on the UART port, leaving it for the application to handle.  Any attempt to use a cellular API while direct mode is active should probably return a specific error code, e.g. `U_CELL_ERROR_DIRECT_LINK_ACTIVE`.
- Add an API `uCellSockDirectLinkStop(int32_t cellHandle, int32_t sockHandle)`: this would wait for 2 seconds, send the "+++" sequence to leave direct mode and then reconnect the AT parser once more, probably sending "AT" and waiting for "OK" to determine that it is done.
- Should probably also add an API to configure the various direct link parameters.

Does that sound like the right approach?  I can't give you a timescale for doing it right now, will need to discuss that internally, just wanted to check if this is what you were thinking of.

manemannen, 2024-11-11:
Project is discontinued

closed 2024-11-11 by manemannen