u-blox/ubxlib #33: config settings for ESP and ublox cellular communication

Paryavi, 2021-09-28:
Hi Rob,

So recently, I was able to upload runner to Adafruit ESP32 feather board successfully;
https://github.com/u-blox/ubxlib/tree/master/port/platform/esp-idf/mcu/esp32/runner

And I connected its GPIO16 and GPIO17 as RX and TX to Sarar4 breakout board(Sparkfun shield);
https://learn.sparkfun.com/tutorials/lte-cat-m1nb-iot-shield-hookup-guide/all

I get the following error when running ExampleSockets(first choice);
idf.py -p COM5 -D TEST_COMPONENTS="ubxlib_runner" flash monitor

Running ExampleSocket... by clicking "1"
Added network with handle -5
Bringing up the network...
Unable to bring up network!
Done.

I modified the server part in example >sockets> main.c to the following;

// Echo server URL and port number
#define MY_SERVER_NAME "www.google.com"
#define MY_SERVER_PORT 80

Also, how can I register operator and APN in this example?

Here is how I was able to set APN and operator in Sparkfun;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/examples/00_Register_Operator/00_Register_Operator.ino

Details:
Btw, I tested Sparkfun and hologram Simcard; Hardware works, and I was able to run their Arduino codes, like sending SMS or registering AT&T network...; These are easy Arduino examples you might like to include in your Arduino Platform...;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/tree/master/examples

Then I disconnected the Arduino board from Sparkfun lte shield and connected 4 pins RX,TX,VCC, and ground from ESP32 feather to the late shield and I run the runner in IDF, which did not work as I explained.

My u_cfg_app_platform_specific.h configuration in folder ...ubxlib\port\platform\esp-idf\mcu\esp32\cfg:

what I did is just connect rx and tx as gpio 16 and 17 from ESP feather board uart2 to sparkfun lte shield;
define U_CFG_APP_PIN_CELL_TXD 17
define U_CFG_APP_PIN_CELL_RXD 16

Also, it looks like below, I set "cell uart" to 1, but I guess I should set it to -1 like other pins, too? not sure what it does...
define U_CFG_APP_CELL_UART 1

Thanks,
MoZen

RobMeades, 2021-09-29:
Hi again, and thanks for raising this as a separate issue.

There are a number of pins you need to be concerned with in the [cellular section of u_cfg_app_platform_specific.h](https://github.com/u-blox/ubxlib/blob//master/port/platform/esp-idf/mcu/esp32/cfg/u_cfg_app_platform_specific.h#L105).  I will go through them one by one.

[U_CFG_APP_CELL_UART](https://github.com/u-blox/ubxlib/blob//master/port/platform/esp-idf/mcu/esp32/cfg/u_cfg_app_platform_specific.h#L98): this is not a pin, it identifies the UART HW block _inside_ your MCU which `ubxlib` should use for its UART interface.  Without this there can be no communication with the cellular module, so it should not be set to -1 if you are using a cellular module.  ESP-IDF by default uses UART HW block 0 for debug prints and so we by default use UART HW block 1 for our communications with the cellular module.  This value doesn't usually need to be changed, you should leave it at 1 unless you are using UART HW block 1 for something else.

[U_CFG_APP_PIN_CELL_ENABLE_POWER](https://github.com/u-blox/ubxlib/blob//master/port/platform/esp-idf/mcu/esp32/cfg/u_cfg_app_platform_specific.h#L109): if you have a relay or some such gating power to the cellular module then you need to set this to the MCU pin which controls that relay.  If you don't you should set this to -1.

[U_CFG_APP_PIN_CELL_PWR_ON](https://github.com/u-blox/ubxlib/blob//master/port/platform/esp-idf/mcu/esp32/cfg/u_cfg_app_platform_specific.h#L116): the cellular module has a pin named `PWR_ON` which may be connected to one of your MCU pins.  The `PWR_ON` tells the cellular module to power on and to power off again.  If it is connected to one of your MCU pins then you must set the value of `U_CFG_APP_PIN_CELL_PWR_ON` to match.  If it is not connected (i.e. you are relying on the cellular module just powering up when power is initially applied, which I believe most of them do) then set the value of `U_CFG_APP_PIN_CELL_PWR_ON` to -1.

[U_CFG_APP_PIN_CELL_VINT](https://github.com/u-blox/ubxlib/blob//master/port/platform/esp-idf/mcu/esp32/cfg/u_cfg_app_platform_specific.h#L123): the cellular module has an _output_ pin, named `VInt`, which supplies 1.8V for interface circuitry (noting that all u-blox modules are 1.8V IO, not 3.3V).  If this pin is connected to your MCU then `ubxlib` can use it to detect that the module is powered-up.  So please either set `U_CFG_APP_PIN_CELL_VINT` to the pin of your MCU that is connected across to the `VInt` pin of the cellular module or, if there is no such connection, set it to -1; `ubxlib` will work with or without this connection but this #define MUST be set to -1 if there is NO SUCH connection.

[U_CFG_APP_PIN_CELL_TXD and U_CFG_APP_PIN_CELL_RXD](https://github.com/u-blox/ubxlib/blob/master/port/platform/esp-idf/mcu/esp32/cfg/u_cfg_app_platform_specific.h#L131): these I think you have understood.

[U_CFG_APP_PIN_CELL_CTS and U_CFG_APP_PIN_CELL_RTS](https://github.com/u-blox/ubxlib/blob/master/port/platform/esp-idf/mcu/esp32/cfg/u_cfg_app_platform_specific.h#L145): from what you have said above you have not connected the UART flow control lines between the MCU and your cellular module, in which case please make sure to set both of these values to -1 or no UART communication will succeed.

I you need to set the APN, i.e. your SIM card/provider requires you to use a specific APN and not just accept the network default, then, for the [sockets](https://github.com/u-blox/ubxlib/blob/master/example/sockets/main.c) example, you need to look at the [config](https://github.com/u-blox/ubxlib/blob/master/example/sockets/main.c#L99) structure.  You will see that there is a comment against one of the fields "/* APN: NULL to accept default.  If using a Thingstream SIM enter "tsiot" here */"; this is the field where you would enter your APN for this example.

Paryavi, 2021-09-29:
Thanks, Rob,

Yes, I was suspicious U_CFG_APP_CELL_UART 1 has a more important role being at the top of the code, that is why I set it to 1 previously, but thanks for the clarification about it and other pins, I knew some of them reading the Sara-r4 datasheet and I had set most of the pins as you mentioned.

As you mentioned ESP32 has several uarts(3 I guess), Since I am using feather board(esp wroom-32) and one of its available uart pins; gpio 16 and 17 are u2rxd and u2txd, so is it uart hardware block 2? then should I set U_CFG_APP_CELL_UART to be 2? 

APN in my case is hologram, so I will modify NULL in NULL, /* APN: NULL to accept default
But for setting let's say AT&T, which one should I modify?

Thanks,
MoZen

RobMeades, 2021-09-29:
The ESP32 is capable of using any pins with any HW UART block so it's really up to you; you could leave it at 1 or try changing it to 2, it shouldn't matter.

On the APN it is the same field you modify, you just put in it whatever string your SIM/service provider tells you to.  I doubt that AT&T needs a specific APN though, the AT&T network should provide the necessary APN to the module at network registration and you should be able to leave the APN for that case at the default of `NULL`.

Paryavi, 2021-09-30:
Hi Rob,

Thank you for your precious feedbacks, 

I was able to upload the sockets.ino to ESP board; in the terminal after showing a few lines of AT, it says the cellular network does not connect, and the on Sparkfun Sara R4 breakout board that I power with a Lipo does not turn blue, I loaded code on Adafruit ESP feather also ESP dev board, same results...

Config setting; I directly manipulate u_cfg_app_platform_specific.h in Arduino libraries>src folder as follows;
// specify ublox module as Sarar4102B
define U_CFG_TEST_CELL_MODULE_TYPE U_CELL_MODULE_TYPE_SARA_R410M_02B

First Everything on u-cfg-short-range I set to -1
Then U_CFG_APP_CELL_UART                  1
U_CFG_APP_PIN_CELL_ENABLE_POWER      -1
U_CFG_APP_PIN_CELL_PWR_ON            -1
U_CFG_APP_PIN_CELL_VINT              -1
U_CFG_APP_PIN_CELL_TXD              17
U_CFG_APP_PIN_CELL_RXD              16
U_CFG_APP_PIN_CELL_CTS              -1
U_CFG_APP_PIN_CELL_RTS              -1
Lastly everything on U_CFG_APP_GNSS_UART is -1
I did not change any other .h files.

in Arduino code, I left "simpin" and APN to NULL,

I am using these pins on Sara-R4 breakout boards; In this video, Jim explains at minute 1:01 you can use it directly with uart;
https://www.youtube.com/watch?v=Eb6Ld0QAdj4&t=2s
similarly I am connecting to these uart pins.
 
Also you can find it in "UART interface" title here; https://learn.sparkfun.com/tutorials/lte-cat-m1nb-iot-shield-hookup-guide/all#hardware-overview

So the uart serial shown in the video has 6 pins, 2 NC, TX(I connected to Esp gpio17), RX(esp gpio17), Ioref(I connected to 3.3V on esp32 that will convert esp uart to 1.8 volts for sarar4 if you look at their schematic), and Gnd(I connected to feather esp grnd), 

my question is 
1. should I use a parallel pull-down resistor, a let's say 5K resistor to gnd on tx and rx pins between esp and sparkfun shield?
2. how can we increase the time it sends AT commands to the board, I guess I need to increase x here; int32_t x = 3; x >= 0; x--
3. we bought 10 sara-r410 chips, if sparkfun does not work with esp, do you have a breakout board eagle file for sarar4 so that we use its schematic for designing our prototype board?
4. Have you tested this sockets code with USA networks(AT&T,...)? because in the sparkfun hookup guide and from my experience it takes time until the network is connected(30 sec to 3 minutes), but in sockets.ino code it sends AT for I say 10  seconds and gives up.

Thanks,
MoZen

RobMeades, 2021-10-01:
You say "in the terminal after showing a few lines of AT": would you be able to post what you see here, just so that I can understand what is happening better?

What you have set in `u_cfg_app_platform_specific.h` seems sensible.

On the pins etc, can you confirm that you have pin 17 of your MCU connected to the `TXD` input line of the module and pin 16 of your MCU connected to the `RXD` output line of the module (i.e. the module uses the [DCE convention](chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/viewer.html?pdfurl=https%3A%2F%2Fftp1.digi.com%2Fsupport%2Fcabling%2Fdte_vs_dce.pdf&clen=48940&chunk=true), so the `TXD` line is the input and the `RXD` line is the output).

I'm not a HW engineer (despite having a degree in it, from many years ago) but I very much doubt that any pull-up or pull-down resistors are required on the `TXD` or `RXD` lines, that would adversely affect the dynamic operation of those pins.

You should not need to make any changes in the `ubxlib` code, anything that we think should be configurable (timeouts etc.) are brought out as conditional compilation flags you can tweak but lets not do anything like that yet, there is not yet any reason to believe that timing is a factor.  On your question (3) our official advice is in the [SARA-R4 system integration manual](https://www.u-blox.com/en/ubx-viewer/view/SARA-R4_SysIntegrManual_UBX-16029218?url=https%3A%2F%2Fwww.u-blox.com%2Fsites%2Fdefault%2Ffiles%2FSARA-R4_SysIntegrManual_UBX-16029218.pdf) but we did an [energy harvesting project](https://www.u-blox.com/en/publication/white-paper/energy-harvesting) a few years ago which used SARA-R4 and includes a schematic for the board we made which you may find useful.  There's also the C030-R410M we designed for use with `mbed-os` which was never publicly released but I've [attached the schematic](https://github.com/u-blox/ubxlib/files/7265835/c030-r412m-0-00_schematics.pdf) in case it is useful to you.  Finally, there was an internal project which made a small M.2 form-factor board containing SARA-R412M, NINA-W10 and a ZOE-M8 GNSS receiver, the [schematic](https://github.com/u-blox/ubxlib/files/7265931/whre-AIC-KeyE-whre-AIC-KeyE_v1_1.in_review.pdf) for which I've also attached.  IMPORTANT NOTE: these schematics are offered in good faith for your information, neither I nor anyone in u-blox can support you with them since they are age-old projects that were never intended for external consumption in the first place!  Just so's you know...

On the US network thing, the network registration timeout is set in that same `config` structure where you set the APN: you will see the line `240, /* Connection timeout in seconds */`, so you can set it to what you wish.  The network registration time-out is in now way connected with the AT time-out or the sockets time-out etc, it is entirely separate.

Paryavi, 2021-10-04:
Hi Rob,

How is it going?

Yes, RTS and DTR are connected to GND in Sparkfun shield, I did a continuity check to board's GND with Voltmeter.

The Arduino Terminal report;

U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
Added network with handle -7.
Bringing up the network...
Network is not available!
1 Tests 1 Failures 0 Ignored

My board is an Adafruit  ESP feature board, 
https://cdn-learn.adafruit.com/downloads/pdf/adafruit-feather-huzzah-esp8266.pdf

so I connect feather board gpio 16 which is RX2 to Sara R4 RXD pin, and its gpio 17 which is TX2 to Sara R4 TXD pin.
And in arduino src folder I modify the code to 
define U_CFG_APP_PIN_CELL_TXD              17
define U_CFG_APP_PIN_CELL_RXD              16

So I guess 16 means gpio 16 and 17 gpio17, as I changed them to their esp32 pin 27 and 28 (ESP wroom pin numbers), got an uart error in the terminal, so I brought it back to 16 and 17 as seen above.

I swap the rx tx cables, it will have the same terminal report; Network is not available!

To be able to make the Sparkfun LTE modem working when I was connecting it to Arduino uno Redboard, setting MNO_ATT was important as follows;
IN Sparkfun code line 57;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/examples/Network_Info/Network_Info.ino
in order to have "LTE shield connected", I had to set
const mobile_network_operator_t MOBILE_NETWORK_OPERATOR = MNO_ATT
and 
const String APN = "hologram";

To do the same thing In the ublox arduino code for feather board, in config part I set;
NULL, /* SIM pin, must be NULL */
"hologram", /* APN: NULL to accept default.  If using a Thingstream SIM enter "tsiot" here */

OR

I set it as 
NULL, /* SIM pin, must be NULL */
"MNO_ATT", /* APN: NULL to accept default.  If using a Thingstream SIM enter "tsiot" here */

I set both to NULL, it does not work, unfortunately.

I wanted to see which pins you connect to rx and tx in devkit c;
 https://github.com/u-blox/ubxlib/blob/preview_feature_arduino_updated/port/platform/common/automation/DATABASE.md
 
These are U_CFG_TEST pins (which I did not touch test pin configs) not U_CFG_APP pins, also in your ESP32; 
U_CFG_TEST_UART_B=1 U_CFG_TEST_PIN_UART_A_CTS=-1 U_CFG_TEST_PIN_UART_A_RTS=-1 U_CFG_TEST_PIN_UART_A_RXD=26 U_CFG_TEST_PIN_UART_B_TXD=27 U_CFG_TEST_PIN_UART_B_RXD=14

Also in u_cfg_app_platform_specific.h You connect tx and rx as follows;
define U_CFG_APP_PIN_CELL_TXD              4
define U_CFG_APP_PIN_CELL_RXD              15
I do not understand why you connected tx and rx to gpio4 and gpio 15 or pin 4 or 15 of ESP32 devkit C
Because these are not uart pins of ESP32 devkit;
https://randomnerdtutorials.com/getting-started-with-esp32/

Thanks,
MoZen

RobMeades, 2021-10-04:
The problem we have here, MCU UART apparently transmitting commands to the module but getting no response at all from the module, is likely due to one of the following reasons:

- no power supplied to cellular module [I assume you've checked this],
- cellular module is not powering-up [see below],
- transmit or receive UART lines are somehow not connected [you've checked this I think],
- UART flow control line (RTS) is telling the cellular module not to send data [you've checked this],
- baud rate is wrong [I guess this is unchanged from the default of 115200 which will work with all u-blox modules],
- IO voltages are wrong: please confirm that the UART lines at the cellular module are 1.8V logic, not 3.3V logic (e.g. for the standard ESP32 devkit C a level shifter would be required).

I've checked the [integration manual for SARA-R4](https://www.u-blox.com/en/docs/UBX-16029218) and on page 20 it says: "Once a valid VCC supply is applied to the SARA-R4 series modules, they remain switched off in the power-off  mode.  Then the proper toggling of the `PWR_ON`/`PWR_CTRL` input  line is necessary to trigger the switch-on routine of the modules that subsequently enter the active mode".

So the SARA-R4 series modules will NOT power on without a pin from the MCU being connected to the `PWR_ON` pin of the cellular module and the `U_CFG_APP_PIN_CELL_PWR_ON` in `u_cfg_app_platform_specific.h` being set to that MCU pin.  The `ubxlib` code will then toggle that pin in the correct way to wake the module up.

Concerning the UART pins we use on ESP32 devkit C, it doesn't matter which lines are chosen: the UART HW inside the ESP32 chip can be muxed to any IO line on the chip (with a small number of exceptions); Espressif happened to label some pins of their IO connector as the UART pins but there is nothing special about those pins, they are just 3.3V digital inputs or outputs.

Paryavi, 2021-10-05:
Thanks Rob, you're right, in sparkfun shield schematic also they connect the power-on and reset pins to the module, so I connected gpio 14 as power-on pin to Sparkfun shield D5 pin which toggles the power-On pin, I see the shield blue LED blinking as AT commands are sent to the board;

U_CELL: initialising with enable power pin not connected, PWR_ON pin 14 (0x0e) (and is toggled from 1 to 0) and VInt pin not connected.
U_CELL: initialising with enable power pin not connected, PWR_ON pin 14 (0x0e) (and is toggled from 1 to 0) and VInt pin not connected.
AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
Added network with handle -7.
Bringing up the network...
Network is not available!
1 Tests 1 Failures 0 Ignored

So I think we are one step closer to make it work! 
Actually I saw the connected sign in dashboard.hologram.io where it shows last active! Which is a very promising sign!
I set APN to "hologram"
In terminal it says Network is not available!, but in hologram dashboard website it says connected! it looks like I should work more on rx tx on feather!

Does your code connect for a while and disconnects?
Can you develop some code to get IP address of a website, or connect to something like request library, to upload pictures?
Also, I guess in your Arduin instructions you mentioned how to transfer your IDF examples to arduino.ino codes, I will check that out!

Also I will strat designing our own Eagle sara-r4 breakout board with accessing more pins! Since this Sparkfun shield is designed mainly for Arduino Uno...

Btw, To set the mobile network operator, MNO, in Sparkfun library it is set is as follows;
MNO_ATT = 2
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/src/SparkFun_LTE_Shield_Arduino_Library.h

In ublox, in some modules, they use +UMNOPROF, not sure it is supported by Sara-r4.

Regards,
MoZen

RobMeades, 2021-10-05:
Progress, excellent.  I would guess that the Hologram website showing a connection means that they have some form of back-door connection into the cellular operator and that the cellular module has powered up and registered with the network (which it will do automagically when a SIM is inserted).  This is a good step forward, we have a `PWR_ON` connection, but clearly there is absolutely no communication yet between the module and your MCU.  I guess that the blue LED you see blinking is connected to the module status pin, which matches what the Hologram website is already telling you but doesn't say anything about the lack of serial communications.

Did you check the IO voltages: remember that your MCU will be 3.3V logic, the module is 1.8V logic so without a level shifter somewhere the serial port will not work and the module may be damaged by over-voltage.

Setting the MNO profile is supported by `ubxlib` with the function [uCellCfgSetMnoProfile()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_cfg.h#L204).  You would simply add a call to that function with `networkHandle` your desired MNO profile number between the [uNetworkAdd()](https://github.com/u-blox/ubxlib/blob/preview_feature_arduino_updated/example/sockets/sockets.ino#L145) and the [uNetworkUp()](https://github.com/u-blox/ubxlib/blob/preview_feature_arduino_updated/example/sockets/sockets.ino#L152) calls in the Arduino example, e.g. `uCellCfgSetMnoProfile(networkHandle, 2);`.

But none of that is going to do anything until the UART issue is resolved of course.

Paryavi, 2021-10-05:
Thanks, Rob, yes Sparkfun lte shield has a level shifter, mentioned in the left part of page 1 of their schematic;
https://cdn.sparkfun.com/assets/3/9/d/5/f/lte_cat_m1_shield_schematic.pdf

So when I connect UART from ESP feather it goes to Sparkfun level shifter first, converting 3.3 V to 1.8V.
Yes, I bought those bidirectional level shifters too, but since Sparkfun had its own, and I can't directly access SaraR4 pins on Sparkfun shield, I refrained from using those, now I am actually designing in Eagle our own level shifter and ESP LTE shield using such level shifters you shared(I probably will have a prototype in two weeks)!

 Meanwhile, if you got time, you can take a look at these popular examples of the Sparkfun shield and see if anything in them is useful being part of your codes;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/tree/master/examples

Thanks for helping with the MNO code too.

Regards,
MoZen

RobMeades, 2021-10-05:
Hmm, that's all fine then.  One more thing to check: do make sure that the cellular module does NOT detect that the USB interface is connected to it; if it does it will ignore anything on the UART port.  In other words JP6 must be open (and, while we're at it, JP4 and JP5 must be closed).

After that I'm out of ideas, I can't see any reason why the UART comms between the MCU and the cellular module never work.

closed 2022-03-02 by Paryavi