u-blox/ubxlib #34: AssistNow Online Support

Lukanite, 2021-12-03:
Hey, I'm using the SARA-R510M8S and wanted to know if there was any way I could use AssistNow Online to assist the GPS chip? I've got CellLocate working but I'd like to get continual information from the GPS chip (or if there's another way to jump-start the GPS chip and just keep it on from there, please let me know).

Thanks!

RobMeades, 2021-12-04:
Hi there and thanks for getting in contact.

You should indeed just be able to do this yourself. For Cell Locate you probably followed the [main_loc_cell_locate.c](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_cell_locate.c) example.  To use the GNSS chip inside the SARA-R510M8S module directly you would follow the [main_loc_gnss_cell.c](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_cell.c) example; you will see that brings up a GNSS "network", which is tunnelled through the AT interface to the GNSS chip.

You can then use that GNSS network handle with the [GNSS API](https://github.com/u-blox/ubxlib/tree/master/gnss) to talk to the GNSS chip and you will see that one of those APIs is [uGnssUtilUbxTransparentSendReceive()](https://github.com/u-blox/ubxlib/blob/master/gnss/api/u_gnss_util.h#L42) where you can just talk to the GNSS chip using any [ubx protocol](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf) command you wish.

There is one complication if you want to use Cell Locate at the same time, which you can read about [here](https://github.com/u-blox/ubxlib/blob/master/gnss/api/u_gnss_pwr.h#L81), thtis is that when you use the GNSS chip directly in this way the Cell Locate code inside the cellular-side of the SARA-R510M8S module will not use it.  Don't ask me why, this is just the way the FW in the module does things, to avoid conflicts I guess.  So your code that uses the GNSS chip directly will need to give it up again for Cell Locate to use it.

RobMeades, 2021-12-04:
Hmm, just realised I may have misunderstood your question: you say "get continual information from the GPS chip" , I assumed that meant that you wanted to communicate with the GNSS chip yourself: is that correct or is it more that you would like the AssistNow service to just get on and operate in conjunction with the GNSS chip?

Lukanite, 2021-12-06:
Ah, sorry, let me clarify. 
I assume (and I might be mistaken in doing so) AssistNow Online and CellLocate are both methods to leverage the radio's internet connection to speed up the first fix of the GNSS chip. I'd just like to use one of these methods, and once a GNSS fix is available, have some method to continually get the current location from the GNSS chip each second or so from then on (whether that's via an API call or some NMEA callback function would be fine with me).

I'm currently hesitant to use CellLocate through the network handle all the time since I don't want it reaching out to the internet every time I make a uCellLocGet call once the GNSS chip is able to get a fix, and I don't know if CellLocate on the SARA-R510M8S would even keep the GNSS chip powered from call to call. 

AssistNow Online seems to be the more direct approach, but I don't see any API that would allow me to tell the cell radio to perform that information exchange with the integrated GNSS chip. Thanks for linking me that ubx protocol page; I guess I could provide my own HTTP client to interact with the radio over the sock library to download the AssistNow data, and then send that over myself to the GNSS chip? Ideally, I'd like the radio to take care of that work for me, just like when using the AT command interface.

Thanks again!

RobMeades, 2021-12-06:
Thanks, that's understood now; continuous position is your goal.  I suspect we've not addressed this use case directly in ubxlib yet, more the "one shot" behaviour of Cell Locate.  You can, of course, weave your own as you suggest but it would be a good idea for us to address this.  I am just trying to get hold of the right expert internally to advise me on the best approach and will respond here.

RobMeades, 2021-12-07:
I've found the right expert now and conclusion is as follows:

- the `+ULOC` command will always ends up talking to a server somehow or other, either to obtain position when Cell Locate is specified as the sensor or to submit the position found by GNSS if GNSS is specified as the sensor; so `+ULOC` is a good thing for  one-shots or if GNSS is not available but is the wrong thing for your "continuous position" use case.
- it seems that instead the right approach would be to use the `+UGPS` command to switch on the GNSS receiver with AssistNow aiding, and then use the `+UGRMC=1` command to enable storing of the last set of recommended minimum coordinates; the `+UGRMC?` command could then be used every second to retrieve the last stored set of coordinates until GNSS is eventually switched off.

I think that's the right approach, but of course we don't have it implemented right now.  It is probably not a great deal of work, we could maybe take a look at doing that in January?

Lukanite, 2021-12-07:
Thanks for the clarifications, Rob, and if you guys could add support that'd be awesome - I think it'd be useful since it opens up the suite of use cases where someone might want to use the GNSS in this radio for tracking on demand.

In the meantime, if I just want to keep the GNSS chip on and receive location data without any kind of acceleration, would the best approach be to call uGnssPwrOn and then just keep calling uGnssPosGet over and over?

RobMeades, 2021-12-07:
OK, it is on the list.  As indicated, it doesn't _look_ like a large job and so, provided that remains the case when we get down to the detail, I'd hope we can squeeze it in in January.

Yes, you should be able to gain position frequently as you stated provided that Cell Locate is not on (since Cell Locate would otherwise grab the GNSS chip).  Please open another issue if that doesn't work for you.

Lukanite, 2022-06-07:
Just wanted to check in and ask if there's any update on this, even though I'm sure you guys are busy with the v1 release. Cheers!

bbyers-UBCO, 2022-12-19:
@RobMeades I'm also interested in this. Any update?

owain-incus, 2023-01-24:
@RobMeades and I too am interested.

RobMeades, 2023-01-24:
Apologies for the silence, I realize this has been sitting here for a looooong time now.  We should have streaming position out for 1.2 (aiming for March), so we can see how AssistNow would (or would not) work with that then.

RobMeades, 2023-08-09:
Configuration of AssistNow is now supported, see https://github.com/u-blox/ubxlib/releases/tag/v1.3.0.

Closing this issue: if you have problems, please open another.

closed 2023-08-09 by RobMeades

elektron314, 2024-03-27:
Hello! I have a silly question perhaps. How to use AssistNow in SARA-R510M8S?
This is the quote from the assist_now_main.c
```
/** @brief This example demonstrates how to use the u-blox AssistNow
 * services to improve the time to first fix of your GNSS device; it may
 * be used where you have a GNSS device connected directly to this MCU
 * (so not connected via an intermediate cellular module).
```

RobMeades, 2024-03-27:
@elektron314 : will answer in #221.