u-blox/ubxlib #35: CME Error: memory full when using MQTT

Lukanite, 2021-12-14:
So I've just gotten started using the MQTT client on the ESP32 with the SARA-R510M8S and I seem to be running into 'memory full' issues on the radio. This causes incoming messages to be delayed, and sometimes causes outgoing messages to fail. For reference, I'm using MQTT over TLS to try to connect to AWS IoT. 

I'm not doing anything else with the radio (I was calling uGnssPosGet over and over in another thread to get a constant position, which is my end goal, but even commenting that out and just triggering this publish to occur over the command line with dummy data still causes this issue).

Any thoughts on what might fix this issue? Some logs are below.
Memory full on receive:
```
AT+UMQTTC=9,1,0,"$aws/things/sara-tracker/shadow/update",100
>{"state":{"reported":{"gnss":{"latitude":12.345600,"longitude":-23.448000}}},"clientToken":"000001"}
OK

+UUMQTTC: 9,1

+UUMQTTC: 6,1
The broker says there are 1 message(s) unread.
I (51955) SHADOW: Published!

AT+UMQTTC=6,1

+CME ERROR: memory full
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=6,1

+CME ERROR: memory full
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=6,1

+CME ERROR: memory full
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=6,1

+UMQTTC: 6,1,286,47,"$aws/things/sara-tracker/shadow/update/accepted",239,"{"state":{"reported":{"gnss":{"latitude":12.3456,"longitude":-23.448}}},"metadata":{"reported":{"gnss":{"latitude":{"timestamp":1639522422},"longitude":{"timestamp":1639522422}}}},"version":44,"timestamp":1639522422,"clientToken":"000001"}"

OK
I (54455) SHADOW: Got message from topic $aws/things/sara-tracker/shadow/update/accepted: {"state":{"reported":{"gnss":{"latitude":12.3456,"longitude":-23.448}}},"metadata":{"reported":{"gnss":{"latitude":{"timestamp":1639522422},"longitude":{"timestamp":1639522422}}}},"version":44,"timestamp":1639522422,"clientToken":"000001"}
```

Memory full on send:
```
AT+UMQTTC=9,1,0,"$aws/things/sara-tracker/shadow/update",100
+CME ERROR: memory full
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
W (170145) SHADOW: Unable to publish our message "{"state":{"reported":{"gnss":{"latitude":12.345600,"longitude":-23.448000}}},"clientToken":"000006"}"!

```

RobMeades, 2021-12-15:
Hi again: that's very weird indeed, not a `+CME ERROR` I've ever seen from a module before, which I know is not what you want to hear.  Let me check internally who knows about this sort of thing.

Could you post here the response to `ATI9` (should pop out in the AT log near power-on)?

RobMeades, 2021-12-15:
It might also be worth posting a full AT log, from power up, just to see if there's anything `ubxlib` might be doing to cause such a pile-up in the module.  Obviously delete any MQTT credentials etc. from such a log.

philwareublox, 2021-12-15:
I've seen this before when there have been a lot of MQTT messages arriving in bursts. 
What it the message reception rate typically? Do the messages come down in bursts?

Lukanite, 2021-12-15:
Hi Rob, thanks once more for the help :)

Here's that full log (with a few application log entries mixed in there). Looking at it, I think it's my application logic causing it to repeatedly get that 'memory full' error (I call `uMqttClientGetUnread` after every `uMqttClientMessageRead` regardless of return code in a loop until there are no more unread messages), but it's kind of amazing that it eventually yields the message that was pending.
<details>
<summary>Log</summary>

```
U_CELL: initialising with enable power pin not connected, PWR_ON pin 26 (0x1a) (and is toggled from 0 to 1) and VInt pin 33 (0x21) (and is 0 when module is on).
I (435) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (445) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
U_CELL_PWR: powering on, module is already on.
ATE0

OK
AT+CMEE=2

OK
ATI9

02.06,A00.01

OK
AT&C1

OK
AT&D0

OK
AT+CPSMS=0

OK
AT&K0

OK
AT+CFUN=0

OK
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type ubx AT.
ATI

SARA-R510M8S-00B-01

OK
I (1755) UBX: Network handle is 256
I (1755) UBX: GNSS handle is 1025
Bringing up the network...
U_CELL_PWR: powering on, module is already on.
ATE0

OK
AT+CMEE=2

OK
ATI9

02.06,A00.01

OK
AT&C1

OK
AT&D0

OK
AT+CPSMS=0

OK
AT&K0

OK
AT+CFUN=0

OK
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=2

OK
AT+CIMI

310260944977183

OK
AT+CIMI

310260944977183

OK
U_CELL_NET: APN from database is "epc.tmobile.com".
AT+CGDCONT=1,"IP","epc.tmobile.com"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
AT+CFUN=1

OK
AT+CREG?

+CREG: 2,0

OK
0: NReg
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?

+CEREG: 2,0

OK
0: NReg
AT+CREG?

+CREG: 2,0

OK
0: NReg
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?

+CEREG: 2,0

OK
0: NReg

+CREG: 6,"5406","00c59105",7

+CEREG: 1,"5406","00c59105",7
7: RegS
7: RegH
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"Red Pocket",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 0,0

+CGACT: 1,1

+CGACT: 3,0

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
AT+UPSDA=0,3

OK

+UUPSDA: 0,"25.67.193.14"
U_CELL_NET: connected after 4 second(s).
Checking credentials...
AT+USECMNG=4,1,"ubxlib_test_client_cert"

+USECMNG: 4,1,"ubxlib_test_client_cert","--removed--"

OK
AT+USECMNG=4,2,"ubxlib_test_client_key"

+USECMNG: 4,2,"ubxlib_test_client_key","--removed--"

OK
AT+USECMNG=4,0,"ubxlib_test_server_cert"

+USECMNG: 4,0,"ubxlib_test_server_cert","--removed--"

OK
Opening connection...
AT+USECPRF=0

OK
AT+USECPRF=0,1,3

OK
AT+USECPRF=0,3,"ubxlib_test_server_cert"

OK
AT+USECPRF=0,5,"ubxlib_test_client_cert"

OK
AT+USECPRF=0,6,"ubxlib_test_client_key"

OK
AT+USECPRF=0,0,0

OK
AT+USECPRF=0,10,"--removed--.iot.us-east-1.amazonaws.com"

OK
Connecting to MQTT broker "--removed--.iot.us-east-1.amazonaws.com:8883"...
AT+UMQTT=2,"--removed--.iot.us-east-1.amazonaws.com",8883

OK
MQTT Response error: 0
AT+UMQTT=11,1,0

OK
MQTT Response error: 0
U_CELL_MQTT: trying to connect...
AT+UMQTTC=1

OK
U_CELL_MQTT: waiting for response for up to 30 second(s)...

+UUMQTTC: 1,1
U_CELL_MQTT: connected after 1 second(s).
AT+UMQTTC=8,1

OK
Subscribing to topic "$aws/things/sara-tracker/test"...
AT+UMQTTC=4,1,"$aws/things/sara-tracker/shadow/update/delta"

OK

+UUMQTTC: 4,1,1,"$aws/things/sara-tracker/shadow/update/delta"
AT+UMQTTC=4,1,"$aws/things/sara-tracker/shadow/update/accepted"

OK

+UUMQTTC: 4,1,1,"$aws/things/sara-tracker/shadow/update/accepted"
AT+UMQTTC=4,1,"$aws/things/sara-tracker/shadow/update/rejected"

OK

+UUMQTTC: 4,1,1,"$aws/things/sara-tracker/shadow/update/rejected"
Subscribing successful
AT+UMQTTC=9,1,0,"$aws/things/sara-tracker/shadow/update",100

>{"state":{"reported":{"gnss":{"latitude":12.345600,"longitude":-23.448000}}},"clientToken":"000001"}
OK

+UUMQTTC: 9,1

+UUMQTTC: 6,1
The broker says there are 1 message(s) unread.
I (22035) SHADOW: Publish successful!

AT+UMQTTC=6,1

+CME ERROR: memory full
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=6,1

+CME ERROR: memory full
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=6,1

+CME ERROR: memory full
AT+UMQTTER

+UMQTTER: 13,0

OK
U_CELL_MQTT: error codes 13, 0.
AT+UMQTTC=6,1

+UMQTTC: 6,1,286,47,"$aws/things/sara-tracker/shadow/update/accepted",239,"{"state":{"reported":{"gnss":{"latitude":12.3456,"longitude":-23.448}}},"metadata":{"reported":{"gnss":{"latitude":{"timestamp":1639561730},"longitude":{"timestamp":1639561730}}}},"version":60,"timestamp":1639561730,"clientToken":"000001"}"

OK
I (25525) SHADOW: Got message from topic $aws/things/sara-tracker/shadow/update/accepted: {"state":{"reported":{"gnss":{"latitude":12.3456,"longitude":-23.448}}},"metadata":{"reported":{"gnss":{"latitude":{"timestamp":1639561730},"longitude":{"timestamp":1639561730}}}},"version":60,"timestamp":1639561730,"clientToken":"000001"}
```
</details>

Lukanite, 2021-12-15:
Hi @philwareublox, in this case it's occurring even with just one transaction - the device publishes a message, and the server sends down one message in response (the channels are dead silent otherwise). I'm currently just manually triggering messages to be sent, and it seems to occur sometimes on the first message pair, and I seem to always get this error by the third message pair, even if I wait a minute between sending messages.

philwareublox, 2021-12-15:
@Lukanite, oh - that's not good. I don't think this is because of your application. You should be able to read out a message if there is one there or not, and continue to do this in a loop, and when one arrives, it should just pop out. 
I've only seen this issue after I've received hundreds of messages. 
Is this very repeatable? 
Does this happen after you've powered on the module, or only after a number of hours of being on?
Are you able to take a debug trace log of the module using the USB port of the Module?

philwareublox, 2021-12-15:
@Lukanite  Thanks for the full log of the application. Ok I can see this just happens at the start of the application. Let me get back to you...

Lukanite, 2021-12-15:
@philwareublox It's quite repeatable; it happens basically instantly on the first few tries after powering on the module. I should be able to get to the USB on this thing; it'll just just take some soldering which requires me to get some sleep first, haha.

Lukanite, 2021-12-15:
@philwareublox Thanks, and ditto - I'll let you know once I get a cable fashioned and hooked up to this thing.

Lukanite, 2021-12-15:
I've got the cable made and connected now, but I seem to be having issues with not having a driver - it enumerates as "CDC-NCM Trace" with VID 1546 and PID 1223. Is this a thing I can download from somewhere?

Lukanite, 2021-12-16:
Was fooling around with the device in the meantime and discovered that it only happens on some SIMs - my Red Pocket (T-Mobile) and Mint Mobile (T-Mobile) SIMs show this issue, but my Hologram.io (T-Mobile) SIM does not. 

I'm really not sure what to think now, but hopefully that helps you out (and at least I have a bit of a workaround now, even though it still takes around 10 seconds to send/receive each transaction, is that normal?)

Lukanite, 2021-12-16:
Drat, spoke too soon. The Hologram.io SIM also fell victim to it; completed about 7 transactions and then couldn't do any more due to 'memory full'.

RobMeades, 2021-12-21:
Apologies for the delay in getting back to you.  Can you give me an e-mail address I can send the [Windows 10] NCM drivers to?

Lukanite, 2021-12-22:
No worries Rob, sent you an email :). 

It looks like it's something with the MQTT client itself; I've managed to hack together core_mqtt on top of the TLS socket driver here and that seems to be working, but trying to use this client either encrypted or not seems to run into this issue.

RobMeades, 2021-12-22:
Interesting: have sent you an e-mail with the logging drivers and instructions on how to capture a log.  Would be really useful to get to the bottom of this.

Lukanite, 2022-01-10:
Any update on the status of this? No rush, since coreMQTT is working for now but I'm curious about this odd issue.

RobMeades, 2022-01-11:
Apologies for the delay, our internal module-related support processes now picking up on this will reply with more detail via e-mail where I can name names.

RobMeades, 2022-01-20:
Hi again.  I _think_ we have determined what is going on here and, in so-doing, I have learned something; I know what I can do to prevent it happening in `ubxlib` and what you could do to workaround the problem in the application.

The issue is that the inactivity timeout in the module (which is set with `uCellMqttSetInactivityTimeout()`, i.e. `AT+UMQTT=10,<timeout>`) is at the default value, which is 0.  Then `uCellMqttSetKeepAliveOn()` is called (which sends `AT+UMQTTC=8,1`) .  The MQTT implementation in the module takes these values literally, i.e. just before zero seconds have elapsed a ping should be sent to the server.  These outgoing MQTT ping messages pretty much immediately fill up the outgoing message queue and you get the "queues full" result.

I knew about the dangers of setting an inactivity timeout value of 0 and so I check for this and stop it being set, what I hadn't appreciated is that setting ping to "on" _without_ setting a non-zero inactivity timeout first would be just as bad.  It so happens that in my testing I do set the inactivity timeout first and I have to leave it there because of my cross-check, so I do not see this in any of my testing, the inactivity timeout is never at 0 when I set ping to "on".

From the point of view of the application, if it called `uCellMqttSetInactivityTimeout()` first, then provided the inactivity timeout is set it to a minute or few, the problem should go away.  From the point of view of `ubxlib`, I will make a change to read the inactivity timeout before I set ping to "on"; if the inactivity timeout is 0 I will return an error instead of switching ping on.

Does all of that makes sense?  As you say, rather a good learning experience, thanks very much for raising this!

Lukanite, 2022-01-20:
Hi Rob,

What a breakdown! That makes a lot of sense, and 'my device has turned into a ping-flooding weapon' is a fun answer to this query. I switched my application back to using ubxlib for MQTT, added the timeout (60 seconds), and was able to get 50 messages up and 50 messages down, so I do believe the workaround is working on my end. Resolution sounds good, too.

Thanks again for all of your help with this, and for enlightening me with the details! You guys are awesome, and I look forward to using more of ubxlib as it grows :)

(feel free to close this at your leisure)

RobMeades, 2022-01-20:
Kinda scary that we can create ping-flooding weapon to be frank with you ;-(.  I will link the eventual resolution of the problem to this issue so will leave it open for now.

RobMeades, 2022-02-09:
Now fixed and pushed here in commit 169fc35353ae05e379c51a5709125eccec0b7dad.

closed 2022-02-09 by RobMeades