u-blox/ubxlib #37: [GNSS Module] Streaming Position Processing

adpjhu, 2022-01-04:
The GNSS module provides a mechanism to poll the GNSS for its current location and to get a call back when the that location has been received. However it does not seem to support a mechanism to process and store points if they are being sent by the GNSS not as a response to a poll. This would be a helpful feature!

RobMeades, 2022-01-04:
Hi there and yes, agreed this is a missing feature. Did we discuss it already? I have a feeling I did with someone but there's been Christmas and the new year since then....

What would the API look like?  Would it be along the lines of starting an asynchronous position request but leaving it running, so your callback would be called with new position fixes as they occur until you tell the API to stop, possibly by having the callback function return a Boolean result which would be `true` for continue and `false` to stop (or, if you're not in the callback, just setting the callback function to `NULL`)?

RobMeades, 2022-01-05:
Apologies, getting my feet back under the desk in 2022 and now realize that this, or at least the resolution to this, is kind of similar to [https://github.com/u-blox/ubxlib/issues/34](https://github.com/u-blox/ubxlib/issues/34).  We'll try to get this done ASAP, though power saving may have to come first since there are quite a few people waiting on that now.

adpjhu, 2022-01-12:
I agree power saving is more important!  Yes there are some parallels w/ #34  but i think this is perhaps a sub ticket of some sort. That other ticket talks about constantly streaming the assistance data into the GNSS as well as streaming the GNSS navigation/position out to the MCU. I am referring specifically to support in the GNSS module ( [u_gnss_pos.c](https://github.com/u-blox/ubxlib/blob/master/gnss/src/u_gnss_pos.c#L280) ) not in the location module. We may be saying the same thing just making sure. 

> What would the API look like? 

In terms of a mechanism to get the GNSS data to the "client", I think your suggestion of repeatedly calling the callback as new location data arrives would certainly work! 

An alternative could be to add add the locations to a queue ( as supported by [u_port_os.h](https://github.com/u-blox/ubxlib/blob/master/port/api/u_port_os.h#L170) ).

RobMeades, 2022-01-13:
Understood, yes, I think it is probably a subset.  Will get to it after power saving, realistically March I think.

adpjhu, 2022-01-27:
This might be obvious but the GNSS can be configured to send position at a regular interval rather than sending a poll request and waiting for a response. Configuring a regular update and asynchronous processing the result is what i have in mind here.

RobMeades, 2022-01-27:
Understood, though I don't think you need most of `ubxlib` for that: just open the serial port and absorb NMEA into a thing of your own.

The reason I was going down the route described in #34 is because that would work just as well for a GNSS chip attached via a cellular module (rather than attached directly to the MCU) without having to implement support for a 3GPP `CMUX` to support another asynchronous stream.

The default epoch is one second and so streaming it or reading it once a second (which could be from file storage in the cellular module) would lead to a worst-case latency of 2 seconds; not sure if that's a problem of course.

Anyway, there's time to thing about this, power saving first...

adpjhu, 2022-02-03:
In our particular case, we are using the GNSS connected directly to the MCU and benefit from the use of ubxlib as we need to configure the GNSS based on our needs (we don't use all of the default settings). Perhaps #34 represents a more likely use case.

RobMeades, 2022-10-01:
FYI, the GNSS API now includes a [uGnssMsg](https://github.com/u-blox/ubxlib/blob/master/gnss/api/u_gnss_msg.h) API which allows you to register an interest for a given message type, in any supported protocol (UBX, NMEA or RTCM) and have one of your callbacks called when such a message arrives, which would allow you to receive such a stream, this for the case where your GNSS chip is connected directly to your MCU.

You can find an example of how to do this [[here](https://github.com/u-blox/ubxlib/blob/master/example/gnss/msg_main.c)](https://github.com/u-blox/ubxlib/blob/master/example/gnss/msg_main.c).

This doesn't yet work for the case where the GNSS chip is connected via the cellular module, that will come when we've added `CMUX` support.

owain-incus, 2023-01-24:
Any updates of when the CMUX support will come?
We are now wishing we had connected Sara (combined cellular and GNSS) in such a way that we separate comms channels for GNSS and the modem.
We are seeing the serialisation of modem commands is the limiting factor of the rate at which we can get GNSS updates.
We may move away from MQTT as a transport to avoid these issues.

RobMeades, 2023-01-24:
Hi @owain-incus, funny you should ask that but it's what I've been writing/debugging for the last little while.  Had to take a break to re-architect the tests system in the middle (for other reasons), which is why it has taken so long.  Plan is for it all to be released by mid March.

closed 2023-03-13 by RobMeades

elektron314, 2023-11-02:
Hello!
Could anybody please help me set up SARA-R510M8S so that I could interact with the GNSS module via multiplexed main UART (RX/TX only)?

What's the channel of interaction of the cell module with the internal GNSS module? Should I consider it? Is it AT-command-based as well?

Example from main_loc_gnss_cell_continuous.c doesn't work because `uDeviceOpen(&gDeviceCfg, &devHandle); `returns -5 error, which means that some parameters are wrong. But which parameters can be wrong if devHandle is NULL and gDeviceCfg is just `static const uDeviceCfg_t gDeviceCfg = {.deviceType = U_DEVICE_TYPE_NONE};` ?

I have also tried to set up CMUX, so multiplexer is enabled successfully but the position is not coming. So I first set up the cell network, tested it after connection, and tried to add the GNSS part with CMUX. CMUX is enabled. uCellMuxAddChannel is ok. uGnssAdd is ok. Then something strange is happening. I don't understand why before uGnssPosGet everything is sending AT-commands and when the uGnssPosGet is executing, it is sending some data somewhere.
like
`U_GNSS: sent command b5 62 01 07 00 00 08 19`
Why is it sending some data and where? I'm trying to get the location. It should use AT commands, no?
Before that, I could see a lot of AT commands setting up the cellular module and testing it and then some strange hex data instead of AT commands. As a result - no position data.