u-blox/ubxlib #38: invalid heap address used in zephyr uNetworkInit()

owain-incus, 2022-01-06:
I was using lib 0.9.1 OK; but just rebased to master latest commit  bb05f55583e90105758c90febcebc293fa21005f; as I wanted to pick up some changes to the AT client.

Here is description as sent to Rob....

Progressed OK with 0.9.1 and the SMS related stuff; coded to use your at partial string write; for assembling the actual sms modem command; have already switched modem to sms text mode via other sequences. Then saw this API was not in your latest 0.9.1 release; so I have tried to move to your latest commit on master and this breaks my build.

Break is in initialisation code where uPortQueueCreate() is called; faulting in the k_msq_alloc_init() heap allocation. Working back from this checking where heap stuff is init'ed I see you have some kind of kludge in u_port_os.c and have called k_thread_system_pool_assign().

The heap that is trying to be used later is (0xAAAAAAAA random unitialised).

But what if that initialisation is not called from z_main_thread; which in my case it is not. In my test harness I am calling your init functions from my debug console shell. You added kludge for main thread; but is this zephyr bug you mention the same for all threads; the current thread resource_pool is for me not initialised (0xAAAAAAAA).

Just looking if I can just cherry pick your changes for the uAtClientWritePartialString(); but if I can then I will probably only be bitten by the same issue later.

Any suggestions to progress?
[image]

owain-incus, 2022-01-06:
I can confirm running the initialisation from main thread appears to work ok.

antevir, 2022-01-10:
Hi Owain!
Unfortunately there is a bug in Zephyr that leaves the main thread allocation pool uninitialized. I am hoping to get a fix for this upstream when we have some slack. However, even with a fix for the uninitialized allocation pool the default behaviour of zephyr is to not allocate any resource pools to any threads:
```
Thread Resource Pools
*********************

Certain kernel APIs may need to make heap allocations on behalf of the
calling thread. For example, some initialization APIs for objects like
pipes and message queues may need to allocate a private kernel-side buffer,
or objects like queues may temporarily allocate kernel data structures
as items are placed in the queue.

Such memory allocations are drawn from memory pools that are assigned to
a thread. By default, a thread in the system has no resource pool and
any allocations made on its behalf will fail. The supervisor-mode only
:c:func:`k_thread_resource_pool_assign` will associate any implicit
kernel-side allocations to the target thread with the provided memory pool,
and any children of that thread will inherit this assignment.

If a system heap exists, threads may alternatively have their resources
drawn from it using the :c:func:`k_thread_system_pool_assign` API.
```

To make this work for ubxlib `uPortOsPrivateInit()` will setup the Zephyr main thread to use the system pool, and as new threads inherits the parent thread allocation pool all new child threads will also get the pool correctly setup.
If you are calling ubxlib from a thread doesn't inherent the fix you need to call `k_thread_system_pool_assign()`. The function takes a `k_thread *` as argument which is the same thing as a `k_tid_t` so if you want to assign it for current thread you can use: `k_thread_system_pool_assign(k_current_get())`.
In the Zephyr shell case you can also get a hold of the k_thread through the `shell` variable: 
```c
static int cmd_dummy(const struct shell *shell, size_t argc, char **argv)
{
   k_thread_system_pool_assign(shell->thread);
}
```

owain-incus, 2022-01-10:
Its not causing me any issues at the moment. No need to escalate on my part.
I have changed the way I do things and am currently back on 0.9.1.

mekm, 2022-10-12:
A fix fb63a1f22333c840464a658c708327c04191826e is now available that initializes the memory pool early during boot. Thus threads created afterwards will get the pool correctly setup.

closed 2022-10-12 by mekm