u-blox/ubxlib #39: HTTP API

ferremarco, 2022-02-09:


RobMeades, 2022-02-09:
Thanks for the suggestion, it's on the list.

RobMeades, 2022-10-13:
FYI, I've just pushed a [preview branch](https://github.com/u-blox/ubxlib/tree/preview_feature_http_rmea) of HTTP support.  It is only a preview branch because we have some test issues to sort out which prevent me getting it onto `master` and because it doesn't yet work over `wifi`, just `cell`, hence there may be small API tweaks when we add support for `wifi`.

However, if you are waiting for HTTP support and want to get started, this branch should be good.  I will update this issue when HTTP support arrives on `master` and will delete the preview branch some time after that.

RobMeades, 2023-01-18:
HTTP support is now pushed to `master` here, latest relevant commit 925403bd434b9ae05b6ebba2ac6ad7512e6c5a0e, though note that it doesn't yet support HTTP over Wifi, that will come along at the end of March.  I will close this issue now and delete the preview branch in a few weeks.

closed 2023-01-18 by RobMeades