u-blox/ubxlib #41: Testing lte-cfg example with ESP32 Feather and Sparkfun LTE shield

Paryavi, 2022-03-16:
Hi Rob and other Ublox people!

I hope all is well, 

We had good progress in soldering our SARA modules on our ESP camera board, and the modules boot up now with correct values for Power_ON and V_INT pins (we have bought 30 modules for prototyping so far).
Now I need to make sure of the software side. So I am testing your lte-cfg example code on ESP-IDF 4.2, using ESP32 feather board connected to Sparkfun shield(as a break-out board for SARA R410M_02B-01) with a hologram Simcard.

Previously I tested this Sparkfun LTE shield with Arduino board and Sparkfun code, so Simcard, antenna, and the shield are fine, I used AT&T as MNO there and "hologram" as APN in their code;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/examples/00_Register_Operator/00_Register_Operator.ino

Ok, to test ublox lte-cfg code now, ESP32 is connected to SARA shield Serial port with four pins GND, V_REF(3.3 from ESP32 reaches Sparkfun shield level shifter), and of course RX and TX pins. 

# define U_CFG_APP_CELL_UART                  1
# define U_CFG_APP_PIN_CELL_TXD              17
# define U_CFG_APP_PIN_CELL_RXD              16

Flashing the code, the ESP-IDF prompt, for example 9 (lte-cfg) starts printing; 
U_CELL: initialising with enable power pin not connected, PWR_ON pin not connected and VInt pin not connected.
AT
AT
...
U_CELL_PWR: powering on.
AT
AT
...

And In the end, the errors I get in the prompt terminal are;

### Added network with handle -7.
### This module does not support setting an MNO profile.
### unable to read bandmask!
### Bringing up the network...
### Unable to bring up the network!
### Done.
D:/UBLOX/ubxlib/example/cell/lte_cfg/lte_cfg_main.c:455:exampleCellLteCfg:FAIL:Function [example].  Expression Evaluated To FALSE
Test ran in 6645ms

I am not sure why it is not able to read bandmask or MNO profile is not supported on R410M_02B.
My firmware version is; L0.0.00.00.05.08,A.02.04

My simcard, hologram, should be MNO agnostic but reading about MNO settings in AT command datasheet for the United States, I set the MNO to 2 or 5 for AT&T and TMobile respectively and APN to "phone" or "hologram" or a few other values mentioned for Tmobile in u_cell_apn_db.h like "fast.tmobile.com" , but I get the same error. Any suggestions?

Things I modify in ubxlib library files;

# define MY_MNO_PROFILE 2    //i.e. AT&T, I tried 0 and 5 too
#define U_CFG_TEST_CELL_MODULE_TYPE U_CELL_MODULE_TYPE_SARA_R410M_02B
#ifdef U_CFG_TEST_CELL_MODULE_TYPE
static const uNetworkConfigurationCell_t gConfigCell = {U_NETWORK_TYPE_CELL,
                                                        U_CFG_TEST_CELL_MODULE_TYPE,
                                                        NULL, /* SIM pin */
                                                        "phone", /* APN: NULL to accept default. 
                                                        240, 

Btw, in the Hologram Dashboard website, sometimes it says the Simcard was connected so it is probably related to MNO, RAT, and CATM1 settings...

Thanks,
Mohsen

RobMeades, 2022-03-16:
Hi @Paryavi: from what you've pasted in above we haven't even got to any bi-directional AT interface activity yet, we should re-check the basics.

From what I can see you do not have the `PWR_ON` pin of the module connected to the ESP32; as far as I am aware SARA-R410M-02B can _only_ be powered up by that pin being toggled for the correct time period by the `ubxlib` code so I don't believe the module is on...?

[image]

`ubxlib` knows all about the correct toggle time etc. for SARA-R4 modules, but for it to be able to power the module up that `PWR_ON` pin must be connected to the MCU, SARA-R410M-02B does not just power on when power is applied to the VCC pins.

The other thing is that you should make sure that the module's RTS line is grounded, otherwise it will float to "not-asserted" and the module will assume that the MCU's serial port is not ready to accept its data and it won't send anything back even when it does receive an AT command.

Paryavi, 2022-03-16:
Aloha Rob, 

Yes RTS is grounded in Sparkfun shield. 

For Power_ON: At first, To have Open Drain, I was using a transistor on Sparkfun level shifter, connecting the Gate to MCU GPIO and its drain to Sparkfun shield power_On pin;  and source to ground. I need to double-check and implement that again.  In code, I need to have Power_ON from MCU as inverting in IDF, so I guess I need to use flags for that.
https://github.com/u-blox/ubxlib/tree/master/port/platform/esp-idf/mcu/esp32/runner
Yes, sometimes I  was pushing the Sparkfun Power-On button manually and it was booting up but no bi-directional AT activity yet.

It was a mystery to me how they turn the module on without connecting the Power_ON signal to Arduino in their design, now I see they sneakily(in PCB eagle file) connected the Power_ON signal from Power_ON hardware switch to Arduino D5, D6;
https://cdn.sparkfun.com/assets/a/c/b/1/0/lte_cat_m1_shield_sara-r4_schematic.pdf
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/examples/00_Register_Operator/00_Register_Operator.ino

Thanks

RobMeades, 2022-03-16:
Thanks for confirming that `RTS` is grounded.  With your Saleae probe what you could now do is have it connected to `PWR_ON` and `VINT`  (closest to the module), plus the `TXD` and `RXD` lines (probably closest to the module again) then capture a trace of, say 15 seconds around the time of powering on and we should be able to see what's going on.

Paryavi, 2022-03-16:
Sure, will do. 

I have a question, Looking at the Sparkfun LTE shield schematic, as I said they directly connected D5 to module PWR_ON pin, so it looks like I do not need a level shifter between the POwer_ON pin and an ESP MCU pin, right? PWR_ON has 200 KOhm resistance. So Can I directly connect ESP 32 GPIO pin(which ranges from 0 to 3.3 Volts) to the SaRA module Power_ON pin? 
Normally when I connect Lipo to LTE shield, PWR_ON is 0.8 V.

RobMeades, 2022-03-16:
I'm no HW expert but in our test farm, wired together by a SW engineer (me), I do connect an ESP32 pin *directly* to the `PWR_ON` pin of the module, no inverting, no level shifting; the code will set drive of the MCU pin that is connected to `PWR_ON` as open drain in this case and it seems to work, at least to a SW engineer's standards ;-).

Paryavi, 2022-03-16:
Thanks Rob, I directly connected MCU GPIO 19 to PWR_ON, and it boots up! Blue LED on Sparkfun shield blinks and after it prints in IDF terminal; U_CELL_PWR, I measure V_Int signal on the shield toggles from 0 to 1.8 V for a few seconds, then goes to zero, AT commands still print, and with second print of U_CELL_PWR it goes to 0 to 1.8 V again for 20 to 30 seconds and then Unable to bring up the network;

More detail:
Running exampleCellLteCfg...
I (8336) uart: queue free spaces: 20
U_CELL: initialising with enable power pin not connected, PWR_ON pin 19 (0x13) (and is toggled from 1 to 0) and VInt pin not connected.
I (8356) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
AT
AT...
Added network with handle -7.
This module does not support setting an MNO profile.
unable to read bandmask!
Bringing up the network...
Unable to bring up the network!
Done.
D:/UBLOX/ubxlib/example/cell/lte_cfg/lte_cfg_main.c:455:exampleCellLteCfg:FAIL:Function [example].  Expression Evaluated To FALSE
Test ran in 21293ms

I switched MNO back from # define MY_MNO_PROFILE 2  to # define MY_MNO_PROFILE U_CELL_TEST_CFG_MNO_PROFILE and APN back to "hologram".

Before or after that I see a connection on hologram sim;
[image]

Btw, Shield LED randomly blinks blue (in several minutes intervals) even after lte-cfg example prints are done in IDF terminal! Probably because it scans through a big NB-1 or 2G RATs list for finding the right frequency.... Now I start working on Salea 15 sec capture accordingly.

RobMeades, 2022-03-16:
Very good, we're getting a bit closer.  That Saleae trace should, hopefully, tell us a lot.

Paryavi, 2022-03-17:
Using Salea Probe, Starting when IDF prints the U_CELL_PWR, using example 9;
[image]
Zoom on the first 15 seconds of same session;
[image]
V_INT provides the 1.8 Volt on the TX/RX level shifter of Sparkfun shield(TXB0104), not sure why it goes to zero after 20 seconds When it toggles to zero, I assume the AT commands of MCU TX won't reach the module probably(I was not able to clip TXD or RXD on module side yet).

RobMeades, 2022-03-17:
Good traces, interesting.

In the trace I can see that the `PWR_ON` toggle is correct, `VINT` rises as expected and that there is a 6 second wait before the `TXD` activity, which is the `ubxlib` code sending `AT` to the module to see if it is awake, but there is never any response from the module.

The suspicious thing, of course, is that `VINT` starts glitching, even before that attempt is made, at around 5 seconds.  If `VINT` really is going low then that's very bad, the module is not running properly; the most likely cause is that the module is not getting sufficient current to sustain itself and hence VCC is drooping below the required level.  It is quite likely that at around 5 seconds the module is powering up its radio to talk to the network so there could be a surge of 1 Amp or more at that time.  What is the source of power to the Sparkfun shield?  You likely need more than USB, unless this is USB 3, and even then USB 3 may not be capable of delivering the peaks required in time.  Anyway, let's check this: could you attach a Saleae probe to the module's `VCC` line (as close to the module as possible) and try again?

One thing, just to double-double-double check (sorry if I've asked this before and apologies in advance when I forget and ask it again!): can you confirm that the `TXD` output from the MCU is connected to the `TXD` input of the module, and the `RXD` input of the MCU to the `RXD` output of the module, i.e. they are not "crossed over", it is `TXD`->`TXD` and `RXD`->`RXD`?  Just that it's a mistake many people (me included) make.

Paryavi, 2022-03-18:
Thanks ROb, yes it was weird VINT going low, and I was connecting a 2000 mAh Lipo battery with 4 volts voltage remained on it.(fully charge is 4.2V). Yes I always check the TX and RX connections of MCU and module.
So today I hooked up a Power supply instead and it looks like the problem is not the power supply.
[image]
Zoom on first 30 seconds;
[image]

I define MY_MNO_PROFILE as 2 .On the hologram dashboard it says you are connected to AT&T. Also as far as I remember with connecting Arduino instead of ESP32 to sparkfun shield, I was using MNO as AT&T which is 2 here;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/src/SparkFun_LTE_Shield_Arduino_Library.h#L69
And APN I used was "hologram".
So I connected to the network more than ten times but I do not receive Serial response from the module.
It sounds like we should disable NBIOT for this code. right? The code sends AT commands for less than a minute and then give up,  and after a while, I see in Hologram dashboard website that my simcard is connected. Sparkfun guide says it might take up to 3 minutes to connect to a network in USA. I am not sure if you have tested your codes in the United States.

RobMeades, 2022-03-18:
In these traces, the ones with the power supply used, `VINT` is definitely behaving properly, so that's good (though the LiPo you describe should have been quite sufficient).  `PWR_ON` is now other way up though; have you gone back to using the inverter?

Anyway, from what you say, and from what `VINT` says, the module is on but we never, ever, get ANY serial data back from it, ever.  Nothing you set in the code, nothing to do with MNO profile or APN, is going to make the slightest difference because the `ubxlib` code can only communicate with the module via the AT interface.  Until we see `AT`->`OK`, either in your Saleae trace or in your `printf()`'ed trace, we have a much more fundamental problem than network configuration.

One of the following must be true:

1. the `TXD` pin you are using on the MCU (which I gather is 17) is not connected to the `TXD` pin of the module,
2. there is something wrong in the level shifting of the `TXD` pin so that it is not at 1.8V-logic at the module, or not the right way up, or something.
3. the `RXD` pin from the module is not connected to the pin you are using for `RXD` on the module (which I gather is 16),
4. the level shifting on the `RXD` pin is somehow going wrong and it does not sense 1.8V-logic from the module correctly.
5. the `RTS` pin of the module is not grounded properly somehow.

To eliminate some of these, would it be possible to probe all of these pins right up close to the module, **on the module-side of the level shifter**?

Paryavi, 2022-03-19:
Anyway, from what you say, and from what VINT says, the module is on but we never, ever, get ANY serial data back from it, ever.

Thanks, Rob, I checeked the connections, they are fine, the level shifter also works fine;
[image]

 because I get TXD signal from MCU on the module side(1.8 volts side of the level shifter), and its not in this trace, but holding the clipper I get

Paryavi, 2022-03-19:
A quick question, I want to test lte-cfg IDF example on our integrated camera board too, but I have some signals inverted.

In the documentation; https://github.com/u-blox/ubxlib/tree/master/port/platform/esp-idf/mcu/esp32#sdk-usage
Where should I set an environment variable for inverting pins?

For example, to invert the PWR_ON pin, Should I add the following flag; 
set U_FLAGS=-DMY_FLAG -DU_CELL_PWR_ON_PIN_INVERTED

RobMeades, 2022-03-19:
This is getting a bit difficult to follow: assuming TXD is all good and you say that you get "stable 1.8V" on the RXD line but in the posts on this thread I've not seen any indication yet that you are receiving any actual _data_ back from the module on the RXD pin; can you, in the Saleae application, attach a serial analyzer to the TXD and RXD lines so that it displays the data, and confirm that you are receiving responses to AT commands from the module at your MCU?

On the compilation flag, how you pass that into the build will depend on how you are building the code. If you are using arduino-cli you would pass the build options "--build-property "compiler.c.extra_flags=-Dblah"" and "--build-property "compiler.cpp.extra_flags=-Dblah"" to the "compile" command, but you will need to check exactly how this is done for whatever build mechanism you are using.

Paryavi, 2022-04-14:
Hi Rob, happy to tell you I was able to run u-blox Arduino code directly from Arduino IDE;

+CGREG=2
OK
AT+CEREG=2
...U_CELL_NET: user-specified APN is "hologram".
AT+CGDCONT=1,"IP","hologram" ...

I implemented the cell enable power inversion I needed with pin definitions, and for Power-ON and Reset-N pins actually we don't need inversion and we're going to remove these extra transistors. So I skipped Arduino CLI and inversion steps and the code runs on ESP board for my Sparkfun R410 LTE shield and today on R42M8S EVKR4 board I purchased recently. 
last month I also modified Sparkfun u-blox codes to work on ESP32; https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/tree/master/src

I have a question; For our integration boards, we have UART0 connected to the module, when I was using Sparkfun codes I moved debug pins to other UARTs;

#define SerialMonitor Serial2  //uses UARTS 2 pins for Serial monitor (pins 16 and 17 of ESP32)
#define LTEShieldSerial Serial  //uses UARTS 0 pins for Serial monitor, as I wanted because in my board UART0 is connected to the module.

So how can I have similar funcionality for printf in the u-blox Arduino code example? I skimmed through u_port_uart.c and h a bit.
I tried fclose(stdout) in the setup and loop but it wipes out everything.
Also, I tried something like the following at the beginning of the code but the compiler says printf sometimes has more than one argument. 

#define DEBUG 0
#if DEBUG == 1
#define printf(x) printf(x)
#else
#define printf(x)               //right, if debug is 1 then I want to remove printf to be nothing!
#endif

Thanks,
MoZen

RobMeades, 2022-04-14:
Woohoo!  Well DONE!  Great that you have it working.

As to your question about debug output, all that `ubxlib` does is to call the ESP-IDF `printf()` function, so wherever the output from the ESP-IDF library function `printf()` is directed to by Arduino is where it will come out.  If you were using ESP-IDF standalone then it [would be possible to redirect this](https://demo-dijiudu.readthedocs.io/en/latest/api-reference/kconfig.html#console-uart) but I believe that under Arduino a [pre-compiled ESP-IDF library](https://github.com/u-blox/ubxlib/tree/master/port/platform/arduino#limitations-of-esp-idf-under-arduino) is used and so the destination for `stdout` is, I think, fixed, but I might be wrong.

To switch all debug output from `ubxlib` off then you would set the conditional compilation flag `U_CFG_ENABLE_LOGGING=0` into the build.

Paryavi, 2022-04-14:
Thank you Rob for all your advice, it most probably was a faulty UART on my ESP32-feather, since when I used a different ESP32, ESP32 DEVKit-V1, it started working! Thanks for the info about the debug output too, we might use a different UART for communicating with the module then.

Since I have the EVKR4 board, my next goal is to read GPS data from R422M8S, TXD_GNSS pin, pin 47 of the module(19 on J301). On our prototype board, we have connected this pin to our ESP32 MCU. Which u-blox gnss example do you recommend I look into? Some examples assume external GPS modules compared to R422M8S which has a GNSS chip inside.
Here is a picture of ESP32, level shifter, and EVKR4;
[image]

RobMeades, 2022-04-14:
Hmm, so your EVK is doubly-endowed with GNSS chips: one inside the SARA-R422M8S chip and one separately on the EVK itself. The `ubxlib` code supports both but, in the above diagram, you have connected to a pin of the SARA-R422M8S thar is a kind of weird half-way house I didn't even know it had: GNSS output only, no input (i.e. no corresponding GNSS RXD line).

The simplest thing to do is to entirely ignore that pin: the `ubxlib` code will just talk to the GNSS chip inside the SARA-R422M8S module via `AT+UBX` AT commands. All you need to do is to say there's an M8 GNSS chip present and set the transport type to `U_GNSS_TRANSPORT_AT` as described [in this example](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_cell.c) **but**, actually, it is MUCH easier if you follow [this example](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_cell_locate.c): this uses the u-blox Cell Locate service which will employ the GNSS signal if it is present (i.e. you get the GNSS accuracy) or, if GNSS is not available for some reason, it will give you an approximate fix based on cell towers. The radius of position number well let you know what accuracy has been achieved.

If you _really_ what to use that `GNSS_TXD` pin then it is _possible_ that the cellular module configures the internal GNSS chip to leave NMEA messages on in the GNSS chip output stream and you can parse these yourself by tapping into that pin and bringing in your own NMEA parsing library but (a) it might not and (b) this is not something that `ubxlib` will do for you, you'll need to figure it out yourself I'm afraid.

Paryavi, 2022-07-19:
Hi Rob and ublox friends! 
Congrats on the new code release(V1.0), I realized Arduino code is also updated so I gave it a shot today. With the first encounter, I got the following compilation error, unfortunately;

sketch_jul18c:89:1: error: 'uDeviceCfgCell_t' has no non-static data member named 'pPin'
 };
 ^
C:\Users\myuser\Documents\Arduino\sketch_jul18c\sketch_jul18c.ino: In function 'void loop()':
sketch_jul18c:277:27: error: too few arguments to function 'int32_t uDeviceClose(uDeviceHandle_t, bool)'
     uDeviceClose(devHandle);
                           ^
In file included from C:\Users\myuser\Documents\Arduino\libraries\ubxlib\src/ubxlib.h:69,
                 from C:\Users\myuser\Documents\Arduino\sketch_jul18c\sketch_jul18c.ino:23:
C:\Users\myuser\Documents\Arduino\libraries\ubxlib\src/u_device.h:324:9: note: declared here
 int32_t uDeviceClose(uDeviceHandle_t devHandle, bool powerOff);
         ^~~~~~~~~~~~
exit status 1
'uDeviceCfgCell_t' has no non-static data member named 'pPin'

I did not change 'pPin' as I do not use SIM pin in my application. Any comments on how to fix it?

Regards,
MoZen

RobMeades, 2022-07-19:
Hi MoZen: yes, apologies, noticed that myself yesterday, looks as though Arduino-CLI isn't rebuilding the main application file for this example for some reason in our test system and so we hadn't spotted this.  Will post a fix later today.

Paryavi, 2022-07-19:
Thanks for quick reply, now I am using ESP32S3 for our camera board application, (we needed more GPIOs) so I tried rolling back to the previous ublox release like 0.10? after compile error you mentioned.
but in the Previous release of this socket code, the code was compiling for the esp32 older chip version(like if I select esp32 dev module in board selection of IDE, which is for ESP32 wroom chip we had previously), but as I have updated the chip to ESP32 S3 physically (now I select their ESP32 S3 Dev board in Arduino IDE), and the compiler(for the Previous release of this socket code(v 0.10)) suggested I change the UART_SCLK_REF_TICK to UART_SCLK_RTC in u_port_uart.c, so I changed and it compiled, but in my UART instead of seeing prints like fox jumped on lazy cat I see jibberish unreadable signs at 115200 baudrate, but it connects to the network (blue LED also Hologram dash shows connection) as I previously initialized the S3 R410 with Sparkfun LTE codes, so I lost my serial prints somehow!

Paryavi, 2022-07-19:
Btw, as you know I wanna send a camera frame buffer(fb) to our website using our ublox modem, if I bypass the modem and use WiFi, I have tested it successfully with HTTP client, do we have a kinda gsm client in your API?

A typical ESPCAM HTTP CLinent code I used;
// Take Picture with ESP Camera
  fb = esp_camera_fb_get();
  HTTPClient http;
  Serial.print("[HTTP] begin...\n");
  // configure traged server and url
  http.begin(post_url); //HTTP
  Serial.print("[HTTP] POST...\n");
  // start connection and send HTTP header
  int httpCode = http.sendRequest("POST", fb->buf, fb->len); // we simply put the whole image in the post body.
  // httpCode will be negative on error
  if (httpCode > 0)
  {
    // HTTP header has been send and Server response header has been handled
    Serial.printf("[HTTP] POST... code: %d\n", httpCode);
    // file found at server
    if (httpCode == HTTP_CODE_OK)
    {
      String payload = http.getString();
      Serial.println(payload);
    }
  }
  else
  {
    Serial.printf("[HTTP] POST... failed, error: %s\n", http.errorToString(httpCode).c_str());
  }
  http.end();
  esp_camera_fb_return(fb);
  
  So this line is the meat of the code: int httpCode = http.sendRequest("POST", fb->buf, fb->len);
  
  Now instead of WiFi, we want a sort of LTE(or GSM) Client!
  Since the framebuffer is uint_8; unsigned integer 8 bit = 0 to 255-pixel values, I needed a type casting for using ublox string codes probably; 
(const char *)fb->buf
Actually, esp camera code suggested this casting too; https://github.com/espressif/esp32-camera#jpeg-http-capture
So in your code; https://github.com/u-blox/ubxlib/blob/master/example/sockets/sockets.ino#L222
I need to modify the code a bit, probably changing

uSockWrite(sock, message, txSize)
to
uSockWrite(sock, (const char *)fb->buf, fb->len)
Also changing uSockGetHostByName(devHandle, "ubxlib.it-sgn.u-blox.com",
                           &(address.ipAddress));
to
uSockGetHostByName(devHandle, "example.com/uploads",
                           &(address.ipAddress));                           
                           
Btw that easy http client code was from https://gist.github.com/klucsik/711a4f072d7194842840d725090fd0a7#file-esp32-image-to-flask-ino-L169

Please let me know if you find problems with this approach or if I should use a different approach like AT+UHTTPC=0,4,"/Post""result.txt","postdata.txt",1
So I am after http posts actually.

Best Regards,
MoZen

Paryavi, 2022-07-20:
Thank you, I downloaded the new fixes and it upload the code to S3 chip, but serial monitor of Arduino IDE only shows AT commands and I do not get any print from modem, I use uart zero for serial print and modem connection.

_CELL: initialising with enable power pin not connected, PWR_ON pin 35 (0x23) (and is toggled from 1 to 0) and VInt pin not connected.
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
U_CELL_PWR: powering on.
AT
AT
Unable to add network, error 1070492948!
Network is not available!
1 Tests 1 Failures 0 Ignored
CORRUPT HEAP: Bad head at 0x3fce5824. Expected 0xabba1234 got 0x3fce0014

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace:0x40377696:0x3fce29b00x4037a449:0x3fce29d0 0x4037fe75:0x3fce29f0 0x4037fae7:0x3fce2b20 0x40377965:0x3fce2b40 0x4037fea5:0x3fce2b60 0x42007bd7:0x3fce2b80 0x42008c74:0x3fce2bb0 0x42001f68:0x3fce2bd0 0x420020cd:0x3fce2bf0 0x42001ede:0x3fce2c10 0x42001d6b:0x3fce2c30 0x4200e721:0x3fce2ce0 

ELF file SHA256: 0000000000000000

Rebooting...

RobMeades, 2022-07-20:
Odd; nothing concerning the UARTs etc. has been changed in 1.0.0 aside from including the baud-rate in the configuration structure, which I guess you must be populating if you are running the example.  Have you definitely applied all the changes you require for your particular HW setup into the updated code?

If you are sure everything is configured correctly, can I suggest we have a video call to go through your setup, etc., maybe do some live debugging?  Just that at this stage we usually get into a several weeks-long exchange through this issues mechanism and we could maybe resolve what's going on much more quickly.  Steps would be:

1. Verify that all of the pins (`UART TXD`, `UART RXD`, `UART RTS`, `UART CTS`, `ENABLE_POWER`, `PWR_ON`, `VINT`) are correct for your HW.
2. Verify that the baud rate is being set correctly in the cellular device configuration structure (should be 115200).
3. Verify that the UART HW block is being set correctly for both the cellular device and for your logging output.
4. Connect Saleae probe to the pins listed in (1) to verify, physically, the levels on the pins and the exchange of UART data.

I can be available in an evening UK time to do this.

Paryavi, 2022-07-20:
Aloha Rob,

The odd crash after AT commands is probably because we realized yesterday that with our new S3 design, we used a few GPIOs occupied by and dedicated to PSRAM! so we sent out a new PCB order today to fix that.
Yes, I am sure I configured pins connecting the modem to the microcontroller correctly.
Another thing we have added recently in this design is connecting D+ and D- of the modem to a micro USB port. I am suspicious. That is why I lost the logging output. So after I did not see logs with the FTDI programmer connected to uart0, I removed that and connected a micro-USB connection for Arduino serial monitor, and it was just printing the AT commands.

Verify that the baud rate is being set correctly in the cellular device configuration structure (should be 115200). -Yes, it is 115200, I did not change it in code I mean.

Verify that the UART HW block is being set correctly for both the cellular device and for your logging output.
-Yes, I use UART 0, but I do not set logging output usually, ublox previously was showing the serial prints automatically, but I added Serial.begin(115200) too; at the beginning of the setup loop, still no output. I printed a test sentence below it, it shows that in the serial print terminal.
 
[image]
Thanks for offering to meet, I think since we are going to solder a new board with PSRAM fixes very soon, let's meet after my new tests, I might have fewer board issues that time!

Thanks

Paryavi, 2022-07-20:
Oh in new  sockets Arduino code you have added baudrate setting in this code, where can I set U_CELL_UART_BAUD_RATE? I will look for it too:

.transportType = U_DEVICE_TRANSPORT_TYPE_UART,
    .transportCfg = {
        .cfgUart = {
            .uart = U_CFG_APP_CELL_UART,
            .baudRate = U_CELL_UART_BAUD_RATE,

Paryavi, 2022-07-21:
Hi Rob, 

The problem was as I mentioned before, we connected ublox LTE Power_ON pin to ESP32 S3 GPIO35 which is occupied with PSRAM! so I disabled PSRAM in Arduino IDE setting for my other board(our R422M8S board), and I have ublox serial prints, attached file! So as seen, it correctly sets the hologram Access point that I have set, but then it gets stuck, I am suspicious that it is because I cant set MNO with your code, how can I add MNO to your Arduino code? It is a piece of cake in Sparkfun LTE code;

Please take a look at line 69 in their source code and line 53 of their Arduino code;
The source code for defining MNO as AT&T:
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/src/SparkFun_LTE_Shield_Arduino_Library.h#L69

The arduino code for setting AT&T;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/examples/00_Register_Operator/00_Register_Operator.ino#L53

My Serial Print is attached. I assume R422 gets stuck as I can't set MNO in your Arduino code. I will try to figure it out, but as we are buying 250 R422 boards very soon to assemble and deploy the camera boards, I need to make it work asap. I can upload Sparkfun code to set network to AT&T first then use your code, but it is not an scalable approach so if you add that to your Arduino code that would be great. I was not 100% sure getting stuck is because of not setting MNO, that's why I have attached the modem report...

[R422 Prints.txt](https://github.com/u-blox/ubxlib/files/9162984/R422.Prints.txt)

Thanks,
MoZen

RobMeades, 2022-07-21:
Haven't read your recent posts properly yet but, on setting the MNO profile you just call [uCellCfgSetMnoProfile()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_cfg.h#L196).

Paryavi, 2022-07-21:
So I added the following to your Arduino sockets code and it compiled!
....
// NETWORK configuration for cellular
int32_t mnoProfile = 2;
int32_t uCellCfgSetMnoProfile(uDeviceHandle_t cellHandle,
                              int32_t mnoProfile);
                              
static const uNetworkCfgCell_t gNetworkCfg = {
    .type = U_NETWORK_TYPE_CELL,
    .pApn = "hologram", /* APN: NULL to accept default.  If using a Thingstream SIM enter "tsiot" here */
    .timeoutSeconds = 240 /* Connection timeout in seconds */
};
....

Paryavi, 2022-07-21:
I get this now:

OK
U_CELL_NET: user-specified APN is "hologram".
AT+CGDCONT=1,"IP","hologram"
AT+CGDCONT=1,"IP","hologram"

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?
AT+COPS?

SARA-R422M8S-00B-00

ERROR

+COPS: 0

OK
AT+CFUN=1
AT+CFUN=1

OK
AT+CREG?
AT+CREG?

+CREG: 2,0

OK
0: NReg
AT+CGREG?
AT+CGREG?

+CGREG: 2,4

OK
0: Search

+CREG: 5,"8504","07A19B0F",7

+CEREG: 5,"8504","7A19B0F",7,,,,
7: RegR
7: RegR
AT+COPS=3,0
AT+COPS=3,0

OK
AT+COPS?
AT+COPS?

+COPS: 0,0,"AT&T",7

OK
AT+CGATT?
AT+CGATT?

ERROR

+CGATT: 1

OK

ERROR
AT+CGATT?
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?
AT+CGACT?

ERROR

+CGACT: 1,1

OK
AT+CGACT=1,1
AT+CGACT=1,1

OK
AT+CGACT?
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1
AT+UPSD=0,100,1

OK
U_CELL_NET: connected after 25 second(s).
Looking up server address...
U_CELL_SOCK: looking up IP address of "ubxlib.it-sgn.u-blox.com".
AT+CGDCONT?
AT+CGDCONT?

+CGDCONT: 1,"IP","hologram","100.69.155.137",0,0,0

OK
AT+UDNSRN=0,"ubxlib.it-sgn.u-blox.com"
AT+UDNSRN=0,"ubxlib.it-sgn.u-blox.com"

+UDNSRN: "185.215.195.136"

OK
U_CELL_SOCK: found it at "185.215.195.136".
Address is: IPV4 185.215.195.136
Creating socket...
AT+USOCR=6
AT+USOCR=6

ERROR

+USOCR: 0

OK
AT+USOER
AT+USOER

+USOER: 65

OK
U_SOCK: underlying socket layer could not create socket (errno 105).
Unable to connect to server!
Closing socket...
1 Tests 1 Failures 0 Ignored

RobMeades, 2022-07-22:
Hi MoZen: definitely progress but there's something very weird indeed going on.  Taking a segment of the AT trace above and adding my annotations prefixed by `<<<`:

```
U_CELL_NET: user-specified APN is "hologram".
AT+CGDCONT=1,"IP","hologram"
AT+CGDCONT=1,"IP","hologram"    <<< The out-going AT command is being echoed back to us by the module;
                                <<< I checked the full log and that shows clearly that we have sent ATE0 
                                <<< to switch echo off, as we always do so the only way this can happen
                                <<< is if the modem has reset underneath us (which would cause it to
                                <<< forget that we had sent it ATE0).
OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?
AT+COPS?
SARA-R422M8S-00B-00             <<< This is bizarre!  It is the response to the command ATI, which we haven't sent!
ERROR                           <<< This is also bizarre - an ERROR response might be possible here but we
+COPS: 0                        <<< have an error response followed by the expected response to AT+COPS? (+COPS: 0);
                                <<< this should not be possible.
OK
```

It is almost like there are two modules and/or two hosts talking down the same bit of serial pipe at the same time.

A reason for the module resetting might be to do with the power circuitry; this could be checked by hooking up your Saleae probe to the module's `VINT` output line and having it trigger on a negative edge, which would tell us if the module ever loses power.

One point on the MNO profile thing, having looked at your attached code: in putting the call to `uCellCfgSetMnoProfile()` where you have it is not actually being called, you have effectively added an additional prototype of that function to your C code.  To call a function in C you need to put it inside another function somewhere in the call-tree beneath `main()`, which in the Arduino case would be in the call-tree beneath `setup()` or `loop()`.  It also needs to be somewhere after we have powered the device up, because setting the MNO profile means talking to the module.  For instance, your `setup()` function might become this:

```
void setup() {
    //Serial.begin(115200);
    pinMode(CELL_ENABLE_POWER, OUTPUT);
    digitalWrite(CELL_ENABLE_POWER, LOW);
    delay(5000);
    printf("Hi MoZen! \n");
  
    int32_t returnCode;
    // Initialise the APIs we will need
    uPortInit();
    uDeviceInit();
    
    // Open the device.  Once this function has returned a
    // non-negative value then the transport is powered-up,
    // can be configured etc. but is not yet connected.
    returnCode = uDeviceOpen(&gDeviceCfg, &devHandle);
    if (returnCode >= 0) {
        // Set the MNO profile
        returnCode = uCellCfgSetMnoProfile(devHandle, 2);
        if (returnCode >= 0) {
            printf("Added network with handle %d.\n", devHandle);
            // Bring up the network layer, i.e. connect it so that
            // after this point it may be used to transfer data.
            printf("Bringing up the network...\n");
            errorCode = uNetworkInterfaceUp(devHandle, U_NETWORK_TYPE_CELL,
                                            &gNetworkCfg);
        } else {
            printf("Unable to set MNO profile, error %d!\n", returnCode);
        }
    } else {
        printf("Unable to add network, error %d!\n", returnCode);
    }

    // An errorCode of zero indicates success
}
```

RobMeades, 2022-07-22:
One other thing you might do is to set a greeting message, which the module would send out when it boots, then you may see it visually in the log.  You would do this with a call to [uCellCfgSetGreeting()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_cfg.h#L308), so, putting everything together, your `setup()` code might become:

```
void setup() {
    //Serial.begin(115200);
    pinMode(CELL_ENABLE_POWER, OUTPUT);
    digitalWrite(CELL_ENABLE_POWER, LOW);
    delay(5000);
    printf("Hi MoZen! \n");
  
    int32_t returnCode;
    // Initialise the APIs we will need
    uPortInit();
    uDeviceInit();
    
    // Open the device.  Once this function has returned a
    // non-negative value then the transport is powered-up,
    // can be configured etc. but is not yet connected.
    returnCode = uDeviceOpen(&gDeviceCfg, &devHandle);
    if (returnCode >= 0) {
        // Set a greeting message so that we can see when the module
        // has booted
        uCellCfgSetGreeting(devHandle, "Module has booted");
        // Set the MNO profile
        returnCode = uCellCfgSetMnoProfile(devHandle, 2);
        if (returnCode >= 0) {
            printf("Added network with handle %d.\n", devHandle);
            // Bring up the network layer, i.e. connect it so that
            // after this point it may be used to transfer data.
            printf("Bringing up the network...\n");
            errorCode = uNetworkInterfaceUp(devHandle, U_NETWORK_TYPE_CELL,
                                            &gNetworkCfg);
        } else {
            printf("Unable to set MNO profile, error %d!\n", returnCode);
        }
    } else {
        printf("Unable to add network, error %d!\n", returnCode);
    }

    // An errorCode of zero indicates success
}
```

Paryavi, 2022-07-22:
Thanks, Rob, not sure why I did mistake with function call, I will read a chapter about it this weekend to refresh my C coding! Anyway, the device boots perfectly after uploading the new setup code, but can't find MNO (AT&T) as follows.

BTW, I am curious why Sparkfun code for MNO setting, also adds MNO-keys, also why do they add 2 to the MNO command string length?
Source:
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/src/SparkFun_LTE_Shield_Arduino_Library.cpp#L1658
Also, they use the "+UMNOPROF" command:
const char LTE_SHIELD_COMMAND_MNO[] = "+UMNOPROF"; // MNO (mobile network operator) Profile
Source:
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/src/SparkFun_LTE_Shield_Arduino_Library.cpp#L52

Hi MoZen! //I think I should remove this, haha!
U_CELL: initialising with enable power pin not connected, PWR_ON pin 35 (0x23) (and is toggled from 1 to 0) and VInt pin not connected.

AT
AT
AT
...  //     By ... I mean repeated AT prints!
U_CELL_PWR: powering on.
AT
...
AT

Module has booted
AT
AT
AT

OK
ATE0
ATE0
AT

OK
ATE0

OK

OK

OK
AT+CMEE=2
AT+CMEE=2

OK
AT+UDCONF=1,0
AT+UDCONF=1,0

OK
ATI9
ATI9

00.12,A00.00

OK
AT&C1
AT&C1

OK
AT&D0
AT&D0

OK
AT+UCGED=2
AT+UCGED=2

OK
AT&K0
AT&K0

OK
AT+UPSV=4
AT+UPSV=4

OK
AT+UPSMR=1
AT+UPSMR=1

OK
AT+CPSMS?
AT+CPSMS?

+CPSMS:0,,,"01100000","00000000"

OK
AT+CEDRXS?
AT+CEDRXS?

+CEDRXS:

OK
AT+CFUN=4
AT+CFUN=4

OK
AT+CSGT=1,"Module has booted"
AT+CSGT=1,"Module has booted"

OK
AT+UMNOPROF=2
AT+UMNOPROF=2

+CME ERROR: Operation not supported

U_CELL_CFG: unable to set MNO profile to 2.
Unable to set MNO profile, error -10!
Network is not available!
1 Tests 1 Failures 0 Ignored

SARA-R422M8S-00B-00

NO CARRIER

Paryavi, 2022-07-22:
It looks like in SParkfun code they first check if APN is set correctly;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/examples/00_Register_Operator/00_Register_Operator.ino#L119
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/src/SparkFun_LTE_Shield_Arduino_Library.cpp#L677
 Then scan for operator after setting the APN for up to 3 minutes;
https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/examples/00_Register_Operator/00_Register_Operator.ino#L130

lte.getOperators

https://github.com/sparkfun/SparkFun_LTE_Shield_Arduino_Library/blob/master/src/SparkFun_LTE_Shield_Arduino_Library.cpp#L832
So can we set APN first too?

RobMeades, 2022-07-22:
Not sure I understand: you are already setting the APN, I saw it in your code...?

Paryavi, 2022-07-22:
It is not setting APN after rebooting with this new setup code from these prints, the AT commands are much shorter now, right?

RobMeades, 2022-07-22:
Looking at your most recent AT log, though, that is utterly bizarre, I just can't square it with reality. The module is sent `ATE0`, which switches echo off, and yet you are still getting echoes of the sent AT commands.  Then towards the end `SARA-R422M8S-00B-00` is emitted without any prompting.

On the MNO profile, I've checked the AT manual for SARA-R4 and it says (chapter 7.26.3) the allowed values for SARA-R422M8S are:

`SARA-R422 / SARA-R422M8S / SARA-R422S - 1, 19, 31, 46, 90 (factory-programmed value), 100, 201`

...which is why it is telling you that 2 is not allowed.  I wouldn't worry about this, MNO profile is a kind of "meta setting" thing that tunes the module for some operators but it is by no means a requirement, anything you can set with the MNOPROF at command can be set with individual AT commands if they are really required.

RobMeades, 2022-07-22:
It's not setting the APN because it hasn't go to the bit where the APN is relevant; it would set it if it got there.

Now you're also getting `NO CARRIER` appearing in the log, which is very strange indeed, like something is asking for a circuit switched connection, not something any `ubxlib` code will ever do.  It's not possible that the USB interface of the SARA-R422M8S module is also connected to something, e.g. a PC, is it, which might be trying to access it as a modem?

Paryavi, 2022-07-22:
Sorry the logs you saw was for I connected FTDI programmer Receive pin to ESP32 TX line, now I have connected FTDI receive pin to RX pin of ESP32, so the log is as follows; Btw. the log is seen on Serial terminal of Arduino as FTDI is connected to COM3 of my PC.

Module has booted
AT

OK
AT

OK
ATE0

OK

OK

OK

OK

OK

00.12,A00.00

OK

OK

OK

OK

OK

OK

OK

+CPSMS:0,,,"01100000","00000000"

OK

+CEDRXS:

OK

OK

OK

+CME ERROR: Operation not supported

SARA-R422M8S-00B-00

NO CARRIER

RobMeades, 2022-07-22:
OK, so now we don't have the out-going AT commands at all, which makes it a bit difficult to tell where we are in the process.  However it matches the log above in that the `+CME ERROR: Operation not supported` is being sent by the module because of the attempt to set the MNO profile to 2, which SARA-R422M8S does not support (I'd just take this out, it is only an optimisation, things will work perfectly well without it), then there's the mysterious "NO CARRIER", back from the module, which makes no sense to me because we would never ask for a circuit-switched connection.

RobMeades, 2022-07-22:
To summarise, take out the MNO Profile setting, 'cos the one you want (2) is not supported by SARA-R422M8s, and the code I gave to you won't proceed unless that works, then the last two things in the log of stuff received back from the module:

```
SARA-R422M8S-00B-00

NO CARRIER
```
...don't make a great deal of sense to me: we didn't ask for either of these things, I can't see why the module would ever send them.

Paryavi, 2022-07-22:
Yes we don't have outgoing AT commands because as I said I am now connecting the red wire in the picture(RXI of FTDI cable which goes to COM3 of PC) to RX pin of ESP32, which is RX0 of the board and connected to RXD pin of the module.

Ok I will remove the MNO profile setting, thanks for your time. I have an R410 board too, will test this code on that(if I get serial prints on that board).

RobMeades, 2022-07-22:
Understood: I'm sure you've got this in hand but, obviously, having `printf()`-style trace output from the MCU, which is where the `ubxlib` code will copy the AT exchanges to for tracing purposes, is essential for your development in general, otherwise you're not going to be able to see what is going on to debug anything.

Let me know how it goes: still happy to have a call to help if you think that will make things happen faster.

Paryavi, 2022-07-22:
Thank you, I appreciate it, let's setup a meeting sometime you are available(on Zoom?)
This is my email: mparyavi@hawaii.edu
The thing that make me worried is we populated USB wires from modem(3 pins USB_5V, USB_D-, USB_D+) to another tiny USB output on our board, which I am not using it now(not connected to PC COM port), I am using FTDI programmer as in picture, I am not sure it USB connection to modem is necessary, I did not modify the ublox code to tell it, it is there...

Paryavi, 2022-07-22:
Ok, I did not get prints from R410 modem loading new code(like before that I did not have modem prints), and it was weirdly showing network connections (probably because on R410 I loaded sparkfun code previously)
Ok back to R422 again, I commented setting MNO line and uploaded same code to R422.
//returnCode = uCellCfgSetMnoProfile(devHandle, 2);

And it connects to AT&T after 27 seconds as module report below but it has another error;
underlying socket layer could not create socket (errno 105).
Btw, all these prints are from USB port.

CREG: 2,2

OK

+CREG: 5,"8504","07A19B0F",7

+CEREG: 5,"8504","7A19B0F",7,,,,
0: Search
7: RegR
7: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"AT&T",7

OK
AT+CGATT?

ERROR

+CGATT: 1

OK

ERROR
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

ERROR

+CGACT: 1,1

OK
AT+CGACT=1,1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1

OK
U_CELL_NET: connected after 27 second(s).
Looking up server address...
U_CELL_SOCK: looking up IP address of "ubxlib.it-sgn.u-blox.com".
AT+CGDCONT?

+CGDCONT: 1,"IP","hologram","100.69.155.137",0,0,0

OK
AT+UDNSRN=0,"ubxlib.it-sgn.u-blox.com"

+UDNSRN: "185.215.195.136"

OK
U_CELL_SOCK: found it at "185.215.195.136".
Address is: IPV4 185.215.195.136
Creating socket...
AT+USOCR=6

ERROR

+USOCR: 0

OK
AT+USOER

+USOER: 65

OK
U_SOCK: underlying socket layer could not create socket (errno 105).
Unable to connect to server!
Closing socket...
1 Tests 1 Failures 0 Ignored

Paryavi, 2022-07-22:
I resurrected another R410 board that had defective camera for testing your new MNO code and surprisingly it has serial prints again, it connect to network and sets the MNO but still I got socket issue, I guess in sockets part it is looking at ublox and hologram server at the same time? I will replace ublox with hologram server and see what is the difference, but before that I have attached the Serial report below.
[R410 prints.txt](https://github.com/u-blox/ubxlib/files/9172693/R410.prints.txt)

Paryavi, 2022-07-22:
I changed ublox server and port to the hologram server provided in Sparkfun code, but still error number 105!

+COPS: 0,0,"AT&T Hologram",7

OK
AT+CGATT?
AT+CGATT?

ERROR

+CGATT: 1

OK

ERROR
AT+CGATT?
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?
AT+CGACT?

ERROR

+CGACT: 1,1

OK
AT+CGACT=1,1
AT+CGACT=1,1

ERROR
AT+CGACT?
AT+CGACT?

+CGACT: 1,1

OK
U_CELL_NET: connected after 5 second(s).
Looking up server address...
U_CELL_SOCK: looking up IP address of "cloudsocket.hologram.io".
AT+CGDCONT?
AT+CGDCONT?

+CGDCONT: 1,"IP","hologram","100.69.155.137",0,0,0,0

OK
AT+UDNSRN=0,"cloudsocket.hologram.io"
AT+UDNSRN=0,"cloudsocket.hologram.io"

+UDNSRN: "3.210.32.221", "34.233.146.107"

OK
U_CELL_SOCK: found it at "3.210.32.221".
Address is: IPV4 3.210.32.221
Creating socket...
AT+USOCR=6
AT+USOCR=6

ERROR
AT+USOER
AT+USOER

+USOCR: 0

OK
U_SOCK: underlying socket layer could not create socket (errno 105).
Unable to connect to server!
Closing socket...
1 Tests 1 Failures 0 Ignored

+USOER: 65

Paryavi, 2022-07-23:
So we cut the tracks to IO 35 and 36 on ESP32 that were connecting to PSRAM wrongfully, which I am suspicious to cause error 105, buffer error, now Power_ON is connected to GPIO 3 of R410 and R422 boards, and I have enabled PSRAM now, but still I get error 105, attached.
R410 report, btw your MNO code works as attached.
[R410 Report.txt](https://github.com/u-blox/ubxlib/files/9172985/R410.Report.txt)
R422 board report:
[R422 report.txt](https://github.com/u-blox/ubxlib/files/9172988/R422.report.txt)

RobMeades, 2022-07-23:
One thing to make sure of is that the module's USB port must NOT be connected to anything.  If the module detects that its USB port is connected to something it will only respond to AT commands over its USB port and will ignore AT commands coming over the UART from your MCU.

The other thing is to make sure of that you set the correct module type when switching between modules: you must use `U_CELL_MODULE_TYPE_SARA_R422` for SARA-R422 and `U_CELL_MODULE_TYPE_SARA_R410M_02B` for SARA-R410M-02B; the way the cellular connection is made available to the internal IP stack on the module is different for the two cases so SARA-R422 will not set up an IP connection correctly if you tell the `ubxlib` code that you've got a SARA-R410 module and vice-versa.  Looking at your SARA-R422 AT log above there are a couple of things I would expect to see (`AT+UPSD` commands) that do not appear so it might well be that the module type was not set correctly, at least on that run.

Paryavi, 2022-07-25:
Thanks Rob, I was commenting out module type in Arduino, now I created a separate Arduino file for R422, attached.
Now it has AT+UPSD command, but still I get error #105.
Also I commented setting MNO code like this, not sure if it works properly as I commented returnCode and I still check if statement for R422 board!

....
returnCode = uDeviceOpen(&gDeviceCfg, &devHandle);
    if (returnCode >= 0) {
        // Set a greeting message so that we can see when the module
        // has booted; not bothering to check the return code here
        // as this is non-critical
        uCellCfgSetGreeting(devHandle, "Module has booted");
        // Set the MNO profile
        //returnCode = uCellCfgSetMnoProfile(devHandle, 2);
        if (returnCode >= 0) {
            printf("Added network with handle %d.\n", devHandle);
....
[R422 with rerouted pins.txt](https://github.com/u-blox/ubxlib/files/9184980/R422.with.rerouted.pins.txt)
Arduino code;
[Arduni code for R422.txt](https://github.com/u-blox/ubxlib/files/9184982/Arduni.code.for.R422.txt)

philwareublox, 2022-07-26:
> *** This is an EXTERNAL email. It was sent from outside of u-blox. ***

The main issue in this AT log is that we are seeing other AT responses within this AT sequence. For example:

143 AT+COPS?
144
145 SARA-R422M8S-00B-00
146
147 ERROR
148
149 +COPS: 0
150
151 OK

We should not be getting SARA-R422M8S-00B-00 and ERROR in between the AT+COPS? And +COPS: 0 response.
We believe there is another device talking to this R422 module, which is upsetting your AT sequence/application.

What UBXLIB is requiring is an AT Log like this:

143 AT+COPS?
144
145 +COPS: 0
146
147 OK

(Without the SARA-R422M8S-00B-00 and ERROR - These are NOT part of the +COPS? query)

We can see this also happens when UBXLIBX creates a socket, and why it is failing:

680 AT+USOCR=6

681

682 ERROR

683

684 +USOCR: 0

685

686 OK

The +USOCR command was successful as we got back a +USOCR: 0 response (SocketID: 0) - but because ERROR was emitted UBXLIB saw this and thought the Create Socket command failed.

We need to figure out why/where the module is being controlled from elsewhere, as this is disturbing the AT sequence.
Is this device connected to a Laptop via the USB port of the module?
It looks like Windows is seeing this Module as a regular MODEM and is trying to send MODEM configuration AT commands to it. I.e. to check the name of the MODEM (SARA-R4222M8S-00B-00) and in an earlier AT log you shared "NO CARRIER".

Without understanding how this device is being sent other AT commands I don't know how we can move forward. UBXLIB is logging the responses to these AT commands, but unfortunately UBXLIB is not logging the sending of these strange hidden AT commands - so we suspect them coming from some other controlling app/Windows/device...

Phil.

Paryavi, 2022-07-26:
Thanks, Phil, 

Now I just use two pins of FTDI progrmmaer(GND and RXI, and NO TX from Windows PC) connected to COM3 Windows (Arduino IDE serial print below) to GND and TX of MCU on the board. In our design, we use UART0 of MCU to connect to R422 modem also the board spits out its ubxlib code serial prints on same UART 0! Can I disable serial prints of MCU? Can we change the serial prints baud rate?

[image]

Serial Prints with these two FTDI wires now:
AT+CREG?
AT+CREG?

+CREG: 2,2

OK
0: Search
AT+CGREG?
AT+CGREG?

+CGREG: 2,4

OK
0: OoC
AT+CEREG?
AT+CEREG?

+CEREG: 4,2

OK
0: Search

+CREG: 5,"8504","07A19B0F",7

+CEREG: 5,"8504","7A19B0F",7,,,,
7: RegR
7: RegR
AT+COPS=3,0
AT+COPS=3,0

OK
AT+COPS?
AT+COPS?

+COPS: 0,0,"AT&T",7

OK
AT+CGATT?
AT+CGATT?

ERROR

+CGATT: 1

OK

ERROR
AT+CGATT?
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?
AT+CGACT?

ERROR

+CGACT: 1,1

OK
AT+CGACT=1,1
AT+CGACT=1,1

OK
AT+CGACT?
AT+CGACT?

+CGACT: 1,1

OK
AT+UPSD=0,0,0
AT+UPSD=0,0,0

OK
AT+UPSD=0,100,1
AT+UPSD=0,100,1

OK
U_CELL_NET: connected after 26 second(s).
Looking up server address...
U_CELL_SOCK: looking up IP address of "cloudsocket.hologram.io".
AT+CGDCONT?
AT+CGDCONT?

+CGDCONT: 1,"IP","hologram","100.69.155.137",0,0,0

OK
AT+UDNSRN=0,"cloudsocket.hologram.io"
AT+UDNSRN=0,"cloudsocket.hologram.io"

+UDNSRN: "34.233.146.107", "3.210.32.221"

OK
U_CELL_SOCK: found it at "34.233.146.107".
Address is: IPV4 34.233.146.107
Creating socket...
AT+USOCR=6
AT+USOCR=6

ERROR
AT+USOER
AT+USOER

SARA-R422M8S-00B-00

ERROR

+USOCR: 0

OK
U_SOCK: underlying socket layer could not create socket (errno 105).
Unable to connect to server!
Closing socket...
1 Tests 1 Failures 0 Ignored

+USOER: 65

OK

RobMeades, 2022-07-26:
Oh no!  I'd not realised the debug trace was ending up on the same UART output line as the AT commands!  That explains where the `SARA-R422M8S-00B-00` comes from: the string "creating socket..." contains the characters "ATI" which would elicit the response `SARA-R422M8S-00B-00` from the module.  The module will be getting VERY confused indeed - a modem receiving an AT command while processing an existing AT command, in many cases, aborts the existing AT command (that's a well define V.250 spec thing).

While you could disable trace output from `ubxlib` (just pass `U_CFG_ENABLE_LOGGING=0` into your build), that's not gonna help you 'cos you will _need_ trace output from your device for your development, it is somewhat of a fundamental requirement.

Let's discuss when we have our call.

Paryavi, 2022-07-27:
Thank you, I modified u_cfg_sw.h file instead to mute the trace output with # define U_CFG_ENABLE_LOGGING                 0
and the socket problem was resolved, thanks Rob and Phil.
[image]
It was kinda long AT+CREG? AT+CGREG? request from MCU, can we make modem connect sooner?

RobMeades, 2022-07-27:
How long a cellular module takes to register with the network will depend on many things:

- was the the first time (first time always takes longer as the module has no knowledge of the network environment),
- is this a roaming SIM (roaming SIMs always take longer to search),
- what RAT is being used (either cat-M1 or NB in the case of SARA-R410/SARA_R422; having NB in the list and having it first always takes longer since it is required to search well below the noise floor before giving up),
- is the device only going to be used in a single area/operator (in which case you may be able to limit the radio bands/RAT to just what you need),
- and of course, network coverage.

Unless the registration time continues to be prohibitively long, I would suggest just letting the module to its thing for now, optimize later when you know what it is best to optimize first.

Paryavi, 2022-09-16:
Aloha @RobMeades and @philwareublox 
Howzit?
Today I went outdoors to check R422M8S GNSS, but it does not report location so far, I tried two versions of my codes(added to your sockets.ino code!)
Rob says I need two things to make it work; 

All you need to do is to say there's an M8 GNSS chip present(I did on my line 93: #define U_CFG_TEST_GNSS_MODULE_TYPE U_GNSS_MODULE_TYPE_M8) and set the transport type to U_GNSS_TRANSPORT_AT as described [in this example](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_cell.c)
I am not sure of the second one(U_GNSS_TRANSPORT_AT).
Rob: https://github.com/u-blox/ubxlib/issues/41#issuecomment-1099615988

All relevant codes:void Setup ...
    uLocation_t location;      
    location.timeUtc = -1;   // Set an out of range value so that we can test it later
    uLocationAssist_t locationAssist;
    locationAssist.disableGnss =true; //If you wish to use the GNSS chip directly from this code then set this to true. 

    //============================================================
    // Get location
    printf("Loc:\n");
    //if (uLocationGet(devHandle, U_LOCATION_TYPE_GNSS,
                     //&locationAssist, NULL, &location, NULL) == 0) {

    if (uLocationGet(devHandle, U_LOCATION_TYPE_GNSS,
                     NULL, NULL, &location, NULL) == 0) {

     printf("I am here: https://maps.google.com/?q=%3.7f,%3.7f\n",
                    ((double) location.latitudeX1e7) / 10000000,
                    ((double) location.longitudeX1e7) / 10000000);
        }
      //=====================================================================

Paryavi, 2022-09-16:
Should I modify the following(**U_CFG_APP_CELL_PIN_GNSS_POWER**) in my configuration file, I am using R422M8S, what should I set it too?

#ifndef U_CFG_APP_CELL_PIN_GNSS_POWER
/** Only relevant when a GNSS chip is connected via a cellular module:
 * this is the cellular module pin (i.e. not the pin of this MCU,
 * the pin of the cellular module which this MCU is using) which controls
 * power to GNSS. This is the cellular module pin number NOT the cellular
 * module GPIO number.  Use -1 if there is no such connection.
 */
# define U_CFG_APP_CELL_PIN_GNSS_POWER  -1 // AKA GPIO2
#endif

My config file:
[u_cfg_app_platform_specific.txt](https://github.com/u-blox/ubxlib/files/9581394/u_cfg_app_platform_specific.txt)

closed 2023-07-23 by Paryavi