u-blox/ubxlib #43: Problem with SARA-R500S-00B which now is critical

nimams, 2022-04-03:
Hi,

I'm using Zephyr with porting layer of ubxlib and all the HALs and Drivers.
Currently, I'm working on SARA-R500S-00B bringup.
I had a success and everything was working normal except the shutdown procedure. So basically the cell shutdown method was never returning. Once I did more investigation, I figured that the modem never send a message to ACK the request. Now the problem has become worse. Right after we finish initialization and try to see if the modem is alive, as soon as we send the first command, we start to get response of 07E and it never stops. This is very likely to not be ublox related, but I was wondering if a team had any similar experiences.
Our HW platform currently is Ublox.
Thank you

RobMeades, 2022-04-03:
Hi there, and sorry to hear that you're having problems.  Can you explain in more detail what you mean by "the modem never send a message to ACK the [shutdown] request"?  Do you mean that there was no `OK` to the `AT+CPWROFF` command, or do you mean that your MCU has `VINT` wired to it from the module and the MCU never saw the `VINT` line go low?  As you will see from our integration manual, we always recommend monitoring `VINT` to detect that the module has fully and properly powered off. 

As to your situation with the `07E`, I will double check what that means but I _believe_ it means that the module's attempt to boot itself has failed because the flash has a CRC error.  If this is really the case you will need to contact your supplier to ask for a replacement.  But it is probably more important to establish why this might have occurred, for which I will need to find the right application engineering channel for your company: if you could send me a direct e-mail message with your details I'll see if I can do that.

RobMeades, 2022-10-05:
Having a bit of a clean-out of the issues list - gonna close this one, please re-open if there is still an issue.

closed 2022-10-05 by RobMeades