u-blox/ubxlib #44: how to use ntp service

alexmaron81, 2022-04-05:
Does anyone have an example of how I can reach an NTP service via socket?

RobMeades, 2022-04-05:
Hi Alex: we used to use obtaining time from an NTP service as an example in our [old-as-the-woods C027 example code](https://os.mbed.com/teams/ublox/code/example-ublox-cellular-interface//file/2962a8eaacb7/main.cpp/).  Basically it just did a UDP `sendTo()` to the URL `2.pool.ntp.org` of a buffer exactly 48 bytes long, in which the first character was `0x1b` and the remaining bytes were `memset()` to 0.  The NTP server would then respond with a response of the same length that we would decode with this function:

```
static void printNtpTime(char *buf, int len)
{
    time_t timestamp = 0;
    struct tm *localTime;
    char timeString[25];
    time_t TIME1970 = 2208988800U;
 
    if (len >= 43) {
        timestamp |= ((int) *(buf + 40)) << 24;
        timestamp |= ((int) *(buf + 41)) << 16;
        timestamp |= ((int) *(buf + 42)) << 8;
        timestamp |= ((int) *(buf + 43));
        timestamp -= TIME1970;
        localTime = localtime(&timestamp);
        if (localTime) {
            if (strftime(timeString, sizeof(timeString), "%a %b %d %H:%M:%S %Y", localTime) > 0) {
                printf("NTP timestamp is %s.\n", timeString);
            }
        }
    }
}
```

Now I can't say that this still works 'cos I haven't used this code since 2017 but I guess it might still work...?

alexmaron81, 2022-04-06:
Hi Rob, I have the following output...
[image]

RobMeades, 2022-04-06:
Obviously not that one anymore then, it was 5 years ago and these things change, try these:

https://www.galsys.co.uk/time-server/uk-time-server.html

alexmaron81, 2022-04-06:
Still getting "Unable to connect".
As far as I understand it correctly, I just have to run this example with UDP on the address "0.uk.pool.ntp.org"?
https://github.com/u-blox/ubxlib/blob/master/example/sockets/main.c

RobMeades, 2022-04-06:
Remember that the time server is UDP and not TCP.  Find [attached](https://github.com/u-blox/ubxlib/files/8427382/main.c.txt) that example source file modified to read the time from the NTP server (please rename the attachment from `main.c.txt` to `main.c` since Github does not allow file attachments with a `.c` extension) and below the corresponding log output:

```
Looking up server address...
U_CELL_SOCK: looking up IP address of "0.uk.pool.ntp.org".
AT+CGDCONT?
+CGDCONT: 1,"IP","internet.mnc001.mcc001.gprs","192.168.3.14",0,0,0,2,0,0,0,0,0,0,0
OK
AT+UDNSRN=0,"0.uk.pool.ntp.org"
+UDNSRN: "217.114.59.3"
OK
U_CELL_SOCK: found it at "217.114.59.3".
Address is: IPV4 217.114.59.3
Creating socket...
AT+USOCR=17
+USOCR: 0
OK
U_SOCK: socket created, descriptor 0, network handle 257, socket handle 0.
Sending data...
AT+USOST=0,"217.114.59.3",123,48
@[1b][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00]
+USOST: 0,48
OK
Sent 48 byte(s) to echo server.
AT+USORF=0,0
+UUSORD: 0,48
+USORF: 0,48
OK
AT+USORF=0,1024
+USORF: 0,"217.114.59.3",123,48,"[1c][02][03][ec][00][00][01]@[00][00][0c] [9d],[b0][04][e5][f8][10]e[d9][b0][b1]r[00][00][00][00][00][00][00][00][e5][f8][17][f3]b[bc][a2]a[e5][f8][17][f3]b[bd][ea][15]"
OK
Received echo back (48 byte(s).
UTC timestamp from NTP server is 1649252723.
```

alexmaron81, 2022-04-07:
Thank you Rob! It works very well.

RobMeades, 2023-11-30:
Hi Alex.  I suspect it is something to do with this:

https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/migration-guides/release-5.x/gcc.html#espressif-toolchain-changes

...i.e. in their release 5 Espressif has changed `int32_t` from being `typedef`ed to `int` to being `typedef`ed to `long` and that affects the way that `printf()` formatters work and might also affect how the things inside `struct timeval` are typed.

So it might be that `%ll`, which was previously telling the logging function that the thing coming at it was a 64-bit number now ends up telling it that it is a 128-bit number.  Maybe try `%ld` instead?

alexmaron81, 2023-12-07:
OK, now I have it.
I replaced "time_t" with "int32_t".

Unix Time 2038 Overflow
(https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html#unix-time-2038-overflow)
Unix time (type time_t) was previously represented as a 32-bit signed integer, leading to an overflow in year 2038 (i.e., [Y2K38 issue](https://en.wikipedia.org/wiki/Year_2038_problem)). To address the Y2K38 issue, ESP-IDF uses a 64-bit signed integer to represent time_t starting from release v5.0, thus deferring time_t overflow for another 292 billion years.

closed 2023-12-07 by alexmaron81

RobMeades, 2023-12-07:
>  thus deferring time_t overflow for another 292 billion years.

:D

cturvey, 2023-12-07:
Should migrate to NTPv4 with 64-bit integer and fractional parts..

```
if (len >= 43) <<< should be 44, you're reaching into element 43 of the zero based array
{
    timestamp |= ((int32_t) * (pBuf + 40)) << 24;
    timestamp |= ((int32_t) * (pBuf + 41)) << 16;
    timestamp |= ((int32_t) * (pBuf + 42)) << 8;
    timestamp |= ((int32_t) * (pBuf + 43)); ***
    timestamp -= TIME1970;
```