u-blox/ubxlib #45: Cell Locate response with zero

alexmaron81, 2022-04-06:
Hi, I have the following output when using the example?
[image]

RobMeades, 2022-04-06:
It looks like the Cell Locate service was unable to establish your position, I don't know whether the Cell Locate service knows anything about the physical location of the cells on the 1nce network.  I will ask...

RobMeades, 2022-04-06:
I'm told that `262 01` is T-mobile in Germany, so Cell Locate should know about that.  Can you confirm that you have obtained your authentication token for the Cell Locate service (from the Thingstream portal) and copied that [into the example](https://github.com/u-blox/ubxlib/tree/master/example/location#the-cell-locate-example-main_loc_cell_locatec)?

alexmaron81, 2022-04-06:
yes i tried it with
#define U_CFG_APP_CELL_LOC_AUTHENTICATION_TOKEN dylsx0cAQsS3n7cJNuexxx

RobMeades, 2022-04-06:
Not sure what to suggest: Cell Locate appears to have failed even before it did a scan, so it seems likely that there is something wrong with the connectivity to the Cell Locate server.  There are two things that might go wrong as far as I can tell, one is the authentication token being incorrect somehow, the other is if your cellular network/APN does not, for some reason, allow the module to create its internal TCP connection to the cell locate servers (`celllive1.services.u-blox.com` and `celllive2.services.u-blox.com`).  If you like you can send me your authentication token in a direct e-mail and I'll try it out here?

RobMeades, 2022-10-05:
Having a bit of a clean-out of the issues list - gonna close this one, please re-open if there is still an issue.

closed 2022-10-05 by RobMeades