u-blox/ubxlib #46: Taking down network... unable to disconnect

alexmaron81, 2022-04-26:
Somehow the power consumption is too high after trying to break down the connection.
[image]

RobMeades, 2022-04-26:
Hi Alex: the AT trace segment pasted in above shows that no response whatsoever from the module to AT commands - not in any particular circumstance it would seem, in that there is no response to `AT+COPS?`, though of course I don't know how long was spent waiting for one.

You say "the power consumption is too high after trying to break down the connection"; what leads you to that conclusion?  Is there more AT trace, or other data, which leads you to that conclusion?

alexmaron81, 2022-04-26:
That's all that happens after "taking down network". After that, the task expires. If the modem were to shut down successfully, the power consumption should drop.

RobMeades, 2022-04-26:
Understood, but it is not possible to understand what is happening from the AT trace above.  It seems as though the module is unresponsive to AT commands, so something has gone wrong.  Can you paste in what happened before the sequence above, how it got into that state?  Do you have the `PWR_ON` line connected between the module and the MCU, i.e. do you expect the MCU to have used that to power off the module and, if so, did it do so?

Basically, what more can you tell me about the problem you are having, 'cos I am not able to conclude much based on that segment of AT trace.

alexmaron81, 2022-04-26:
Oh, you just gave me a good idea.
Sometimes it happens that the MQTT connection could not be established. After the connection is broken, an attempt is made to switch off the modem. And only then does something go wrong

Yes, unfortunately, as is always the case, I don't have the problem right now. now everything works fine
I still don't know why the connection sometimes can't be established?

[image]
[image]

RobMeades, 2022-04-26:
Ah, that's useful: there is an issue in some cases where the MQTT entity inside the module doesn't get informed that the data connection is available until slightly _after_ the AT interface (and hence the attached MCU) has been told.  Hence it is possible to get an immediate failure from the MQTT entity if the MCU is a bit too quick in asking for an MQTT connection after the cellular connection has been established.

We have fixed this in the most recent `ubxlib` code by making sure there is a delay between the MCU completing the commands for the data connection and asking for an MQTT connection, however that code has not been pushed to this repo yet.  In case that is your issue, you could try adding a short delay (say 500 ms, to be safe) between establishing the cellular connection and calling `uMqttClientConnect()`.

But that doesn't explain why the module would ever become unresponsive to AT commands of course.  Should you see that again, please post as much of the AT trace as you can here.

alexmaron81, 2022-04-28:
thanks Rob It worked great. Hadn't had any problems until now.
two things done at once.

closed 2022-04-28 by alexmaron81

RobMeades, 2022-04-28:
Great, thanks for confirming Alex.  Our version of the fix for this problem is now pushed here in commit 9d14a2080d170c4fad1e7d01f0d46c389029ec3c.