u-blox/ubxlib #47: Support UART Break Condition

philwareublox, 2022-04-27:
Many MCU UARTs can detect a 'break condition' where the Rx line is held low for longer than a frame period. https://en.wikipedia.org/wiki/Universal_asynchronous_receiver-transmitter#Break_condition

Our cellular modules will perform this break condition when shutting down. 

I believe ESP32, nRF52 and STM32 MCUs support a break interrupt. We should investigate how we could use this break condition to detect PSM and power down states, for when V_INT monitoring is not possible, or +UPSMR: 1 URC is not implemented.

RobMeades, 2022-04-27:
Excellent idea: will raise an internal JIRA to track this also.

manemannen, 2024-11-11:
Project is discontinued

closed 2024-11-11 by manemannen