u-blox/ubxlib #48: wrong ICCID

alexmaron81, 2022-05-05:
Hi, everyone,

I am trying to query the ICCID with the "uCellInfoGetIccidStr" function. But I get one character at the end of the number too many. what could be the problem?
bytesRead = uCellInfoGetIccidStr(networkHandle, ICCID_buffer, sizeof(ICCID_buffer));

here is the representation of the portal of my provider
[image]

and here what the modem outputs
[image]

RobMeades, 2022-05-05:
Hi Alex: looks like you may have a case of this:

https://www.netmorem2m.com/iot-wiki/iccid-explained/#:~:text=NN%20C(X)-,An%20ICCID%20typically%20consists%20of%2019%20to%2020%20characters.,13%20digits%20will%20be%20visible.

alexmaron81, 2022-05-05:
only the 19 characters are on the card.
OK that explains everything.
Will cut off the last character for me ;-)

closed 2022-05-05 by alexmaron81