u-blox/ubxlib #49: multiple transmission of a mqtt message with only one sent

alexmaron81, 2022-05-16:
I have the problem that when I publish a message, four always arrive?
the next time i send, there are already 8 in the queue
Rabbitmq broker (MQTT)

RobMeades, 2022-05-16:
Hi Alex - what are you setting the [QoS](https://github.com/u-blox/ubxlib/blob/master/common/mqtt_client/api/u_mqtt_common.h#L43) to?  If you want to guarantee that only one will arrive you should use `U_MQTT_QOS_EXACTLY_ONCE`.

alexmaron81, 2022-05-16:
[image]

RobMeades, 2022-05-16:
That would seem correct: could you possibly paste into here an AT log of the exact sequence, which I could use to show your problem to our application engineering guys?

alexmaron81, 2022-05-16:
[image]

RobMeades, 2022-05-16:
Great, thanks, let me see what they say...

RobMeades, 2022-05-16:
Right, have had a discussion with AE, the AT sequence is fine so we need visibility from another angle.  A few options:

1. Are you able to get at the USB port of the module?  Specifically, would it be possible to attach a Windows PC to that USB port _only after powering the module up_?  If so, you could obtain a log from it (I will tell you how).  It needs to only be connected _after_ powering the module up as, if the module detects something plugged into the USB port at boot it will switch to only accepting AT commands over the USB port and NOT over the UART port, which would prevent your MCU talking to it.
2. Are you able to take a log (e.g. Wireshark, or just an MQTT message log) on the broker side at all?
3. Would you be willing to share log-in details for the broker with us, in which case we could reproduce the problem here?

alexmaron81, 2022-05-16:
I would like to try the first variant.
I just have to figure out how to connect the USB port. Is it a problem if the VCC pin gets the V_SYS voltage and then the RSVD/VUSB_DET is powered by the USB? The circuit board was developed for us, so I don't have much of a clue.
[image]

RobMeades, 2022-10-05:
This was moved to direct comms, closing - please re-open if there are still concerns.

closed 2022-10-05 by RobMeades