u-blox/ubxlib #51: Unable to bring up the network!

alexmaron81, 2022-06-14:
Hi,

sometimes the connection is not successfully established, so that the module no longer works.

Unfortunately I don't use the following options:
U_CFG_APP_PIN_CELL_ENABLE_POWER
U_CFG_APP_PIN_CELL_RESET
U_CFG_APP_PIN_CELL_VINT

What other option is there to reset the module via software?

RobMeades, 2022-06-14:
Hi Alex: it would depend on the nature of the "no longer works".

If the module is responsive to AT commands then you can call `uCellPwrReboot()` and `ubxlib` will tell the module to reboot via the AT interface.

However, if the module is not responding to AT commands, the only way out that I am aware of would be a physical connection to either the `PWR_ON` or `RESET_N` pins of the module from one of your MCU pins; this could be done using any GPIO that you control of course.  You should check the [SARA-R4 System Integration Manual](https://www.u-blox.com/en/docs/UBX-16029218) for details of exactly what you need to do with those pins to effect a reset.

alexmaron81, 2022-06-14:
ok nice, thanks for the info!

alexmaron81, 2022-06-20:
Hi Rob, 

to use uCellPwrReboot() do I need u_cell_pwr.h? but then comes the following

[image]

RobMeades, 2022-06-22:
Hi Alex: yes, you'll need to #include `u_cell_net.h` just before `u_cell_pwr.h`.  We've removed this problem in the 1.0.0 release that we are about to make by just dumping all required headers into one `ubxlib.h` that you can include.

alexmaron81, 2022-08-09:
Hi Rob, I run the uCellPwrReboot(devHandle, NULL); , but the device still does not turn on

[image]

RobMeades, 2022-10-05:
Can't remember whether we resolved this one or not now, maybe in direct comms?  Gonna close it - please re-open if there is still an issue.

closed 2022-10-05 by RobMeades