u-blox/ubxlib #57: Unable to connect to MQTT broker

alexmaron81, 2022-07-21:
Hi Rob, sometimes the problem is that the connection to the MQTT broker cannot be established, so that it is then not possible to log off from the network.

There is no reaction to a new connection to the module.

[Unable to connect to MQTT broker.txt](https://github.com/u-blox/ubxlib/files/9157682/Unable.to.connect.to.MQTT.broker.txt)

RobMeades, 2022-07-21:
Hmmm, so could the problem somehow be related to MQTT in the module?  If you have time it might be worth trying the experiment of #56 with the MQTT parts omitted, just to see if the disconnect works then.  But first it would be good to get an AT log of #56 with the additional changes in the file I attached there, and with MQTT parts still **included**.

RobMeades, 2022-07-21:
Right, I have pushed a branch here:

https://github.com/u-blox/ubxlib/tree/fix_sara_r412m_disconnect_rmea

...with what I believe to be a fix in it.  I have tested it here and it has no adverse effect but we can't reproduce your problem here so would you be able to give it a try for us?  You can either pull the branch or you can just copy the `u_cell_net.c` file from that into your current code tree and try it out.  If you can, please paste the AT output into here.

alexmaron81, 2022-07-22:
Hi Rob, it works very well. The connection is immediately disconnected.

[disconnected.txt](https://github.com/u-blox/ubxlib/files/9164939/disconnected.txt)