u-blox/ubxlib #58: Hex-Mode attempts oversized socket writes

johannesmay, 2022-07-22:
Hi again,

I found an issue with the cellular socket hex mode implementation.
If we call `uCellSockWrite` with more than 512 bytes of data, invalid write commands are triggered.

For example, we found a scenario where we would attempt a write like this: `AT+USOWR=5,1017,"30...` (I removed the remaining data for security reasons).
In this case, the modem always responds `+CME ERROR: 3`, resulting in an error return code and our application closing the connection.

I think there are some basic mechanics in place to avoid this.
The `thisSendSize` variable is correctly divided in the hex mode case (see [u_cell_sock.c](https://github.com/u-blox/ubxlib/blob/master/cell/src/u_cell_sock.c#L1508)), but this value is never used afterwards.
I guess the while loop should trigger `AT+USOWR` commands with the `thisSendSize` variable as the limit?
Does this also apply to the non-hex mode sending?

RobMeades, 2022-07-22:
Looks like we need to add more range checking and testing of range checking; I'd kind of avoided this in the past on the grounds that, should a module support larger AT command lengths, they might "just work" but this is also an easy way out on my part :-).

I will add range checking to the TODO list and run your PR through the existing tests.

johannesmay, 2022-07-22:
Yeah I think at least one test with larger packet sizes would be good to add.
I hope the fix works, but I think with this change the `thisSendSize` is actually used as originally intended.

Thanks for running the tests!

closed 2022-07-25 by johannesmay