u-blox/ubxlib #61: How do I get a correct DBM value?

alexmaron81, 2022-08-08:
I need the correct dbm value.

is it the right source?
https://github.com/u-blox/ubxlib/blob/75ab57e92bcf485bd16b8556c3b11a5f9c462bfe/cell/test/u_cell_info_test.c

RobMeades, 2022-08-08:
Hi Alex: correct dBm value of what thing?  You have a choice of [uCellInfoGetRssiDbm()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_info.h#L81) or [uCellInfoGetRsrpDbm()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_info.h#L93).  I believe you are on NB1 so you _probably_ want RSRP.  Don't forget to call [uCellInfoRefreshRadioParameters()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_info.h#L69) first!

alexmaron81, 2022-08-08:
what's the difference between Rssi and Rsrp?

RobMeades, 2022-08-08:
RSSI is the total received strength in the received bandwidth, RSRP is the total received strength of the wanted signal in a received bandwidth.  So RSSI will include things other than the specific serving signal.

This article seems quite informative: https://www.cablefree.net/wirelesstechnology/4glte/rsrp-rsrq-measurement-lte.

alexmaron81, 2022-08-08:
but before that I have to build a connection "uNetworkInterfaceUp"?

alexmaron81, 2022-08-08:
ok that works already

[image]

Thanks for the help Rob!

RobMeades, 2022-08-08:
Excellent!

RobMeades, 2022-08-08:
...though it looks like you have a lot of interferers there (with RSSI being considerably higher than RSRP), and RSRP is pretty close to the lowest limit.  Good to know it though.

alexmaron81, 2023-01-11:
Hi Rob,

Since I've been using the new version of the library, the RSSI value is 0.
Do you know what it could have to do with?
[image]

alexmaron81, 2023-01-26:
hello, sometimes the rssi value is zero.
What can be the reason?
[image]

RobMeades, 2023-01-26:
Hi Alex: RSSI is zero because it is derived from the CSQ value and that sometimes just takes a while to update.  I notice that this is shortly after connection, you might just have to wait for a little while.

Frankly, though, RSSI is pretty meaningless for LTE: OK, there's a measured signal in the band but who knows what it is, it could be anything.  The thing that matters is RSRP: the level of the _wanted_ signal, and RSRQ: the level of the wanted signal relative to the noise.  You can see above that the level of the wanted signal is quite low but probably bearable at -112 dBm and RSRQ is -10 dB, so the module is actually pulling the wanted signal out of the noise.

RobMeades, 2023-03-02:
Closing this one: please re-open if necessary.

closed 2023-03-02 by RobMeades

alexmaron81, 2024-09-02:
how can i find out, before a connection has been established, whether it will be possible to establish a connection at all?

My application is battery operated. So far it works quite well when the device is in good network coverage.
Some devices are not in a good position and try to establish a connection too often, so that the battery runs out quickly.

However, I can only query RSRQ once the connection has been established.

Is there a way to check the network quality beforehand?

philwareublox, 2024-09-02:
Hi, 

Unfortunately LTE does not work like that and to get the 3GPP RSRQ/RSRP values you must be connected. 

Phil.