u-blox/ubxlib #63: ubxlib support for Assist Now On-line, off-line and auto

owain-incus, 2022-08-15:
Is support for the various Assist now u-blox services planned. Sara R4 / R5 support these; but I don't think ubxlib does (yet).
If I want to use these services now must I just rely on the AT client functionality of the library and implement my own location service based on the Sara R4/R5 location services app note?

RobMeades, 2022-08-16:
Hi Owain - I have to admit that the range of these services confuses me also, gimme a moment to consult...

bbyers-UBCO, 2022-12-19:
@RobMeades I'm also curious about this, any update here?

RobMeades, 2022-12-20:
Apologies, entirely forgot about this, have assigned it to myself and made a note to actually do something, though likely not until January now as I will need to find the right person internally.

RobMeades, 2023-08-09:
Configuration of AssistNow is now supported, see https://github.com/u-blox/ubxlib/releases/tag/v1.3.0.

Closing this issue: if you have problems, please open another.

closed 2023-08-09 by RobMeades