u-blox/ubxlib #65: WiFi access point support

kozakPCS, 2022-08-16:
Are there plans to add support for configuring short range radios as WiFi access points? Thanks!

RobMeades, 2022-08-16:
Hi there, and a very good question: @manemannen, one for you I think.

manemannen, 2022-08-18:
Currently no but I agree it would be a nice feature. I'll add an internal JIRA ticket to be discussed in the team.

hkro, 2023-02-17:
Check out #86 - it's in the making 🏗️

RobMeades, 2023-08-09:
This is now supported, in the sense of using the Wifi access point as a captive portal, see https://github.com/u-blox/ubxlib/releases/tag/v1.3.0 and the example here: https://github.com/u-blox/ubxlib_examples_xplr_iot/tree/master/examples/captive_portal (though you will need to update the `ubxlib` submodule of that repo to point to the latest `ubxlib` to bring in the functionality).

Is that sufficient for your purposes?

RobMeades, 2024-01-04:
Going to close this now: please re-open if there is more to do.

closed 2024-01-04 by RobMeades