u-blox/ubxlib #66: define the local port when creating a socket.

eeFLis, 2022-08-23:
Hi Rob

When we restore the sockets after 3gpp psm we want to open the socket with the same local port as before so we do not break the DTLS session.

ubxlib skips this option in the AT command
AT+USOCR=<protocol>[,<local_port>[,<preferred_protocol_type>[,<cid>]]]

Maybe an option that could be added.

RobMeades, 2022-08-23:
Hi there: I hadn't even realised this was a possibility! Will see if I can add it in a backwards-compatible way.

RobMeades, 2022-08-23:
Having discussed this internally:

- we don't want to change the `uSockCreate()`/`uCellSockCreate()` API as that would break the world,
- we would naturally do this through `uSockBind()`/`uCellSockBind()` but those APIs required a descriptor so can't be done until after the socket is created,
- so the suggestion is that we could add a `uSockNextLocalPort()`/`uCellSockNextLocalPort()` API, which would set the local port to be used by the very next "create".  This is not thread-safe, of course; if thread-safety is require you would have to provide that via mutex protection yourself.  -1 could be used to cancel a "next local port" setting.  Once `uSockCreate()`/`uCellSockCreate()` has been called the local port setting would be reset back to "automatic", so `uSockNextLocalPort()`/`uCellSockNextLocalPort()` need only be called when required.

Does that approach make sense for you?

eeFLis, 2022-08-23:
Hi Rob

thanks for the quick answer. This solution will work for us.

I have another question related to sockets and psm where I hope you can help me with.
We have observed that after a UDP packet transmission about 25 seconds elapse before the RRC is released.
then the timer T3324 runs down and the module switches to psm.

We are trying to reduce the time befor RRC is released to save more power.
Therefore, we use AT+USOSTF with rai_flag 1 instead of +USOST to release the connection immediately after sending.

But that has no influence. Do you have an idea?

RobMeades, 2022-08-23:
This is one for @philwareublox - could you possibly paste it into a new issue that I could assign to Phil?  Then we can use this one to track the local port thingy.

eeFLis, 2022-08-23:
I have just opened a new issue #67

RobMeades, 2022-08-24:
Hi again: I've just pushed a preview branch, https://github.com/u-blox/ubxlib/tree/feature_sock_local_port_rmea, where the most recent commit implements the feature your requested.  Could you take a look and confirm that it does what you want?  If so, I will push it to `master` [and delete the preview branch].

eeFLis, 2022-08-25:
Hi Rob
I have tested the feature and it works great. Thanks a lot

RobMeades, 2022-08-25:
Thanks for doing that so swiftly, I will let you know when the change is pushed to `master` here.

RobMeades, 2022-09-09:
The change is now available on `master`, see commit 77156a9c1cabfb6d80c93c92852f983cee3d2839; I will delete the temporary branch and close this issue.

closed 2022-09-09 by RobMeades