u-blox/ubxlib #67: RRC Release Assistance

eeFLis, 2022-08-23:
hi there 

We have observed that after a UDP packet transmission about 25 seconds elapse before the RRC is released.
then the timer T3324 runs down and the module switches to psm.

We are trying to reduce the time befor RRC is released to save more power.
Therefore, we use AT+USOSTF with rai_flag 1 instead of +USOST in ubxlib to release the connection immediately after sending.

But that has no influence. Do you have any idea?

RobMeades, 2022-08-23:
@philwareublox : I think this is one for you...

eeFLis, 2022-08-23:
By the way, we use the following module:

Product type number: **SARA-R510S-01B-00**
modem_version: **03.15**
applications version: **A00.01**

philwareublox, 2022-08-23:
Hi,
This is the correct way to use the RAI flag in +USOSTF, but it also requires support by the network you are using. 
Do you know if the MNO you have connected to supports NAS-RAI feature?

(There is no way of checking on the module side, apart from 'it should work, if the network supports it')

Phil.

eeFLis, 2022-08-23:
Hi,

We did the same tests a long time ago with a SARA-R510S-00B-00 connected to the same MNO.
The measurements from back then show that the connection was released immediately after sending.
so i think it is supported. but i can clarify this to be sure.

eeFLis, 2022-09-26:
Hi,

I get confirmation from the network operator that RAI is supported for Cat-M1 and NB-IoT.
RAI works on the module with NB-IoT but not with Cat-M1.

philwareublox, 2022-10-12:
Hi.
NB-IoT is a Non-Access Stratum technology were the messages go over the control-plane. +USOSTF sets the NAS-RAI flag in the NAS message to instruct the network to release the RRC

CatM1 is a Access Stratum technology were the messages go over the constructed user-plane. +USOSTF will not work in CatM1.

CatM1 RRC Release Assistance is controlled by AS-RAI. Here the cellular module will send a Buffer Status Report message instructing the network there is no more data in its buffer, BSR: 0 message. The network will then release the RRC connection.

I will need to get back to you how to send the BSR: 0 message in SARA-R5xx.....

eeFLis, 2022-10-24:
Hi.
Thank you very much for the explanation.
We missed to trigger a packet send after setting buffer status report (BSR) to 0.
CatM1 RAI works now.

closed 2022-10-24 by eeFLis