u-blox/ubxlib #68: Abort / Interrupt long operations

owain-incus, 2022-08-25:
As per probably most of your customers I use your higher level API's to build cell connections etc. uNetworkUp().

This operation may take many seconds to complete; how can I abort this operation from API's at this level.
uNetworkUp() eventually calls the uCellNetConnect() API; which uses a callback to determine whether to keep trying an operation or to abort that operation. Currently this callback has been fixed to just be a timeout but not allow interruption from the top level API.

The timeout on the operation is configurable at compile time; but it would be good if higher level API could specify the keepGoingCallback; or have an abort operation API call.

My user story is user has tried to start the cell connection; but then decides he wishes to abort and power down the unit. Currently unless I resort to using low level API's I have no way of aborting the starting of the connection and shutting down gracefully. One must wait for the connection to ether time out or be made before one can bring the connection back down.

RobMeades, 2022-08-25:
Hi Owain: an excellent point, as ever.  We originally tried to keep the higher-level APIs uncluttered in this respect but maybe that was a mistake; it is, after all, a fact of life that the application may need to abort and you can't decide to go "underneath" the higher level API arbitrarily 'cos it will have hijacked the callback you'd need.

I will try to think up a mechanism that would work in a backwards compatible way, or if you have any suggestions I'd be very happy to entertain them.

RobMeades, 2022-09-15:
The simplest way to do this would be to add a `keepGoingCallback()` item to the configuration structure for a network type, so the [cellular configuration structure](https://github.com/u-blox/ubxlib/blob/master/common/network/api/u_network_config_cell.h#L49) might have the following added to it:

```
bool (*pKeepGoingCallback) (uDeviceHandle_t); /**< if set, this function will be called periodically
                                                   during an "abortable" operation; while the function
                                                   returns true the operation will continue, else it will
                                                   be stopped and this code will return. If this is set
                                                   timeoutSeconds will be ignored.  If you do not need
                                                   this facility, ignore it, i.e. allow you compiler to
                                                   set the field to zero, and timeoutSeconds will be obeyed
                                                   instead. */
```

Would that suffice?

closed 2022-09-26 by RobMeades

RobMeades, 2022-10-01:
The fix in question has now been merged to the `master` branch here in commit 52c91449c32c364d75781db2e105c814eb64902b.

chenlijun99, 2023-03-24:
Hi @RobMeades. Is there a way to get a user-defined pointer in the `pKeepGoingCallback1` callback? I tried to search whether I can attach some user data to the `uDeviceHandle_t` handle, but I couldn't find anything.
Thanks!

RobMeades, 2023-04-27:
@chenlijun99: hi, apologies for the delay in replying, had lost track of this issue because it was closed.

Unfortunately, for the `pKeepGoingCallback` cases, we decided not to include a user-defined pointer in the function signature: these are intended as pretty simple go/no-go functions that we use in quite a few places and, if we had to store a user-defined pointer as well, we'd end up adding storage for that etc. which we didn't think was really worth it.  Sorry!

chenlijun99, 2023-04-27:
@RobMeades I can understand your design choice. While I would prefer to be able to have the user-defined pointer, it is nothing that I can't solve at the application level.
Anyway, thanks for the reply. At least I'm sure I didn't miss any obvious documentation.

RobMeades, 2023-10-14:
> Hi @RobMeades. Is there a way to get an user-defined pointer in the `pKeepGoingCallback1` callback? I tried to search whether I can attach some user data to the `uDeviceHandle_t` handle, but I couldn't find anything.

@chenlijun99: maybe not relevant now but, just in case it is, @alonbl has just proposed and implemented a scheme which, provided you open your cellular device with [uDeviceOpen()](https://github.com/u-blox/ubxlib/blob/7df8ac914de68cce7fec96fa7de79f9c965ae3d0/common/device/api/u_device.h#L409), allows you to attach a user context to the device, which you then retrieve inside any callback.  You can find it at commit b4941958a7438d9e1deeaf43a466759916dae6d9.

chenlijun99, 2023-10-16:
Hi @RobMeades. Many thanks for letting me know!