u-blox/ubxlib #69: ubxlib port to MSP430FR5969

bytespectrumllp, 2022-10-08:
Hi,

We are planning to interface MSP430FR5969 with UBlox SARA-R422S.

So I want to port ubxlib in MSP430FR5969.

I checked ubxlib port are available for STM32, ESP32, nRF5340, nRF52840 etc.

These are more powerful and resource rich host controller than MSP430FR5969.

My doubt is ubxlib is portable to MSP430FR5969 as it is comparatively low end controller.

Can you please confirm, ubxlib is portable or not to MSP430FR5969?

-Chandra

RobMeades, 2022-10-08:
Hi there, and thanks for getting in contact.  That is certainly an interesting choice of MCU: cellular generally sits in the relatively "high-power" world, it being a wide area network with consequential high transmit powers, so we don't often see it paired with low-end (but obviously highly power efficient) MCUs of this nature.

A few things:
- with such a small amount of program and data memory, whether everything will fit depends how big your application is likely to be: do you have any ideas/budget?
- `ubxlib` needs to run on an RTOS, which of course will also take some resources: have you made any choices in this regard?

Looking at this specific MCU, if I have read the data sheet correctly, it seems to have 64 kbytes of program/data memory plus 2 k bytes data memory, so your problem is going to be program space: 64 kbytes is very small indeed, probably about half of what `ubxlib` alone would require.  RAM-wise `ubxlib` runs happily on MCUs with only 64 kbytes RAM but on this MCU you have 66 kbytes _total_ _memory_, program and data included.

So my guess is, assuming that I have read the MCU data sheet correctly, the answer to your question would be "not out of the box".  However...

`ubxlib` is delivered as full source code for a reason, you can take it as a basis and chop/slice as required.  I think a fundamental decision is whether there is room in your design for an RTOS, does your application _need_ an RTOS, or is an RTOS overkill?  FreeRTOS, for instance, takes very few system resource (maybe 10 or 20 kbytes, which would normally be considered "small"!).  *IF* you have room for an RTOS then you could, for instance, take the [port](https://github.com/u-blox/ubxlib/tree/master/port)/[AT client](https://github.com/u-blox/ubxlib/tree/master/common/at_client)/[cell](https://github.com/u-blox/ubxlib/tree/master/cell) bits of `ubxlib` (plus other relevant headers) and use them as a basis for your SARA-R422 interface.  This way you can take advantage of a lot of the basic work that `ubxlib` does - talking to a cellular module "nicely", and ensure the minimum possible overhead.

But there has to be an RTOS for `ubxlib` to work.  It _might_ be possible to make `ubxlib` work without an RTOS, but you would need to somehow provide an [event queue](https://github.com/u-blox/ubxlib/blob/master/port/api/u_port_event_queue.h), required to handle the asynchronous unsolicited result codes (URCs) that, unfortunately, all AT interfaces may send, at any time (on a line-buffered basis). This is essentially a "tasky" kind of thing; it _might_, maybe, be possible to somehow implement it with interrrupts?  It is worth bearing in mind that, whether you use `ubxlib` or not, you will need to have a mechanism for handling such asynchronous responses from the cellular module in your design, they are an unfortunate fact of life with any cellular module.

In case it is of interest, I will also mention this:

https://www.u-blox.com/en/product/SARA-R510AWS-module

This is a module we are making with a new Amazon-driven AT interface which is _specifically_ designed to work with low-end MCUs such as the one you have chosen.  This, quite deliberately, has _no_ URCs, takes care of security for you, etc., and of course only interfaces into Amazon web services (basically MQTT messaging).  For this you would not need `ubxlb` - a child could write the code :-).  However, it is SARA-R5 rather than SARA-R422, so no 2G fallback.

I think I've burbled enough: please ask more questions if you think I can help.

bytespectrumllp, 2022-10-11:
Hi there,

Thank you very much for this detail explanation.

As of now, I do not have exact idea about its end use. 

Yes, we are planning to use FreeRTOS.

We are discussing this internally to reach some conclusion.

Meanwhile, I am just wondering as MSP430 is 16-bit MCU not 32-bit. Will this make any difference?

-Chandra

RobMeades, 2022-10-11:
While we've never used a 16-bit MCU, `ubxlib` has no particular speed requirements, the main difference is going to be native pointer size which will hopefully raise compiler errors should problems exist; for instance, it might be that we assume a pointer is the same size as an `int32_t` somewhere.  This might cause some initial issues during development but not unsurmountable I'd say.

bytespectrumllp, 2022-10-14:
Thanks for information

closed 2022-10-14 by bytespectrumllp