u-blox/ubxlib #70: `uSecurityTlsRemove` Memory Leak

wrh-dev, 2022-10-11:
Thanks for creating this library!

I see a small bug that results in a memory leak in `u_security_tls.c`. I think this code:

```c
void uSecurityTlsRemove(uSecurityTlsContext_t *pContext)
{
    if ((pContext != NULL) && init()) {
...
```

Should be updated to:

```c
void uSecurityTlsRemove(uSecurityTlsContext_t *pContext)
{
    if ((pContext != NULL) && (init() == 0)) {
...
```

to account for the return type of the `init()` function.

Does this change make sense?

RobMeades, 2022-10-11:
Hi there and thanks very much for spotting this: someone in one of our teams spotted it also, not more than a month ago, so you're very on the ball!  You'll find the fix here in this commit 2bce269d5ed03598765e672874b6d0118a871f81.

wrh-dev, 2022-10-11:
Oh, perfect! I'll go ahead and close out this issue. Thanks!

closed 2022-10-11 by wrh-dev