u-blox/ubxlib #73: Allow to override dynamic memory related functions in ubxlib

johannesmay, 2022-10-21:
Do you think it would be possible to implement a way to override all of the functions related to memory allocation or deallocation in a ubxlib port?
Our system is running on multiple heaps due to MPU restrictions, so it would be nice to be able to use a separate heap implementation for ubxlib that lives in a shared memory area.
We did run into some trouble trying to override malloc and free using defines because not all libraries include a common port header after the standard library headers.

If there currently is no possibility to do this, could you imagine adding support for such overrides?

RobMeades, 2022-10-21:
Hi Johannes: this has been on our TODO list for some time, at least partly so that we can monitor for memory leaks more neatly than we currently do, we'd just not got around to it yet.  Would you be thinking of us adopting something like a `uPortMalloc()/`uPortFree()` that we would global search/replace `malloc()`/`free()` with?  Wouldn't be a difficult change for us to make.

I guess using the linker's `--wrap` directive (assuming you are using GCC or similar), for example as we already do [here](https://github.com/u-blox/ubxlib/blob/master/port/platform/nrf5sdk/mcu/nrf52/gcc/runner/Makefile#L209), is not an option for you, as an immediate fix?

RobMeades, 2022-10-21:
Good - we don't use `calloc()`/`realloc()`/`strdup()`/`strtok()` (only `strtok_r()`) and we don't use floats (we check for this by parsing the `.map` file, hope the check isn't failing!).  Lemme have a go at the wrapping thing and get back to you.

RobMeades, 2022-10-21:
I've pushed a preview branch of the proposed solution here: https://github.com/u-blox/ubxlib/tree/preview_malloc_rmea.  It appears to compile and pass all our tests: can you let me know if it meets your needs?  I will need to do some more checking and add some automation which ensure we don't go calling `malloc()`/`free()` by mistake but I think this is basically it.

johannesmay, 2022-10-26:
That was quick, thanks for the work!
It looks exactly like what we had in mind, no surprises here 👍

RobMeades, 2022-10-26:
Thanks for checking: that commit is now pushed here in aaf35720f3e17347f525ce8967bfca8369369176, preview branch will be deleted shortly.

closed 2022-10-26 by RobMeades