u-blox/ubxlib #75: deep sleep state

eeFLis, 2022-10-28:
Hi Rob

is there a way to detect when the module is going to enter deep sleep right before vint = low?
I think the only possibility is to check for `+UUPSMR: 1` URC. I have seen that you already do this in the "private" area.
is it possible to make the `deepSleepState` readable through the API as well?

RobMeades, 2022-10-28:
Hi there: I'll need to refresh my memory, and probably consult @philwareublox, the expert in these matters (who is out of the office this week) but I believe the answer is that there is no good way, which is why I kept that URC detection over in the private area.

I will get back to you...

eeFLis, 2022-10-31:
hi Rob 
thanks for your answer.

It is fine for us if its only work for SARA-R5. we use a SARA R510s module.

we rather think of a get function for the deepSleepState `uCellPrivateDeepSleepState_t ` instead of a +UUPSMR: 1 callback.
it would also be helpful if there is an additional state in `uCellPrivateDeepSleepState_t `when  deep sleep is prevented (+UUPSMR: 2).

What do you think about that?

eeFLis, 2022-10-31:
that would be great.

since there is `U_CELL_PWR_3GPP_POWER_SAVING_STATE_UNAVAILABLE ` would it be possible to have a additional state like U_CELL_PWR_3GPP_POWER_SAVING_PARMETERS_AGREED_BY_NETWORK? which indicates that the parameters have been agreed by the network.

RobMeades, 2022-10-31:
Yup, can do that, let me put something together and I'll post a preview branch here for you to look at.

RobMeades, 2022-10-31:
I have pushed a preview branch here:

https://github.com/u-blox/ubxlib/tree/preview_cell_3gpp_ps_state

Note that I've performed no testing on this as yet, all I know is that it passes compilation/static-analysis/doxygen/code-style checking.  When you get a chance, please let me know if it does what you want and I will then start testing.

eeFLis, 2022-10-31:
that was fast. thank you that is exactly what we need.

RobMeades, 2022-10-31:
Great, thanks for the swift feedback.  Feel free to use the preview branch if you want to get moving quickly; I won't be able to get the testing/reviewing done until next week but I think it is pretty unlikely to change.

When I have update the `master` branch here with the final version I will let you know and will delete the preview branch some time after that.

eeFLis, 2022-11-28:
Hi

There seems to be a issue when `uCellPwrGet3gppPowerSavingState `is called multiple times when the module is in deep sleep (VINT==low).
the function returns `U_CELL_PWR_3GPP_POWER_SAVING_STATE_ACTIVE_DEEP_SLEEP_ACTIVE `in the first call which is correct. 
However, all calls afterwards returns `U_CELL_PWR_3GPP_POWER_SAVING_STATE_AGREED_BY_NETWORK `.

this seems to be because in uCellPrivateIsDeepSleepActive() the deepSleepState is set to `U_CELL_PRIVATE_DEEP_SLEEP_STATE_ASLEEP` afterwards uCellPrivateIsDeepSleepActive() is not called again in `uCellPwrGet3gppPowerSavingState()` .

RobMeades, 2022-11-29:
Ah, well spotted, so would this be the right fix do you think (see the additional "or" check on `deepSleepState` in the `if()` statement):

```
powerSavingState3gpp = U_CELL_PWR_3GPP_POWER_SAVING_STATE_AGREED_BY_NETWORK;
if (pInstance->pSleepContext->powerSaving3gppOnNotOffCereg) {
    if (pInstance->deepSleepBlockedBy >= 0) {
        powerSavingState3gpp = U_CELL_PWR_3GPP_POWER_SAVING_STATE_BLOCKED_BY_MODULE;
        if (pApplication != NULL) {
            *pApplication = pInstance->deepSleepBlockedBy;
        }
    } else {
        if ((pInstance->deepSleepState == U_CELL_PRIVATE_DEEP_SLEEP_STATE_PROTOCOL_STACK_ASLEEP) ||
            (pInstance->deepSleepState == U_CELL_PRIVATE_DEEP_SLEEP_STATE_ASLEEP)) {
            powerSavingState3gpp = U_CELL_PWR_3GPP_POWER_SAVING_STATE_ACTIVE;
            if (uCellPrivateIsDeepSleepActive(pInstance)) {
                powerSavingState3gpp = U_CELL_PWR_3GPP_POWER_SAVING_STATE_ACTIVE_DEEP_SLEEP_ACTIVE;
            }
        }
    }
} else {
    powerSavingState3gpp = U_CELL_PWR_3GPP_POWER_SAVING_STATE_BLOCKED_BY_NETWORK;
}
```

eeFLis, 2022-11-29:
yes that should fix the issue. Thank you

RobMeades, 2022-11-29:
Great, I've updated the [preview branch](https://github.com/u-blox/ubxlib/tree/preview_cell_3gpp_ps_state).  Haven't managed to get this onto `master` yet as I've been re-writing the test system for a few weeks, will let you know when it gets there.

eeFLis, 2022-11-30:
btw 

during the long time test we noticed that sometimes the at client seems to be  confused what was already read from the RX buffer.
The debug output shows messages that were not sent at this time by the module (we have checked the UART communication) and the printed messages are identical with the messages from the past.

Have you observed this behavior on your test system or do you have an idea what could be causing it?

RobMeades, 2022-11-30:
That's interesting: our testing is limited to about 40 minutes of fairly intense AT activity, we don't have a long term test setup at the moment, hence it's not something I've seen.  That said, the test system re-write is intended to allow us a _lot_ more flexibility, so we could set such a thing up.  Can you describe the kind of scenario?

eeFLis, 2022-11-30:
the scenario is as follows:

1.	Wake up from psm.
2.	send data to a UDP socket.
3.	Wait for an response.
4.	wait until the module switches to psm (VINT == low) and sleep for about 60 Seconds
5.	start again at point 1

this works for an hour and more before the error occurs

RobMeades, 2022-11-30:
Great - looks like it is time I set that up as a separate long-term test.  Will aim to do so in the next few weeks.

The buffering code in the AT Client is more complex than it needs to be in order to support the intercept functions, needed by short-range EDM mode and cellular chip-to-chip encryption, so there _could_ be a problem in there somewhere.

eeFLis, 2022-11-30:
is there a way around this when it is not needed? This is because the application will stop working if it thinks it is receiving data that does not really exist.

RobMeades, 2022-11-30:
Unfortunately not - the code is just written to include that functionality (see [bufferFill()](https://github.com/u-blox/ubxlib/blob/master/common/at_client/src/u_at_client.c#L1172)), taking it out would require re-writing the code, which would of course be simpler, but then you've got new AT Client buffer handling code to re-test, whereas the current stuff has been in place for a few years now and so is, at least in other respects, pretty well hardened.

We will work to track it down: if you gain any further insights please update this ticket, or maybe start another dedicated to the subject.

eeFLis, 2022-12-06:
Hi Rob

what we see when the issue occurs is that `urcCallback` reports a lot of readable data.
But definitely no data has been received between "URC checking done" and "data readable 565".
and the "readable data" are those already received at an earlier time.

```
U_AT_CLIENT_0-0: possible URC data readable 13, already buffered 0.
+CSCON: 1
U_AT_CLIENT_0-0: URC checking done.
U_AT_CLIENT_0-0: possible URC data readable 565, already buffered 0.
```

RobMeades, 2022-12-06:
Interesting: depending on how adventurous you feel and how much static RAM you have free, there is a compilation switch [`U_CFG_AT_CLIENT_DETAILED_DEBUG`](https://github.com/u-blox/ubxlib/blob/master/common/at_client/src/u_at_client.c#L2417) in the AT Client which I used when debugging buffering behaviour for chip-to-chip encryption and left there in case of a future need.

When defined this causes log points that are already present in the code to become active and record the AT client buffer state to a static logging buffer.  The logging is started and stopped with `uAtClientDetailedDebugOn()`/`uAtClientDetailedDebugOff()` (without destroying the buffer if I remember correctly); the application should call `uAtClientDetailedDebugPrint()` when it wants to to print the buffer out and empty it, also called automatically on `uAtClientDeinit()`.

I'm not yet done with the test system re-write, should be this week though, then I can try to reproduce the problem.

eeFLis, 2022-12-07:
thanks for the hint. I have seen this option, unfortunately we dont have enough memory for it.

RobMeades, 2022-12-07:
How much RAM might you have free?  We [currently set the number of entries in the logging array to 1000](https://github.com/u-blox/ubxlib/blob/master/common/at_client/src/u_at_client.c#L513), which is likely overkill.  Each entry is 16 `int32_t`'s, so 64 bytes each, making it 64 kbytes total.  If you had, say, 5 kbytes free you could set it to 75 entries and might capture something.

RobMeades, 2022-12-07:
Oooo, interesting, let me refresh my memory as to what the log points mean and I'll post back here...

RobMeades, 2022-12-07:
Hmmm, yes, you're right: between [`LOG_BUFFER_FILL(3)`](https://github.com/u-blox/ubxlib/blob/master/common/at_client/src/u_at_client.c#L1274) and [`LOG_BUFFER_FILL(4)`](https://github.com/u-blox/ubxlib/blob/master/common/at_client/src/u_at_client.c#L1291) `readLength` goes from 0 to 1124.

And, barring stack weirdness, the only way it can do that is if [`uartReadNoStutter()`](https://github.com/u-blox/ubxlib/blob/master/common/at_client/src/u_at_client.c#L1126) returned that value, and the only way it can do that is if `uPortUartRead()` for STM32F4 is returning 1124.  So I think suspicion moves to the UART driver.

Are you using a debugger?  If so, might it be possible to drop some diagnostic information into some variables down in the UART driver and then put a break-point on where the problem occurs and go look at them?  If not, let me try to think up something else.

eeFLis, 2022-12-07:
Yes we use a debugger. We will do it as you suggested.

However, the issue occurs only rarely and sometimes first after hours. 
I will report back when I have more information.

RobMeades, 2022-12-07:
Thanks, and sorry to encumber you with this, I really will get to it next week, but if you happen to find out something first then do let me know.

eeFLis, 2022-12-12:
I think we have found the problem

the error happened when we migrated the STM32F4 porting layer to STM32L4.
at `LL_DMA_GetDataLength `we didn't take care that the channel (“stream” in 
STM32F4 lines) is shifted by one compared to the channel name.

```
#define LL_DMA_CHANNEL_1                  0x00000000U /*!< DMA Channel 1 */
#define LL_DMA_CHANNEL_2                  0x00000001U /*!< DMA Channel 2 */
#define LL_DMA_CHANNEL_3                  0x00000002U /*!< DMA Channel 3 */
```
since then the error has not appeared again
So I am sorry that I have encumbered you with this.

RobMeades, 2022-12-12:
Woohoo!  Well found, the ST HAL is a bit of a nightmare for "out by one" problems.  I really hope that is it; if not, please don't hesitate to raise the problem again.  I won't feel encumbered :-).

RobMeades, 2023-01-18:
I believe the issues raised here are now sorted with commit 0c56afaa79d4c68a8fa64ef24372b16d6565f5dd being pushed to `master` here, will close this issue and delete the [preview branch](https://github.com/u-blox/ubxlib/tree/preview_cell_3gpp_ps_state) in a few weeks.

closed 2023-01-18 by RobMeades