u-blox/ubxlib #77: Certificate Upload fails on fresh modems

johannesmay, 2022-12-14:
Hi!

We have noticed an issue with our modem (Sara R422), which seems to be a modem firmware bug as far as I can tell.
Our normal support channel for these issues is already informed, but we are still waiting for feedback and we can maybe discuss if this is reproducible for you and if a workaround could be integrated into ubxlib.

We are using `uSecurityCredentialStore()` to store a certificate on the device. Since this caused some issues, we started to investigate the AT commands being sent:

The issue is that the device will reject certificate uploads with CME Error 1912 like this (lines starting with 0 are our TX line, lines with one are RX; this data was recorded with a UART sniffer):
```
0 09:52:40.608:
0 AT+USECMNG=0,0,"ROOT_TEST",2090
1 09:52:40.631:
1 
1 +CME ERROR: 1912
```

The modem is a "fresh" modem, i.e. it never had a certificate uploaded before and also never had any other file system interactions.
We noticed that this issue can be fixed by sending an `AT+ULSTFILE` command before uploading the certificate like this:

```
0 09:54:14.473:
0 AT+ULSTFILE
1 09:54:14.625:
1 
1 
1 +ULSTFILE: 
1 
1 OK
1 
0 09:54:14.755:
0 AT+USECMNG=4,0,"ROOT_TEST"
1 09:54:14.761:
1 
1 +CME ERROR: 4
1 
1 
0 09:54:14.896:
0 AT+USECMNG=0,0,"ROOT_TEST",2090
1 09:54:14.913:
1 
1 >
0 09:54:14.989:
0 -----BEGIN CERTIFICATE-----
0 (Certificate data left out)
0 -----END CERTIFIC
0 09:54:15.170:
0 ATE-----
1 09:54:15.486:
1 
1 +USECMNG: 0,0,"ROOT_TEST","A5A
1 09:54:15.489:
1 7FAB5A729845E29BB81B5FF59B818"
1 
1 09:54:15.492:
1 
1 OK
```

This behavior is also reproducible for us after resetting a modem to the factory settings.

Our modems are on firmware version 00.12 and we are currently still using the ubxlib revision `574f3f3c2b6a5c56f167a3879f6e92235a56f631`, which is a couple commits after the 1.0 release.

RobMeades, 2022-12-14:
That's fascinating thanks Johannes, let me discuss internally and get back to you.

RobMeades, 2022-12-19:
Apologies for the delay in replying, haven't been able to get in contact with the right person internally and don't have a SARA-R422 that I can play with at the moment, so maybe we JFDI and add a workaround: could just do a "throw-away" `AT+ULSTFILE` before the first `USECMNGR=0` issued by any cellular instance?

johannesmay, 2023-01-02:
Hi, I was on my christmas break anyway, so no problem :)
Yeah, a workaround like that would work. We can also implement that on our side if you want to wait for some test hardware before adding this to ubxlib.

RobMeades, 2023-01-08:
Belated happy 2023 to you: I've had a look at the implementation now and the problem is that, because the `AT+SECMNG` interface is utterly common between our Wifi and cellular modules, it is completely unaware of the module type it is talking to and, in fact, carries with it no context data whatsoever.  All nice and neat, oh so clever of us, however it means we have no way to store "this is the first access of the module" in the context of that thing, as there isn't one, if you see what I mean.

My inclination would be to "fix it in the documentation": add a note to [uSecurityCredentialStore()](https://github.com/u-blox/ubxlib/blob/master/common/security/api/u_security_credential.h#L131) that a fresh SARA-R4 cellular module may fail certificate storage on the first occasion and that a workaround for this is to call [uCellFileListFirst()](https://github.com/u-blox/ubxlib/blob/master/common/security/api/u_security_credential.h#L131) followed by [uCellFileListLast()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_file.h#L231) before calling [uSecurityCredentialStore()](https://github.com/u-blox/ubxlib/blob/master/common/security/api/u_security_credential.h#L131) for the first time.

The security credential store does have the AT client context so I could, potentially, add a "give me a user context" API to the AT client, allowing a caller to ask for a context space of a given size inside the AT client and pass it back to the caller when required, but it seems overkill for what is a very specific case, and of course there would only be one of it and we'd have to be sure no-one else decided to use it.  Would need further design.

Or maybe I could keep a small array of AT client handles in the security credential storage code and, if it were the first time we saw the handle, I would do the `AT+LSTFILE` thing first (I can, at least, tell whether the device we're talking to is cellular or short range).

What do you think?  Any other ways to square it?

RobMeades, 2023-01-09:
FYI, in case it hasn't got back to you, my esteemed colleague @philwareublox found the ticket that was raised as a result of your bug report: the bug has been fixed and queued into the -02B maintenance release of SARA-R422 FW. I can't, at the moment, be exactly sure of when this release is scheduled but it looks like 2Q23.

johannesmay, 2023-01-09:
Hi Rob, happy 2023 to you as well :)

I'd second the suggestion of the documentation fix.
Doing the workaround in software using manual `uCellFileListFirst()` and `uCellFileListLast` is easy enough and it's a temporary issue anyway.
Since the bug is confirmed by u-blox and will be fixed it shouldn't be necessary to add any workaround to the firmware.

Thanks for letting me know about the fix!

closed 2023-01-18 by RobMeades