u-blox/ubxlib #78: gnss location M10 support

hcglitte, 2022-12-15:
Hi,

Currently I see that there is support for M8 and M9.
Will there be support for M10 (specifically MAX-M10S)?

What changes are necessary to include to the library in order to incorporate support for M10?

From reading online, U-BLOX mentions that the ubxlib library is used for testing of all modules internally at U-BLOX.
If this is the case, wouldn't it make sense that the M10 is included in the supported devices in u_gnss_module_type.h ?

Thanks for any help.

Regards HC

hkro, 2022-12-16:
Hi,

Yes, support of the M10 series planned for Q1 2023, we're working on that.

Regarding the scope. ubxlib is primarily used (and intended to be used) by customers who develop something with u-blox products, typically on a microcontroller. Hereby ubxlib provides a starting point and does the heavy lifting by providing tested source code examples of functionality which is common across many use cases. Find more high-level information here: https://www.u-blox.com/en/product/ubxlib

With test system of ubxlib we ensure that every release is automatically tested across all MCU - module combinations. So testing is a key aspect of ubxlib but not the primary use case.

About which changes are necessary to include M10 in ubxlib @RobMeades might have some insights.

RobMeades, 2022-12-16:
The main change in M10, from a software perspective, is that the previous configuration interface, used in M8 and M9, but deprecated in M9, is now removed.  `ubxlib` supports the new configuration interface but, internally, still uses the old one in various places (e.g. when powering up and when confirming that unacknowledged messages have been accepted by the module).

The changes required to support M10 are to use the new configuration interface in those places if an M10 GNSS chip is connected.

hcglitte, 2022-12-16:
Thanks for your quick response. I will then buy a M9 instead and do implementation in this.

A sidenote: In my application I am interested in the PPS feature mainly. My application currently has a multitude of peripherals, an OS and "my ecosystem". When I want to include yet another peripheral (u-blox GNSS in this case with a few IO) I expect (and hope) to find a library which I can install and do some simple minor HAL modifications to support my MCU. For the GNSS case: UART/I2C/SPI and INT+GPIO - and not a lot of generic code surrounded to these simple HAL files. 

The ubxlib library has support for all your modules which is impressive and great for customers using many of your modules on one board (I understand you use this in your test farm). Removing "wifi", "ble" etc was not possible at first glance due to dependencies. There is even an OS required to instantiate modules as objects and more, lots a debug code and even lib_common to make libraries(!) etc. This means I need to spend weeks possibly to adapt, remove unnecessary "current dependencies", and then finally do the real work to adapt for M10 (if I still want to go for this module).  As a customer I would like to easily find the minimal required source files to implement the one module of interest. Is an OS really required to send I2C messages to the GNSS module? Yes, an OS is great for a big project, but not for a quick evaluation for the GNSS features. Then later, I can adapt your driver to suit my ecosystem with all my own fancy generic IO setups, OS etc. In my opinion a metric should be "How fast can a customer setup your driver code into his own MCU ecosystem with minimum effort". Perhaps I am the only one who thinks this way, but this is at least my opinion. And of course, I may be wrong - perhaps all the other dependencies are required for operation of the GNSS module alone.

I will continue to strip down the library to suit GNSS support only. Feel free to comment and let me know if my thoughts are totally off here.

RobMeades, 2022-12-16:
Thanks for the feedback, perfectly valid comment.

FYI, it is now possible to lose the [short_range](/common/short_range)/[wifi](/wifi)/[ble](/ble)/[cell](/cell) code if you just want to use [gnss](/gnss): see [this preview branch](https://github.com/u-blox/ubxlib/tree/preview_separation_use_this_one_rmea); we've just not yet been able to push this to `master` here for various reasons.  You might be better off starting from that branch, if it is not too late.

On the more general issue of granularity, we decided from the outset to rely on an RTOS.  You're quite correct that it is not necessary to send/receive messages to a GNSS chip but RTOS's are pretty small, MCUs are becoming more capable, and all of the MCUs we chose to support come with some form of RTOS integration, so it was a simplification on our part to rely on one.  Similarly, we use dynamic memory, which I know some people do not like, for much the same reason [I've worked in embedded software since the last century and have never, ever, had an issue with `malloc()`ators, but maybe I've never been constrained enough].

All a bit of a compromise really.

RobMeades, 2023-01-18:
FYI, M10 support is now added, from commit d61cb5097c1791c836bcd73712d3ef0b8b48904a.  I will close this issue, please re-open if you need more.