u-blox/ubxlib #79: Linux support

Hadatko, 2023-01-12:
Hi, i am a bit confused from documentation and code. Am i correct that i currently cannot build gnss application under Linux with using this library which is communicating with gnss device using uart?:
PORT = "/dev/ttyLP2"
DEFAULT_BAUDRATE = 9600
OPERATIONAL_BAUDRATE = 115200

RobMeades, 2023-01-12:
Hi there: short answer, if I've understood your question correctly, is that yes you can build `ubxlib` into a Zephyr "native posix" application and have it talk to a GNSS module connected to that Linux box as a `/dev/tty`.

Starting from the start, our Linux support is limited to being able to use Zephyr with its in-built [native posix](https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html) platform, which is intended for development/test purposes only, rather than being a "production" environment.

The way Zephyr works in this "native posix" mode is that it exposes two pseudo-TTYs, i.e. something like `/dev/pts/5`, where the number at the end is randomly selected by Linux, hence the [note here](https://github.com/u-blox/ubxlib/tree/master/port/platform/zephyr#linuxposix) on how to map those randomly selected PTSs to fixed, real TTYs, which is where your GNSS chip might actually be.

You should be able to clone or sub-module this repo into your project, tell Zephyr to bring in `ubxlib` as a Zephyr sub-module according to the [instructions here](https://github.com/u-blox/ubxlib/tree/master/port/platform/zephyr#integration), have your application `#include <ubxlib.h>` and call `ubxlib` APIs etc. using 0 or 1 to identify one or other of those UARTs as the UART HW block when you call [uDeviceOpen()](https://github.com/u-blox/ubxlib/blob/master/common/device/api/u_device.h#L336).  Then, when you run the Linux executable that you have built, Zephyr will print out which PTS's it has selected and you can call `socat` to do the mapping.

I'm afraid that's the only way, as far as I am aware, that Zephyr interacts with the real world in its Linux flavour, and this mechanism is currently the only way that `ubxlib` supports Linux.  We may, at some point, support plain Linux, no Zephyr in the way (as we support Windows, for development purpose only again, on the WIN32 API), it's just not something that has got high enough up the priority list yet.

But maybe I've misunderstood your question: please restate if that is the case.

Hadatko, 2023-01-12:
Thank you for quick answer.
We want rewrite our python code to c++. In python code we were using https://github.com/semuconsulting/pyubx2. We have application in alpine container running on linux kernel (no zephyr dependencies so far currently.)
This solution you are providing sounds a bit complicated. Will think about that more. THank you.

RobMeades, 2023-01-12:
Actually, let's leave this open and I'll tag it as the feature request.

RobMeades, 2023-08-09:
Linux is now supported, see see https://github.com/u-blox/ubxlib/releases/tag/v1.3.0.

Closing this issue: if you have problems, please open another.

closed 2023-08-09 by RobMeades

Hadatko, 2023-08-09:
Sounds great. But we didn't know you are preparing it so we did our own version. But who knows what will happen in future ;)

RobMeades, 2023-08-09:
@Hadatko: apologies for not mentioning it, kind of decided to do it at the last minute and we weren't sure if it would make the release or not; we should have put something about the possibility in the `UPCOMING.md` note.

RobMeades, 2023-08-09:
@Hadatko: maybe the more substantial change, don't know if you care, or if you did this yourselves, but 64 bit is now supported, so we can run on a Raspberry Pi 4 (which is what we test on).