u-blox/ubxlib #80: How can i download a file over cell tcp socket

alexmaron81, 2023-01-17:
I want to download a file for Over The Air Update (OTA) over TCP socket.
e.g. http://xxx.xxx.xxx.x:8100/file.md

Is there an example?

alexmaron81, 2023-01-17:
Is it like this?
const char socket_message[] = "GET /esp.bin HTTP/1.0\r\n"
uSockSendTo(sock, &address, socket_message, sizeof(socket_message));

RobMeades, 2023-01-18:
Hi Alex: we've just pushed HTTP support here, which uses the HTTP client in the module to do all the work.  You can see an [example of how to use it here](https://github.com/u-blox/ubxlib/blob/master/example/http_client/http_main.c#L224) and the full API is in [u_http_client.h](https://github.com/u-blox/ubxlib/blob/master/common/http_client/api/u_http_client.h).

Otherwise, if you want to do it directly yourself by forming the correct string having opened a TCP socket connection to the server, then yes you want something like that, see maybe this [Stack Overflow question](https://stackoverflow.com/questions/70550983/http-request-using-sockets-on-c) as a good example.

alexmaron81, 2023-01-19:
have now first switched to v4.4.
I'll deal with that later.

I have to take care of another problem now.
https://github.com/u-blox/ubxlib/issues/60

RobMeades, 2023-03-02:
Closing this one: please re-open if necessary.

closed 2023-03-02 by RobMeades