u-blox/ubxlib #81: load ubxlib library in esp idf

alexmaron81, 2023-01-19:
I can't get rid of the error message.
isp-idf v5.0

[image]

alexmaron81, 2023-01-19:
also for
[image]

RobMeades, 2023-01-19:
Hi Alex: we test with ESP-IDF version 4, I didn't even know there was an ESP-IDF version 5.

RobMeades, 2023-01-19:
Seems that version 5 ESP-IDF has a few breaking changes.  My first guess would be that the above might be caused by this one:

https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/migration-guides/release-5.x/build-system.html#specify-component-requirements-explicitly

You could try adding `driver` to your list of required components.

RobMeades, 2023-01-19:
..and `esp_timer` as well.

RobMeades, 2023-03-02:
Closing this one: please re-open if necessary.

closed 2023-03-02 by RobMeades