u-blox/ubxlib #82: uSockWrite does not support "connected" UDP sockets

johannesmay, 2023-02-02:
Hi again, we've run into another issue with ubxlib unfortunately.
We have some code that uses a UDP connection but relies on "connecting" UDP sockets, i.e. setting the default target address by calling `connect()` on a UDP socket once and just using plain `send()` and `receive()` calls.

This is supported by the modem (though discouraged) and also by the cellular layer of ubxlib.
But the general sock layer does not support this and filters by socket type on `uCellSockWrite()` and `uCellSockRead()` calls.

Is there a reason for this limitation?
If so, do you see a way to ease that constraint?

RobMeades, 2023-02-02:
Hi Johannes: it might be something to do with the way the short range stuff works, let me check...

RobMeades, 2023-02-02:
There's a [comment in the sockBasicUdp()](https://github.com/u-blox/ubxlib/blob/master/common/sock/test/u_sock_test.c#L1054) test that we used to test connection of a UDP socket but stopped doing so because SARA-R422 had a bug which prevented it.  However, even then, we didn't actually call `uSockWrite()` on that UDP socket in that test, so it wasn't a very good test.  The sockets code for Wifi [only supports socket writes for TCP](https://github.com/u-blox/ubxlib/blob/master/wifi/src/u_wifi_sock.c#L1133) but, on the other hand, it will just return an error so if the change is as simple as removing [this line in u_sock.c](https://github.com/u-blox/ubxlib/blob/master/common/sock/src/u_sock.c#L2170) I can't see a problem with that.

Would making that change meet your needs?

johannesmay, 2023-02-02:
Thanks for your quick response! Your suggested change is exactly what we did locally to get it to work 👍 It also seems to work on our R422, so not sure what bug occured with that in the past.

RobMeades, 2023-02-02:
Your requested change has been made internally, just waiting for tests to run and review from the short-range people,  hopefully push it to public tomorrow.

johannesmay, 2023-02-03:
Great! Thanks for your swift response as always 👍

RobMeades, 2023-02-03:
Hopefully address by 57f7ca5c17ccc15f737f1e9b0cb1ae8dcd2363aa - let me know if not.

closed 2023-03-02 by RobMeades