u-blox/ubxlib #86: Provisioning over Access point

AlexanderZender, 2023-02-14:
Hello,

does this library support an WiFi access point mode in which we can connect to the chip to provision it over a website?
(configure the local WiFi the chip should connect to, to connect to the internet)
We are currently using the WGM160 from silabs which provide such a functionality, but want to change chips.
On the ublox website I couldn't find much information how one provides a build-in webpage within the host MCU.
Maybe you have some more insight?

Best regards

plerup, 2023-02-16:
Hi Alexander,
We have planned for this but it will not make it for the coming 1.2 release.
I could however provide you with a prototype version as soon as I have something, if you want to help out testing it.

BR
/Peter

AlexanderZender, 2023-02-16:
Hi plerup,

thank you for the information! After a bit of digging deeper, I realized I might be in the wrong library.
Is this the library for the openCPU or u-connect version?

We would like to use the u-connect version of the NINA chip (w15) and connect an external host MCU.
When running the NINA chip in access point mode, connect an external device and provide a website from the NINA/MCU setup to the user.
If this is the correct version for this use case, will there be a webpage template/example on how to connect the requests by the website to the host MCU?

Best regards

plerup, 2023-02-16:
Well, ubxlib handles (to some extent) both opencpu and u-connect. The planned first implementation of WiFi access point in ubxlib is by using u-connect from a host cpu as you describe.

We currently have no web-server example but that is a good idea to include as part of this functionality.

BR
/Peter

AlexanderZender, 2023-02-16:
Alright, that is good news! 
I would be interested in any pre-release when you are ready or if you need a tester's feedback as we are currently in the preparation for a new project that will start in the next months in which we would use the chip in the setup described above.

Best regards
Alex

AlexanderZender, 2023-03-02:
@plerup Due to some planning changes, we are currently fast tracking our project. And would begin development next week.
I would like to take you up on your offer for a preview version, if that is available? 

Best regards

plerup, 2023-03-02:
@AlexanderZender 
Hi, I'm working on it, but nothing usable just yet. The main challenge is to implement a tcp server in ubxlib. That has never been done before. I assume that you need that in order to access the AP once it's started, i.e. i small webserver for provisioning of WiFi credentials.
Maybe some very alfa version could be available next week. I will keep you posted.
BR
/Peter

AlexanderZender, 2023-03-02:
@plerup 
Hey, 
yes, that is correct that is an important feature, but we also need to get familiar with the library first.
That sounds fantastic! Whenever you think you are ready we would be happy to give it a spine. And provide you with feedback.

As a side question, is this library only compatible with RTOS systems or could we also use it with bare metal chips?

Thank you

plerup, 2023-03-02:
No, an RTOS is required, either FreeRTOS or Zephyr

AlexanderZender, 2023-03-06:
Thank you for the info,
At the moment we are going bare metal so, do you have someone in the team that could be asked about AT commands?
I encountering an ERROR but I cant figure out whats the problem.

RobMeades, 2023-03-06:
Hi @AlexanderZender: some questions for clarification:

- You say "bare metal": `ubxlib` does need an RTOS, for tasks/queues/mutexes; I guess you mean that you have an RTOS environment of your own which you are now porting `ubxlib` to, following the [DIY instructions](https://github.com/u-blox/ubxlib/tree/master/port#diy).  Is that correct?
- Can you explain the sequence you are following that results in the error; e.g. an idea of the sequence of `uPort`/`uDevice`/`uNetwork` etc. API calls you are making, the kind of thing you see in our [examples](https://github.com/u-blox/ubxlib/tree/master/example), and maybe a `.txt` file log of the AT exchanges?

AlexanderZender, 2023-03-06:
Hey @RobMeades,

by bare metal I mean we are implementing the AT commands by hand, as there is no underlying OS.
But one AT command does not work as expected and we cant find the issue.
If you want I can explain it here

RobMeades, 2023-03-06:
It is possible that I'm misunderstanding here, please bare with me:

- @plerup is creating code inside `ubxlib` which will allow a user of `ubxlib` to tell a u-blox WiFi module to be a WiFi access point that can serve a simple web page to an attached WiFi client,
- this code will be "`ubxlib` code", and so will rely on RTOS features (tasks, queues and mutexes) like the rest of `ubxlib`,
- is it your plan to somehow make use of what Peter delivers, but without an RTOS?
- if not, can you indicate how Peter's delivery will be of use to you?

Apologies if this is all a bit deliberate/stupid, I just have a feeling there's a misunderstanding and would like to get to the bottom of it so that I can help.

AlexanderZender, 2023-03-06:
Thank you, sure. I will send you an email describing the problem tomorrow if that is alright.
Also can you share your email?

RobMeades, 2023-03-06:
rob.meades@u-blox.com

RobMeades, 2023-08-09:
This feature is now published, see https://github.com/u-blox/ubxlib/releases/tag/v1.3.0 and the example here: https://github.com/u-blox/ubxlib_examples_xplr_iot/tree/master/examples/captive_portal (though you will need to update the `ubxlib` submodule of that repo to point to the latest `ubxlib` to bring in the functionality).

Closing this issue: please re-open, or start a new one, if you have further issues.

closed 2023-08-09 by RobMeades