u-blox/ubxlib #89: Frequent TCP bad packets observed with shorter CHUNK_LENGTH during file uploads

Tagussan, 2023-03-21:
This issue is a continuation of [this PR](https://github.com/u-blox/ubxlib/pull/87). While using NINA-W132, we have observed that uploading a large file via a PUT request with longer `CHUNK_LENGTH` results in more bad TCP packets. I further elaborate on this topic in the ongoing discussion within the PR.

@RobMeades @mekm,
I apologize for the delay in my response. Below are the results of our packet capture using Wireshark, including statistics on error packets. We tested two different chunk lengths: 128 and 2556 (128x20). During our test, we ran our application which communicates with our server and sends 1MB of data via a PUT request. We captured the packets using a port-mirroring hub.

Please find attached a screenshot of the packet capture for `CHUNK_LENGTH=128`. The right-hand bar shows the entire communication landscape.

[image]

And here is a screenshot for `CHUNK_LENGTH=2556`.

[image]

As you can see from the right-hand bar, we have more packets marked 'black' for the case of  `CHUNK_LENGTH=2556`. The black packets represent bad TCP packets.

Please also find attached statistics for `CHUNK_LENGTH=128`. 
[image]

And for `CHUNK_LENGTH=2556`. 
[image]

We captured a total of 2507 packets for `CHUNK_LENGTH=128` and 2906 packets for `CHUNK_LENGTH=2556`. and 2906 packets for CHUNK_LENGTH=2556. The statistics show that the number of error packets is almost double when we use a longer `CHUNK_LENGTH`.

RobMeades, 2023-03-30:
Thanks for all this detail, very interesting: not sure if this is a factor or not but obviously 2507 packets of length 128 bytes is a lot less data than 2906 packets of check lunch 2556 (if I've understood you correctly).  Is it possible that if you ran the "128 byte" test for 20 times longer, i.e. transferring the same amount of data, that the number of errors for that case would rise, making the percentage loss similar?

Tagussan, 2023-04-02:
Each data packet contains about 1 kilobyte of data in both cases. If I understand correctly, the `CHUNK LENGTH` parameter does not correspond to the length of the packet transferred. It is only related to the frame length of UART between NINA-W132 and MCU. The actual length of the packet is determined by NINA-W132. So the total amount of data transferred has no difference between these two situations.

RobMeades, 2023-04-02:
Apologies if I'm being a bit slow with this: could you clarify what you mean by `CHUNK LENGTH`?  I'm having trouble relating the term `CHUNK LENGTH` to the parameters of the [uConnectExpress user manual](https://www.u-blox.com/sites/default/files/u-connectXpress-ATCommands-Manual_UBX-14044127.pdf).  Or do you, for instance, mean that you are running in EDM mode and the length of each EDM packet is `CHUNK LENGTH`?

closed 2023-04-05 by Tagussan