u-blox/ubxlib #90: Support NB2 RAT

chenlijun99, 2023-03-22:
Hi there,

we're experimenting with SARA-R422M8S, which supports LTE Cat NB2. However, I see that as for now `uCellNetRat_t` defines only `U_CELL_NET_RAT_NB1`.
Is there any plan to add support for NB2?

Thanks!

philwareublox, 2023-03-22:
> **** This is an EXTERNAL email. It was sent from outside of u-blox. ****

Hi,

Ubxlib has no control over the NB-IoT standard used by the cellular modules.
Using ubxlib and the uCellNetRat_t setting of RAT_NB1 does not mean it will downgrade NB2 modules to NB1

This is probably an oversight in our #defines.

Regards,
Phil.

chenlijun99, 2023-03-24:
Hi. Does this mean that if I use `U_CELL_NET_RAT_NB1`, then the module will actually use `LTE CAT NB2` and everything will work?

RobMeades, 2023-03-30:
That's correct: for instance the code here just uses "8" with the `AT+URAT` command for the NB case (this described in the AT command manual for SARA-R422, section 7.5, as "LTE Cat.NB1").

closed 2023-03-30 by chenlijun99