u-blox/ubxlib #92: Zephyr SPI Error

bbyers-UBCO, 2023-03-31:
Tried to implement the example found in "pos_main.c" recently on the v1.2.0 release using Zephyr and were getting, "Unable to open GNSS!" over SPI but with a different return code (-5). Not sure what is happening or how to go about debugging the UBXLIB side of this.

bbyers-UBCO, 2023-03-31:
Separating this from https://github.com/u-blox/ubxlib/issues/91

bbyers-UBCO, 2023-03-31:
```
    .deviceType = U_DEVICE_TYPE_GNSS,
    .deviceCfg = {
        .cfgGnss = {
            .moduleType = U_CFG_TEST_GNSS_MODULE_TYPE,
            .pinEnablePower = GNSS_EN_PIN,
            .pinDataReady = -1, // Not used
            // There is an additional field here:
            // "i2cAddress", which we do NOT set,
            // we allow the compiler to set it to 0
            // and all will be fine. You may set the
            // field to the I2C address of your GNSS
            // device if you have modified the I2C
            // address of your GNSS device to something
            // other than the default value of 0x42,
            // for example:
            // .i2cAddress = 0x43
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_SPI,
    .transportCfg = {
        .cfgSpi = {
            .spi = U_CFG_APP_GNSS_SPI,
            .pinMosi = U_CFG_APP_PIN_GNSS_SPI_MOSI,
            .pinMiso = U_CFG_APP_PIN_GNSS_SPI_MISO,
            .pinClk = U_CFG_APP_PIN_GNSS_SPI_CLK,
            .device = U_COMMON_SPI_CONTROLLER_DEVICE_DEFAULTS(U_CFG_APP_PIN_GNSS_SPI_SELECT)
        },
    },
};
```

bbyers-UBCO, 2023-03-31:
`U_CFG_APP_PIN_GNSS_SPI_SELECT` isn't set to -1. Sounds like we should try setting it to -1 to let Zephyr handle the CS?
`U_CFG_TEST_GNSS_MODULE_TYPE=1`
`GNSS_EN_PIN=5`
`U_CFG_APP_GNSS_SPI=3`

RobMeades, 2023-03-31:
If `U_CFG_APP_PIN_GNSS_SPI_SELECT` is at its [default for Zephyr](https://github.com/u-blox/ubxlib/blob/master/port/platform/zephyr/cfg/u_cfg_app_platform_specific.h#L379) then it would be -1 but if you have set it to something other than -1 that should be OK also; `CS` will be handled as a GPIO pin, so provided your device tree sets whatever pin you have chosen to be a valid Zephyr GPIO pin that should also work.

From what you say, the only print you get out of the example is "Unable to open GNSS!", you do not get the "U_GNSS: initialising with ENABLE_POWER pin..." print that `mtfurlan` got, so the invalid parameter is occurring quite early on, at the SPI configuration stage.  Can you post your `.dts` or `.overlay` file here so that can see how the SPI and other ports/pins are allocated?

RobMeades, 2023-08-14:
Hi @bbyers-UBCO: I'm going to close this issue due to inactivity, like some bot :-).  If you need more, please re-open it, or start another issue, etc.

closed 2023-08-14 by RobMeades