u-blox/ubxlib #95: SIM PIN entry is not supported

chenlijun99, 2023-04-05:
Hi,

apparently from here

https://github.com/u-blox/ubxlib/blob/80e76d49b97d8128202d5dd2a2fa9bfa2ecbc47a/cell/src/u_cell_pwr.c#L1778-L1779

SIM cards with PIN are not supported.

It this actually the case or did I miss something?

Thanks!

RobMeades, 2023-04-05:
Hi there: yes, that's correct - so far all of the applications we have come across are in IoT devices where there is no human around to enter a PIN.  That and power saving doesn't generally work when a PIN has to entered (since RAM is off when in power saving mode there is no way for the module to "remember" the SIM PIN, even if that wasn't considered a security risk).

We can, of course, add this as a feature request if required.

chenlijun99, 2023-04-05:
Thanks for the response! We don't need it either for now and for the foreseeable future. I just asked for futureproofness.
Feel free to close this issue.

RobMeades, 2023-04-05:
NP, this issue can serve as a clarification for others.