u-blox/ubxlib #96: Try to use GNSS for Sodaq Sara Sff board

Dorian-Mrll, 2023-04-25:
Hello everyone ! I'm blocking, here I'd like to use the ubxlib library to communicate with my internal gnss module on my sodaq sara sff card. My working environment is the ZorinRTOS framework on the plaftfomIO IDE.
Here is the link to the pins : https://support.sodaq.com/Boards/Sara_SFF/#pin-definitions
and here is the linkschematics of the board : 
I configured my dts, and for the program I retrieved: https://github.com/u-blox/ubxlib/blob/master/port/platform/platformio/example/position.c to adapt it for my case : 
[image]

Already I do not know what to put for .i2c of the structure .cfgI2C, in my case it is the sercom3, I would like to have more clarification on what it is not correct or what I must put in place. Because currently has share of the messages of the type "Failed to initiate the module: -2", I do not know much... Thanks in advance and have a nice day.

RobMeades, 2023-04-25:
Hi there, and thanks for posting.  Some questions for clarification first:

- you mention `ZorinRTOS` above: I assume you mean that you are _building_ on a Linux machine (Zorin OS, Ubuntu-like as I understand it), where you are using the PlatformIO IDE.  Is that correct?
- what RTOS are you using on the microcontroller of the Sodaq SARA SFF board?  We support Zephyr, ESP-IDF and Arduino[ESP-IDF], so I guess that you are using Zephyr?

Dorian-Mrll, 2023-04-25:
Thanks for your time, I am indeed using the PlatformIO IDE, with Zephyr.

RobMeades, 2023-04-26:
Great, thanks for confirming.  A few things then:

1. Error code -2 is [ U_ERROR_COMMON_NOT_INITIALISED](https://github.com/u-blox/ubxlib/blob/master/common/error/api/u_error_common.h#L57), so it seems likely that one or more of the initialisation functions are not being called: are you starting by coping one of our [examples](https://github.com/u-blox/ubxlib/blob/master/example/gnss/pos_main.c#L224) (if so, which one?), or can you maybe post a link to your code?
2. With the above error code you will not have got this far yet but, FYI, since you are using Zephyr, all of the pin allocations must be [done through the device tree](https://github.com/u-blox/ubxlib/tree/master/port/platform/zephyr#device-tree), that's just the way Zephyr works I'm afraid, so you must set `pinSda` and `pinScl` to -1.
3. As to the `i2c` item in the `cfgI2C` structure, that should be set to whichever HW block number inside the SAMD21 chip you want to use for I2C comms.  Looking at the "sercom" section of the [SAMD21 data sheet](https://onlinedocs.microchip.com/pr/GUID-22527069-B4D6-49B9-BACC-3AF1C52EB48C-en-US-18/index.html?GUID-EE060DEA-159B-4FA3-80E8-012D38AE61B3), the term "sercom" is used to refer to all forms of serial generically (IC, SPI and UART); you say that you are using `sercom3`: looking at the [default Zephyr device tree for SAMD21](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/atsamd21_xpro/atsamd21_xpro.dts), it has `sercom3` configured to be a UART, whereas `sercom2` is the one configured to be I2C, hence I would guess `i2c` should be 2 to match with the device tree.  If, of course, you have edited or overridden the device tree with a `.overlay` file so that `sercom3` has become the I2C port, then `i2c` should be 3.

RobMeades, 2023-04-26:
Note: since we don't use/test Atmel devices ourselves, it is possible that the Zephyr bindings (the strings Zephyr uses to identify peripherals in its device tree) are different to the [ones we assume](https://github.com/u-blox/ubxlib/blob/80e76d49b97d8128202d5dd2a2fa9bfa2ecbc47a/port/platform/zephyr/src/u_port_i2c.c#L135).  Zephyr can get very chip-specific in this respect I'm afraid.  We'll have to just hope that Zephyr does some intelligent mapping in the background to find an I2C device, rather than requiring, for instance, `i2c0` to be changed to `sercom2` or `sercom3`.

Dorian-Mrll, 2023-04-27:
I tried as you said to put the uPortI2cInit() function just after uPortInit(). But I always find the same error code: -2. Looking closer at the functions used, I suspect the function called by uDeviceOpen the uDeviceLock() function to return all the time: U_ERROR_COMMON_NOT_INITIALISED. In the [u_device_shared.c](https://github.com/u-blox/ubxlib/blob/master/common/device/src/u_device_shared.c#L62) file, the gMutex variable is initialized to NULL, so it's normal that in the function, we always return the same error since it is not modified in the meantime, so we never go into the [loop that contains the uPortMutexLock(gMutex) function](https://github.com/u-blox/ubxlib/blob/master/common/device/src/u_device_shared.c#L123). Am I right ? To integrate ubxlib I followed this [ReadMe](https://github.com/u-blox/ubxlib/blob/master/port/platform/platformio/README.md), I added the line and got exactly the same [prj.conf](https://github.com/u-blox/ubxlib/blob/master/port/platform/platformio/zephyr_default.conf) for my project. Should I use other files like text files or pyhton files available on the [github ](https://github.com/u-blox/ubxlib/tree/master/port/platform/platformio)?

RobMeades, 2023-04-27:
Aha!  You will probably have already spotted the the return code -6 is [U_ERROR_COMMON_NO_MEMORY](https://github.com/u-blox/ubxlib/blob/master/common/error/api/u_error_common.h#L61), so likely one of the functions called in [uPortInit()](https://github.com/u-blox/ubxlib/blob/master/port/platform/zephyr/src/u_port.c#L88) is unable to `malloc()` sufficient memory to do its work.

How much heap does Zephyr give on the SAMD21?

Dorian-Mrll, 2023-04-27:
I found this : 
[image]
a priori, a priori, it should : [256 KiB de mémoire flash et 32 KiB de RAM ](https://docs.zephyrproject.org/3.1.0/boards/arm/adafruit_feather_m0_basic_proto/doc/index.html)

Then for the blink program, I also have the same return code for the uPortInit() function. But this program works correctly on my board.

RobMeades, 2023-04-27:
Ah.  32 kbytes of RAM in total, for both stack and heap, is likely to be tight.  Blinky will work, 'cos it doesn't use any RAM to speak of, but for the rest you'll need to find out how much of those 32 kbytes of RAM Zephyr is allocating for stack and how much for heap, because it will make a significant difference.

There didn't use to be a way to determine the free heap size in Zephyr, and it is a bit confusing as I believe there can be more than one heap, but it might be that [sys_heap_print_info()](https://docs.zephyrproject.org/apidoc/latest/sys__heap_8h.html#af36db704bd892b657ccaa7a4cebc45e5) is what you want?

Once we know what Zephry is allocating to what, you can tune the stack and heap sizes to fit your particular requirements.

When you have found the right Zephyr heap stats function, you could insert a call to it at the start of [uPortInit()](https://github.com/u-blox/ubxlib/blob/master/port/platform/zephyr/src/u_port.c#L88) and between each of the functions called from [uPortInit()](https://github.com/u-blox/ubxlib/blob/master/port/platform/zephyr/src/u_port.c#L88): this will help show where the tweaks are needed.

Dorian-Mrll, 2023-05-02:
hello, I finally couldn't find the heap and memory size. I didn't find any indication, I tried the functions you indicated without result, but I can still try to adjust these memories with CONFIG_HEAP_MEM_POOL_SIZE if it's the only way ...

RobMeades, 2023-05-03:
OK, I think the only effective we for me to help you is for me to duplicate your HW setup and try it myself.  Can you throw some links into here as to what you are using and I'll see if I can obtain the same?

Dorian-Mrll, 2023-05-15:
Hello again thank you for helping me and for your time, but I managed to use my ublox module directly via Zephyr!
Have a nice day.

RobMeades, 2023-05-15:
Great, in that case I will close this issue. Do let me know if we can help with anything else.

closed 2023-05-15 by RobMeades