u-blox/ubxlib #97: python vs python3

mtfurlan, 2023-04-25:
I'm trying to follow the instructions at [port/platform/zephyr/runner](https://github.com/u-blox/ubxlib/blob/master/port/platform/zephyr/runner/README.md)
to build your examples, to try to prove my setup is wrong in #91 .

[`common/lib_common/makelib.mk` directly calls `python`](https://github.com/u-blox/ubxlib/blob/master/common/lib_common/makelib.mk#L85), not [`python3` as the comments suggest it will](https://github.com/u-blox/ubxlib/blob/master/common/lib_common/makelib.mk#L12), and on my updated debian stable, `python` points to `python2` for some reason.

I propose making the python scripts executable and just executing them so the shebang line gets to specify the interpreter(though maybe also do `#!/usr/bin/env python3` for environment reasons), or if that isn't acceptable for some reason, run them with python3 instead of python.

---

Steps to reproduce:
1. Setup https://github.com/zephyrproject-rtos/example-application according to it's readme
2. Add ubxlib to `west.yml` and use the last release of zephyr, not the main branch (run west update after this)
```diff
diff --git a/west.yml b/west.yml
index 580363c..582b2b9 100644
--- a/west.yml
+++ b/west.yml
@@ -8,9 +8,16 @@ manifest:
   remotes:
     - name: zephyrproject-rtos
       url-base: https://github.com/zephyrproject-rtos
+    - name: u-blox
+      url-base: git@github.com:u-blox
 
   projects:
     - name: zephyr
       remote: zephyrproject-rtos
-      revision: main
+      revision: v3.3.0
       import: true
+    - name: ubxlib
+      remote: u-blox
+      path: ubxlib
+      revision: v1.2.0
+      clone-depth: 1
```
3. go to ubxlib/port/platform/zephyr/runner and follow the README.md to run
```
west build -p auto -b nrf5340dk_nrf5340_cpuapp . --build-dir build_nrf5340dk_nrf5340_cpuapp
```
West will print 
```
...
-- Found Python3: /usr/bin/python3 (found suitable exact version "3.9.2") found components: Interpreter
...
Traceback (most recent call last):
  File "../../../..//common/lib_common/genencryptedbin.py", line 11, in <module>
    import secrets
ImportError: No module named secrets
make: *** [../../../..//common/lib_common/makelib.mk:95: /home/mark/tmp/zephyr-ublox-workspace/ubxlib/port/platform/zephyr/runner/build_nrf5340dk_nrf5340_cpuapp/libfibonacci/build/libfibonacci_code_final.bin] Error 1
```
`pip3 install secrets` has no effect.

RobMeades, 2023-04-26:
Thanks for raising this: `lib_common` was maybe an unwise/unspecific name - this is actually a utility which can load applications into RAM, potentially encrypted, and run them from there.  It was added for a specific use-case in our Wifi/BLE modules, only for the Zephyr platform I believe.  It has never actually been used anywhere: `ubxlib` does not, itself, use it at all, but the code is still built and tested by the test system, hence it forms part of the `runner` build (which is what we test).

So while this certainly needs resolving, it is, unfortunately, unlikely to be connected with #91.

mtfurlan, 2023-04-26:
Oh yeah sorry it's definitely not related to 91, that's just how I found it.
I probably just shouldn't have linked it.

closed 2023-05-03 by RobMeades