u-blox/ubxlib #100: ubxlib with drotek ZED-F9P ?

darkhmiza, 2023-05-03:
Hello I am trying to use ubxlib to get lat and lon from my DROTEK ZED-F9P with I2C connection, I am using nrf52840dk and I am trying to run the pos_main.c example, I've added the overlay file below :

`&i2c1 {
    compatible = "nordic,nrf-twim";
    status = "okay";
	pinctrl-0 = <&i2c0_default>;
	pinctrl-1 = <&i2c0_sleep>;
	pinctrl-names = "default", "sleep";
    mysensor: mysensor@42{
        compatible = "i2c-device";
        reg = < 0x42 >;
        label = "MYSENSOR";
    };
};

&spi1{
status = "disabled";
};`

Additionally I modified the config from 
`static const uDeviceCfg_t gDeviceCfg = {.deviceType = U_DEVICE_TYPE_NONE};`
to
`static const uDeviceCfg_t gDeviceCfg = {.deviceType = U_DEVICE_TYPE_GNSS, .transportType = U_DEVICE_TRANSPORT_TYPE_I2C};`

The build is fine and when I flash my board the udeviceopen() function returns error code -5.

I am using nrf connect SDK for VScode (ncs v2.0.0)

Any help would much appreciated

PS : Its my first time dealing a u-blox device or any GNSS module for that matter.

RobMeades, 2023-05-03:
Hi there, and thanks for posting.

Believe it or not, NCS 2.0.0 is slightly old now (yes, NCS and Zephyr having been moving forward at quite a rapid pace, up to NCS 2.3/Zephyr 3.3 now), but it _should_ work of course, and the error you are getting is not likely to be associated with NCS/Zephyr version.

The error code -5 is [U_ERROR_COMMON_INVALID_PARAMETER](https://github.com/u-blox/ubxlib/blob/master/common/error/api/u_error_common.h#L60).  My first suspicion is `gDeviceCfg`: the one you have modified is an "empty" one, which is fine, but if you look further up in the same file you will see that there are other things that should be populated.  For Zephyr, the pins all need to be -1 so that `ubxlib` knows that you're handlings those yourself in the device tree, and there is also the GNSS module type to populate.

For an F9P over I2C on I2C HW block 1 you probably need something more like this:

```
static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_GNSS,
    .deviceCfg = {
        .cfgGnss = {
            .moduleType = U_GNSS_MODULE_TYPE_M9,
            .pinEnablePower = -1,
            .pinDataReady = -1,
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_I2C,
    .transportCfg = {
        .cfgI2c = {
            .i2c = 1,
            .pinSda = -1,
            .pinScl = -1
        },
    },
};
```

Try that and see if things improve at all.

darkhmiza, 2023-05-04:
Hi, first of all thank you the quick response.
So I added the lines you suggested to my code and now I get the error below:

`*** Booting Zephyr OS build v3.0.99-ncs1  ***
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type I2C.
Opened device with return code -8.
Unable to open GNSS!
[00:00:00.478,790] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.489,196] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.499,572] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.509,948] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.520,324] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.530,700] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.541,076] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.551,452] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.561,828] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.572,204] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.582,611] <err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0
[00:00:00.587,707] <err> os: ***** BUS FAULT *****
[00:00:00.587,738] <err> os:   Precise data bus error
[00:00:00.587,738] <err> os:   BFAR Address: 0x2e002f4
[00:00:00.587,768] <err> os: r0/a1:  0x02e002f4  r1/a2:  0x200072c4  r2/a3:  0x00000001
[00:00:00.587,768] <err> os: r3/a4:  0x00000020 r12/ip:  0x00000000 r14/lr:  0x00018c93
[00:00:00.587,799] <err> os:  xpsr:  0x21000000
[00:00:00.587,799] <err> os: Faulting instruction address (r15/pc): 0x000159b4
[00:00:00.587,829] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.587,860] <err> os: Current thread: 0x20001f18 (main)
[00:00:02.193,237] <err> os: Halting system`

I also tried to do the same for the example in https://github.com/u-blox/ubxlib_examples_xplr_iot/tree/master/examples/position
 And in this case I get this  :

`*** Booting Zephyr OS build v3.0.99-ncs1  ***
[00:00:00.469,573] <err> os: ***** MPU FAULT *****
[00:00:00.469,635] <err> os:   Data Access Violation
[00:00:00.469,635] <err> os:   MMFAR Address: 0x29fc8
[00:00:00.469,665] <err> os: r0/a1:  0x00000002  r1/a2:  0x00029fc4  r2/a3:  0x00000000
[00:00:00.469,665] <err> os: r3/a4:  0x00029fc4 r12/ip:  0x00000004 r14/lr:  0x0000e887
[00:00:00.469,696] <err> os:  xpsr:  0x21000000
[00:00:00.469,696] <err> os: Faulting instruction address (r15/pc): 0x00003020
[00:00:00.469,726] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.469,757] <err> os: Current thread: 0x20001f30 (main)
[00:00:01.378,814] <err> os: Halting system`

How should I go about debugging this?

RobMeades, 2023-05-04:
Great, so ignore the second one for now: in our examples we assume that the device will open, just to avoid clutter, don't check the return code so, if the device doesn't open, the close function will access a flappy pointer, which is likely causing that.  Maybe we should check the return code, just for appearances sake.  Anyway.

From the prints, it looks as if the Nordic I2C stuff "downstairs" is unhappy, which is backed up by the error code, -8, which is [U_ERROR_COMMON_PLATFORM](https://github.com/u-blox/ubxlib/blob/master/common/error/api/u_error_common.h#L63).

Pasting "i2c_nrfx_twim: Error 0x0BAE0001" into Google led me here:

https://devzone.nordicsemi.com/f/nordic-q-a/75770/i2c_nrfx_twi-error-0x0bae0001-occurred-for-message-0

...where Nordic support say that their error code 0x0BAE0001 indicates a NACK on the I2C bus, so either the GNSS chip is not listening on that I2C bus or the I2C address is wrong somehow.  Assuming, in the [uDeviceCfgGnss_t](https://github.com/u-blox/ubxlib/blob/master/common/device/api/u_device.h#L247) part of the [uDeviceCfg_t](https://github.com/u-blox/ubxlib/blob/master/common/device/api/u_device.h#L318) structure you are passing into [uDeviceOpen()](https://github.com/u-blox/ubxlib/blob/master/common/device/api/u_device.h#L386), you have set it as above:

```
.deviceType = U_DEVICE_TYPE_GNSS,
    .deviceCfg = {
        .cfgGnss = {
            .moduleType = U_GNSS_MODULE_TYPE_M9,
            .pinEnablePower = -1,
            .pinDataReady = -1,
            // There is an additional field here:
            // "i2cAddress", which we do NOT set,
            // we allow the compiler to set it to 0
            // and all will be fine. You may set the
            // field to the I2C address of your GNSS
            // device if you have modified the I2C
            // address of your GNSS device to something
            // other than the default value of 0x42,
            // for example:
            // .i2cAddress = 0x43
        },
```

...so leaving out the `i2cAddress` field (which your compiler will initialise to 0), then the `ubxlib` code will assume that the GNSS chip has an I2C address of 0x42, which is the default as programmed in the factory.  Unless you have changed the address in the GNSS chip for some reason, this should not fail.

Which leaves us with either pins being wrong or electrical issues (e.g. presence of pull-ups, or [unlikely] power to the GNSS chip).  Do you have access to something like a Saleae probe, or similar, to monitor the I2C bus, so that we can see what's going on directly?

darkhmiza, 2023-05-04:
Unfortunately I don't have a logic analyzer at the moment but I have a scope at my disposal so I tried to see if I my board sends anything from my i2c1 port (SCL : P0.31, SDA: P030) it does not but on the other hand I see that my i2c0 port is active which tells me that code is using I2c0, so my overlay file is effectively doing nothing.

After using i2c0 pins and commenting the code in the overlay file I am getting this : 

```
*** Booting Zephyr OS build v3.0.99-ncs1  ***
Opened device with return code -8.
Unable to open GNSS!
Done.
```
As you'll notice I still get error code -8 so my board I2c's  "downstairs" is still unhappy.

As for the address I did not change it, I kept the default 0x42 and I am sure its the correct address as I was able to communicate to my module using zephyr i2c driver.

darkhmiza, 2023-05-04:
Ok problem fixed so apparently my overlay file is wrong, this is what I had at the beginning:

```
&i2c1 {
compatible = "nordic,nrf-twim";
status = "okay";
pinctrl-0 = <&i2c0_default>;
pinctrl-1 = <&i2c0_sleep>;
pinctrl-names = "default", "sleep";
mysensor: mysensor@42{
compatible = "i2c-device";
reg = < 0x42 >;
label = "MYSENSOR";
};
};

&spi1{
status = "disabled";
};

```
My pin control definitions are using i2c0_default/sleep instead of i2c1, such an obvious mistake and that's why you don't copy paste.

After changing them to i2c1 and running the program I am getting this :

```
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type I2C.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 04 04 00 00 00 09 00 17 76.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
Starting position stream.
U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 ff ff 21 00 b8 cc.
U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 05 00 21 10 00 03 00 21 20 01 01 00 21 30 e8 03 02 00 21 30 01 00 [body 26 byte(s)].
Opened device with return code 0.
U_GNSS: sent command b5 62 06 01 02 00 01 07 11 3a.
U_GNSS: decoded UBX response 0x06 0x01: 01 07 00 00 00 00 00 00 [body 8 byte(s)].
U_GNSS: sent command b5 62 06 01 08 00 01 07 01 00 00 00 00 00 18 e2.
U_GNSS: decoded UBX response 0x05 0x01: 06 01 [body 2 byte(s)].
Waiting up to 60 seconds for 5 position fixes.
U_GNSS: sent command b5 62 06 01 08 00 01 07 00 00 00 00 00 00 17 dc.
U_GNSS: decoded UBX response 0x05 0x01: 06 01 [body 2 byte(s)].
```
By the way my GNSS module is not connected to any antenna, I am simply testing the communication.

Anyway Thank you very much Rob for your help, you have saved my life.

RobMeades, 2023-05-04:
That's excellent!  You fixed it before I could even type my reply, well done :-)!

closed 2023-05-05 by RobMeades