u-blox/ubxlib #101: using zephyr uart driver with ubxlib

darkhmiza, 2023-05-09:
Hello,

So I was able te get data from an adxl345 connected to nrf52840dk using ubxlib, but when I tried to send this data through uart with the standard zephyr uart driver it does not work.

The problem arises when I try to set the uart callback :

```
    int err;
    err = uart_callback_set(uart, uart_cb, NULL);

		if (err) {
            printk("error %d",err);
			return err;
		}
``` 

I get error -88 which is an error that I've never seen before.

When I use the zephyr uart driver without ubxlib it works fine though.

Can anyone help me with this ?

RobMeades, 2023-05-09:
Hi there: a few questions for clarification:

1. When you say you were able to get data from an adxl345 connected to an nRF52840DK using `ubxlib`, what was `ubxlib`'s involvement in this?  Do you mean that you were running `ubxlib` on an nRF52840 and that you used the `ubxlib` `uPortSpi()` or `uPortI2c()` APIs to talk to the adxl345?
2. If the answer to (1) is "yes", were you using SPI or I2C?
3. In the case of the UART, you are calling `uart_callback_set()`, but obviously that is not a `ubxlib` API.  How is `ubxlib` involved in the UART part?

While you can, of course, use the `ubxlib` `uPortxxx()` APIs as you wish for non-`ubxlib` peripherals, you don't have to, you can just use the native platform functions; the `uPortxxx()` APIs are primarily intended so that we can move `ubxlib` from platform to platform, there is no requirement for you to use them in your application.

If, in your case, you have a GNSS module on the same I2C bus as other non-u-blox modules, then you can set up the I2C API yourself using the native platform I2C API and then, when you want to talk to the u-blox GNSS module, just set the [alreadyOpen](https://github.com/u-blox/ubxlib/blob/master/common/device/api/u_device.h#L170) flag in the [uDeviceCfgI2c_t](https://github.com/u-blox/ubxlib/blob/master/common/device/api/u_device.h#L152) structure and `ubxlib` will use the I2C interface that you have already opened.

Apologies if I've misunderstood what you're trying to do; please add any points of clarification that you think might help.

darkhmiza, 2023-05-09:
Hi Rob thank you for the quick response.

So I am sorry I didn't clarify the issue enough, ok so for the adxl345 I am using the example in https://github.com/u-blox/ubxlib_examples_xplr_iot/tree/master/examples/sensors which I assumed uses uPortxxx() functions, To be sure I removed the code that polls the adxl345 and left just the code for sending data through uart : 
```
void main()
{
   int err;
    err = uart_callback_set(uart, uart_cb, NULL);

		if (err) {
            printk("error %d\n",err);
			return err;
		}
    int j=0;
    char dist_str[30];
    double distance3=2.22;
    int32_t uPortUartOpen(1, 115200,rx_buffer,30,-1, -1,-1, -1);
    
    while(1){
          j=sprintf(dist_str,"    D%0.2lf\r\n;",distance3);
                        dist_str[0]=0x02;
	                    dist_str[1]=0x06;
	                    dist_str[2]=j-4;
	                    dist_str[3]=0x00;
	                    dist_str[j]=Get_CRC8(dist_str,j);
                        uart_tx(uart,dist_str, sizeof(dist_str), SYS_FOREVER_MS);
                        k_msleep(1000);

    }
}
```
And I don't use anything from ubxlib.
 
What is odd is that if remove the lines  : 

```
set(ENV{UBXLIB_DIR} ${CMAKE_CURRENT_LIST_DIR}/ubxlib)
include($ENV{UBXLIB_DIR}/zephyr/ubxlib.cmake)
```

From my CMakeLists.txt the uart works fine hence my suspicions that ubxlib affects the zephyr uart driver which are apparently wrong since you state that it is indeed possible to use native platform functions with ubxlib.

What do you think of this ?

RobMeades, 2023-05-09:
Ah, great, thanks for the clarification, that all makes sense.  We do indeed use the `ubxlib` `uPortxxx()` stuff in the XPLR IoT examples, in case we ever wanted to use the examples on another platform, so I understand your approach now.

If I'm reading the Zephyr documentation correctly, `uart_callback_set()` should return 0 or negative `errno` and, looking at their [errno.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/libc/minimal/include/errno.h), 88 is `ENOSYS` "Function not implemented".  From their [comments against the `uart_callback_set()`](https://docs.zephyrproject.org/2.7.0/reference/peripherals/uart.html#c.uart_callback_set) function:

> Since it is mandatory to set callback to use other asynchronous functions, it can be used to detect if the device supports asynchronous API. Remaining API does not have that detection.

...I'd guess that means Zephyr doesn't think the device pointer you have as `uart` supports an asynchronous callback.  If it is definitely a pointer to a UART (I believe the nRF52840 has two of them) it is probably worth double-checking that `uart` is definitely pointing to one of them from your device tree, and that it is `okay` (as Zephyr puts it).  Of course, I can't see how adding/removing the `ubxlib.cmake` would have any effect: that will just compile additional source files or include additional header files, but I guess you aren't bringing any of those `.h` files into your project anyway, or it wouldn't compile with the lines removed otherwise?

Just to be clear, with your `main()` as pasted in above, i.e. calling no `ubxlib` things`, then just adding:

```
set(ENV{UBXLIB_DIR} ${CMAKE_CURRENT_LIST_DIR}/ubxlib)
include($ENV{UBXLIB_DIR}/zephyr/ubxlib.cmake)
```

...to your `CMakeLists.txt` causes `uart_callback_set()` to return an error, whereas if you leave those two lines _out_ of `CMakeLists.txt`, `uart_callback_set()` returns zero...?

darkhmiza, 2023-05-10:
So I think I figured it out, when I add those line to my `CMakeLists.txt` I call in my build to the default.conf in ubxlib/port/platform/zephyr. When I checked the file I found `the line `CONFIG_UART_INTERRUPT_DRIVEN=y` which apparently does not work well with asynchronous uart `CONFIG_UART_ASYNC_API` after adding the line  `CONFIG_UART_INTERRUPT_DRIVEN=n`  to my `prj.conf` it works fine now.

Anyway Rob thank you for your help.

RobMeades, 2023-05-10:
Well found!  It is a bit weird, though, that you need to use a non-interrupt-driven UART for the Zephyr _asynchronous_ UART API to work; I'd have assumed almost the opposite would be the case, having a non-interrupt driven UART is kinda unusual, and I'd want my asynchronous API to be interrupt-triggered.  We only use that switch because, when we use Zephyr-on-Posix (i.e. Linux), there are no interrupts, so you have to poll the UART interface on a timer; inefficient but it's Linux so of no matter.

Very odd.  Anyway, you have got to the bottom of it.