u-blox/ubxlib #104: HTTP errors out too often

chenlijun99, 2023-05-15:
Hi,

I'm having problems with HTTP both in an old version of ubxlib (80e76d49b97d8128202d5dd2a2fa9bfa2ecbc47a) that I'm using and in the current `master` (a97d994f59f391200b180922e7e071ad0439c045).

Namely, too often making HTTP requests results in -1. Too often meaning that I get more errors than successes.

[http_example_error_log.txt](https://github.com/u-blox/ubxlib/files/11474744/http_example_error_log.txt)
[http_example_success_log.txt](https://github.com/u-blox/ubxlib/files/11474745/http_example_success_log.txt)

In addition to `+UHTTPER: 0,3,10`, I have also observed errors like `+UHTTPER: 0,10,22` and `+UHTTPER: 0,3,11`.

I tried to change the example to see whether retrying could help, but apprently once the HTTP client is in a state where it returns -1, it will keep returning -1.

[retry_patch.txt](https://github.com/u-blox/ubxlib/files/11475125/retry_patch.txt)

Tried to move the `while (1)` loop up, so each time it fails the HTTP profile is closed and then re-opened. Still same problem.

[retry_http_reopen_patch.txt](https://github.com/u-blox/ubxlib/files/11475124/retry_http_reopen_patch.txt)

Tried to move the `while (1)` loop up, so each time it fails the network is put down and turned up again (i.e. cellular connection is re-established). Does not help. Still keep getting -1 error. But then I tried to put a delay of 2 seconds after the network connection is established and it seems to mitigate the problem (stil get -1 sometimes, but it's much likely to succeed).

[retry_network_reconnect_patch.txt](https://github.com/u-blox/ubxlib/files/11475198/retry_network_reconnect_patch.txt)
[retry_network_reconnect_with_delay_patch.txt](https://github.com/u-blox/ubxlib/files/11475200/retry_network_reconnect_with_delay_patch.txt)

Is there an explanation for this?

chenlijun99, 2023-05-15:
After further experimenting, the gist seems to be:

* HTTP profile is orthogonal to cellular network connection, in the sense that it is OK to create the HTTP profile even before establishing the cellular network connection.
* BUT, after establishing the cellular network connection, put a slight delay before making the first HTTP request.
* If HTTP request returns -1, disconnecting and re-connecting to the cellular network is enough. No need to close and re-open the HTTP profile.

RobMeades, 2023-05-15:
Hmph.  I will have to call in the cavalry on this one I'm afraid: there is obviously something going on inside the SARA-R422 module, something not directly visible to us, which is getting in your way.  We do test SARA-R422 in the test system here (two modules, both with FW version 00.12,A00.00) and, though the test system as a whole does sometimes see HTTP failures ('tis the internet, that's just life), I don't recall any particularly landing in the lap of SARA-R422.

The cavalry in this case are in North America so will hopefully get back to you tomorrow.

RobMeades, 2023-05-15:
One thing I will likely be asked is what the signal strength/quality is like at your location.  Could you possible add something like the following, just before you do your HTTP request, and post what is printed-out back here?

```
uCellInfoRefreshRadioParameters(devHandle);
print("### RSRP %d dBm, RSRQ dB %d, SNR %d dB (EARFCN %d).\n", uCellInfoGetRsrpDbm(devHandle), uCellInfoGetRsrqDb(devHandle), uCellInfoGetSnrDb(devHandle), uCellInfoGetEarfcn(devHandle));
```

chenlijun99, 2023-05-15:
Sure. Will do now. Thanks for your help!

RobMeades, 2023-05-15:
Can't draw _too_ much conclusion as to the environment.

One thing to say is that you are on 2G, rather than LTE, so the bandwidth is going to be relatively small.  The HTTP connection seems to be an HTTPS one, so there will be certificate exchanges going on, a relatively large amount of data to transmit.

The signal strength in your logs ranges from -73 dBm down to -100 dBm (`ubxlib` reports the RSSI value from the `AT+UCGED` AT command since that is valid for all RATs, but it is also visible in the `AT+CSQ` command for 2G, and that shows up in your first log as -100 (113 - 13) at one point).  So it is washing around rather a lot; just the nature of RSSI measurement I'm afraid. 
 But whenever `AT+CSQ` reports a value, the value it reports is 0, which is good, and it does this in both the success and fail cases.

Cavalry time.

chenlijun99, 2023-05-15:
The same problem has been observed with plain HTTP (without TLS), so maybe we can also exclude that. Looking forward to reply from the cavalry!

RobMeades, 2023-05-17:
Right, the cavalry says that to find out what's going on for the "internal client error" (`+UHTTPER: 0, 3,10`) case, which seems to form the main source of your issues, requires a diagnostic log.  Unfortunately, for the SARA-R4 module series, the tools to take such a log need an NDA, as well as access to the USB port of the SARA-R4.  Is that something you are able to deal with?  If so I can send you the NDA (contact me at rob.meades@u-blox.com).

If you don't want to go that far, they also say that `+UHTTPER: 0,10,22` is only emitted if the PDP context is not active, so you could maybe call `uCellNetGetIpAddressStr()` before doing your HTTP thang, as that will return an error if there is no PDP context at the time, though of course there may always be a race condition.  There is also an AT command which `ubxlib` does not currently support, `AT+CIND`, which enables URCs on certain conditions (see section 5.4 of the SARA-R4 AT manual): you could maybe DIY-add that with your own callback that watches the "gprs" PS indication status, which will tell you the "live" state of the PDP context.

And, finally, `+UHTTPER: 0, 3,11` can occur when there is packet loss or a server not responding or HTTP profile settings are incorrectly configured (though I see no sign of this latter).

But of course, if you are able to go down the diagnostic-logging route, it will help get true visibility for all of the above.

chenlijun99, 2023-05-21:
Hi Rob. Sorry for the delay in the reply.

> Right, the cavalry says that to find out what's going on for the "internal client error" (+UHTTPER: 0,3,10) case, which seems to form the main source of your issues, requires a diagnostic log. Unfortunately, for the SARA-R4 module series, the tools to take such a log need an NDA, as well as access to the USB port of the SARA-R4. Is that something you are able to deal with? If so I can send you the NDA (contact me at [rob.meades@u-blox.com](mailto:rob.meades@u-blox.com)).

My company should already have an NDA with ublox, I'm just trying to get it from my colleagues and that is not entirely under my control. Anyway, we don't have access to the USB port on our custom boards and the evaluation kit is also being shipped to me, so yeah, that will take a while. ASAP I get the NDA, I'll send it to you.

> If you don't want to go that far, they also say that +UHTTPER: 0,10,22 is only emitted if the PDP context is not active, so you could maybe call uCellNetGetIpAddressStr() before doing your HTTP [...]

> And, finally, +UHTTPER: 0,3,11 can occur when there is packet loss or a server not responding or HTTP profile settings are incorrectly configured (though I see no sign of this latter).

In the meanwhile, I just had time to run some tests with trying to get the IP address. Here is the http example code I used

[with_get_ip_address.txt](https://github.com/u-blox/ubxlib/files/11525786/with_get_ip_address.txt)

Basically, I try to get the IP address before any HTTP request and after any HTTP request error.

Here are two logs

[error_0_3_11_with_no_ip_address.txt](https://github.com/u-blox/ubxlib/files/11525789/error_0_3_11_with_no_ip_address.txt)
I cannot get the IP address but the error is 0,3,11.

[error_0_10_22_with_ip_address.txt](https://github.com/u-blox/ubxlib/files/11525790/error_0_10_22_with_ip_address.txt)
I can get the IP address but the error is 0,10,22.

I find the result quite strange, since the error codes seem to be the opposite of what you explained.

RobMeades, 2023-08-14:
Hi @chenlijun99: I guess you probably got what you needed on this issue, and if not we have a direct channel via e-mail in any case, so I'm going to close this.  Please feel free to re-open it if you disagree.

closed 2023-08-14 by RobMeades