u-blox/ubxlib #107: CellTime support

eeFLis, 2023-05-31:
Hi there 

Are there any plans for CellTime implementation?

RobMeades, 2023-05-31:
Hi there: we have [uCellInfoGetTimeUtc()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_info.h#L281) and [uCellInfoGetTimeUtcStr()](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_info.h#L291), both of which use `AT+CCLK?` under the hood: is that what you mean, or did you need something different?

eeFLis, 2023-05-31:
We have adapted the ubxlib code so that we can use the feature. But I would be happy if it could be integrated "properly" into the library. Since we have an intermediate solution it is not urgent. What do you think?

RobMeades, 2023-05-31:
Yes, we should do it if you need it.  At the moment all efforts are on release 1.3 in July, so would probably have to be after that, but we would have you review the API for it (or you could sketch-out the one that you prefer if you like, for us to start from?).

eeFLis, 2023-05-31:
great, I will do that. Thanks for your help

eeFLis, 2023-06-01:
Hi Rob
Here's a first draft. let me know what you think of it when you've had time to look at it.

`#ifndef _CELL_API_U_CELL_TIME_H_
#define _CELL_API_U_CELL_TIME_H_

typedef struct
{
	uint32_t cellId; 	/** physical cell id of the locked base station */
	uint32_t status; 	/** Status of the operation in progress */
	uint32_t timeInfo; 	/** Source used for timekeeping */
	uint32_t absTime; 	/** Whether time is in UTC or on an arbitrary time scale */
	uint32_t result; 	/** Result of intermediate operation steps */
	int32_t offsetNs; 	/** When synchronizing with UTC time (by means of the GNSS) (<result>=1) the local
							time scale experiences a discontinuity. The knowledge of this discontinuity is needed
							to compare in time two events occurred before and after the re-synchronization. The
							parameter is issued only in the +UUTIMEIND URC */
	int32_t offsetS;

}cellTimeInformation_t;

typedef struct
{
    int32_t mcc; 	/** Mobile Country Code */
    int32_t mnc; 	/** Mobile Network Code */
    int32_t tac; 	/** Tracking area code */
    int32_t ci; 	/** Cell identity */
    int32_t dlf; 	/** Downlink E-UTRAN absolute radio frequency channel number */
    int32_t ulf; 	/** Uplink E-UTRAN absolute radio frequency channel number */
    int32_t pci; 	/** PhysCellID */
    int32_t rsrp; 	/** Current Reference Signal Received Power */
    int32_t rsrq; 	/** Current Reference Signal Received Quality */

} uCellTimeNetLteCell_t;

/**
 * @brief Source of the timing info
 *
 */
typedef enum {
    U_CELLTIME_SOURCE_GNSS_LTE = 0x01,	/** GNSS/LTE (best effort) */
	U_CELLTIME_SOURCE_LTE_ONLY = 0x02,  /** LTE only */
} uCellTimeSource_t;

/**
 * @brief enable continuous time pulse output on pin 19 (only pin 19 can be configured for "Time pulse output functionality on SARA R5 module").
 * 		  Time pulse period is 1 second with a pulse width of 100 milliseconds.
 *
 * @param cellHandle				the handle of the cellular instance.
 * @param offsetInSeconds         	offset compared to local time scale [Seconds].
 * @param offsetInNanoseconds		offset compared to local time scale [Nanoseconds].
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimeContinuousPulseEnable( uDeviceHandle_t cellHandle, uCellTimeSource_t source,  int32_t offsetInSeconds, int32_t offsetInNanoseconds );

/**
 * @brief disable time pulse output. Note: this do not change the output pin configuration.
 *
 * @param cellHandle 	the handle of the cellular instance.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimePulseDisable( uDeviceHandle_t cellHandle );

/**
 * @brief get the current cell time information
 *
 * @param cellHandle					the handle of the cellular instance.
 * @param[out] pCellTimeInformation		pointer to a place to store cellTimeInformation.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimeInformationGet( uDeviceHandle_t cellHandle, cellTimeInformation_t *pCellTimeInformation );

/**
 * @brief sets a callback which will be called when timekeeping status has changed reported by "+UTIMEIND" URC.
 * Such a callback may call uCellTimeInformationGet() to get the cell time information.
 *
 * @param cellHandle			the handle of the cellular instance.
 * @param pCallback				pointer to the function to handle any timekeeping status changes.
 * @param pCallbackParameter 	a pointer to be passed to the call-back. may be NULL.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimeStatusCallback(uDeviceHandle_t cellHandle, void (*pCallback) (void *pCallbackParameter) );

/**
 * @brief do an extended network search "+COPS=5". The detected cells can be used to sync on a specific cell.
 *
 * @param cellHandle					the handle of the cellular instance.
 * @param pArrayDetectedCells [out]		pointer to an array of detected Cells. Memory for the array is allocated by the function and must be freed by the user application.
 * @param detectedCellsCount [out]		count of the detected cells.
 *
 * @return
 */
int32_t uCellTimeNetDeepScan(uDeviceHandle_t cellHandle, uCellTimeNetLteCell_t **pArrayDetectedCells, uint32_t detectedCellsCount);

/**
 * @brief Forces the module to synchronize and lock on a specific LTE-M/NB-IoT cell of any specific MNO.
 *
 * @param cellHandle	the handle of the cellular instance.
 * @param pCell			pointer to a cell we want to synchronize with.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimesyncOnASpecificCellEnable( uDeviceHandle_t cellHandle, uCellTimeNetLteCell_t *pCell );

/**
 * @brief disable synchronization on a specific cell.
 *
 * @param cellHandle	the handle of the cellular instance.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimesyncOnASpecificCellDisable( uDeviceHandle_t cellHandle );

#endif /* _CELL_API_U_CELL_TIME_H_ */
`

eeFLis, 2023-06-01:
Hi Rob
Here's a first draft. let me know what you think of it when you've had time to look at it.

```
#ifndef _CELL_API_U_CELL_TIME_H_
#define _CELL_API_U_CELL_TIME_H_

typedef struct
{
	uint32_t cellId; 	/** physical cell id of the locked base station */
	uint32_t status; 	/** Status of the operation in progress */
	uint32_t timeInfo; 	/** Source used for timekeeping */
	uint32_t absTime; 	/** Whether time is in UTC or on an arbitrary time scale */
	uint32_t result; 	/** Result of intermediate operation steps */
	int32_t offsetNs; 	/** When synchronizing with UTC time (by means of the GNSS) (<result>=1) the local
					time scale experiences a discontinuity. The knowledge of this discontinuity is needed
					to compare in time two events occurred before and after the re-synchronization. The
					parameter is issued only in the +UUTIMEIND URC */
	int32_t offsetS;

}cellTimeInformation_t;

typedef struct
{
    int32_t mcc; 	/** Mobile Country Code */
    int32_t mnc; 	/** Mobile Network Code */
    int32_t tac; 	/** Tracking area code */
    int32_t ci; 	/** Cell identity */
    int32_t dlf; 	/** Downlink E-UTRAN absolute radio frequency channel number */
    int32_t ulf; 	/** Uplink E-UTRAN absolute radio frequency channel number */
    int32_t pci; 	/** PhysCellID */
    int32_t rsrp; 	/** Current Reference Signal Received Power */
    int32_t rsrq; 	/** Current Reference Signal Received Quality */

} uCellTimeNetLteCell_t;

/**
 * @brief Source of the timing info
 *
 */
typedef enum {
    U_CELLTIME_SOURCE_GNSS_LTE = 0x01,	/** GNSS/LTE (best effort) */
	U_CELLTIME_SOURCE_LTE_ONLY = 0x02,  /** LTE only */
} uCellTimeSource_t;

/**
 * @brief enable continuous time pulse output on pin 19 (only pin 19 can be configured for "Time pulse output functionality on SARA R5 module").
 * 		  Time pulse period is 1 second with a pulse width of 100 milliseconds.
 *
 * @param cellHandle				the handle of the cellular instance.
 * @param offsetInSeconds         	offset compared to local time scale [Seconds].
 * @param offsetInNanoseconds		offset compared to local time scale [Nanoseconds].
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimeContinuousPulseEnable( uDeviceHandle_t cellHandle, uCellTimeSource_t source,  int32_t offsetInSeconds, int32_t offsetInNanoseconds );

/**
 * @brief disable time pulse output. Note: this do not change the output pin configuration.
 *
 * @param cellHandle 	the handle of the cellular instance.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimePulseDisable( uDeviceHandle_t cellHandle );

/**
 * @brief get the current cell time information
 *
 * @param cellHandle					the handle of the cellular instance.
 * @param[out] pCellTimeInformation		pointer to a place to store cellTimeInformation.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimeInformationGet( uDeviceHandle_t cellHandle, cellTimeInformation_t *pCellTimeInformation );

/**
 * @brief sets a callback which will be called when timekeeping status has changed reported by "+UTIMEIND" URC.
 * Such a callback may call uCellTimeInformationGet() to get the cell time information.
 *
 * @param cellHandle			the handle of the cellular instance.
 * @param pCallback				pointer to the function to handle any timekeeping status changes.
 * @param pCallbackParameter 	a pointer to be passed to the call-back. may be NULL.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimeStatusCallback(uDeviceHandle_t cellHandle, void (*pCallback) (void *pCallbackParameter) );

/**
 * @brief do an extended network search "+COPS=5". The detected cells can be used to sync on a specific cell.
 *
 * @param cellHandle					the handle of the cellular instance.
 * @param pArrayDetectedCells [out]		pointer to an array of detected Cells. Memory for the array is allocated by the function and must be freed by the user application.
 * @param detectedCellsCount [out]		count of the detected cells.
 *
 * @return
 */
int32_t uCellTimeNetDeepScan(uDeviceHandle_t cellHandle, uCellTimeNetLteCell_t **pArrayDetectedCells, uint32_t detectedCellsCount);

/**
 * @brief Forces the module to synchronize and lock on a specific LTE-M/NB-IoT cell of any specific MNO.
 *
 * @param cellHandle	the handle of the cellular instance.
 * @param pCell			pointer to a cell we want to synchronize with.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimesyncOnASpecificCellEnable( uDeviceHandle_t cellHandle, uCellTimeNetLteCell_t *pCell );

/**
 * @brief disable synchronization on a specific cell.
 *
 * @param cellHandle	the handle of the cellular instance.
 *
 * @return zero on success else negative error code.
 */
int32_t uCellTimesyncOnASpecificCellDisable( uDeviceHandle_t cellHandle );

#endif /* _CELL_API_U_CELL_TIME_H_ */
```

RobMeades, 2023-06-01:
Without looking in too much detail at the AT command itself, likely you understand that better than me, this looks fine.  I'll post back here when I have an implementation that you can review.

RobMeades, 2023-08-06:
Hi there: I've pushed an implementation of CellTime to a preview branch here: https://github.com/u-blox/ubxlib/tree/preview_feature_cell_time_rmea, see [u_cell_time.h](https://github.com/u-blox/ubxlib/blob/preview_feature_cell_time_rmea/cell/api/u_cell_time.h) for the API.

Main differences to your original proposed API:

- supports the other modes of CellTime (which I know you probably aren't interested in),
- the [uCellTimeEnable()](https://github.com/u-blox/ubxlib/blob/794a54c59fc754eb1f0a13fc87fad8f9faac0aa1/cell/api/u_cell_time.h#L323C9-L323C20) function includes setting the callback (which can be `NULL`), no separate function for that,
- the callback is given a pointer to a structure which has all the things in it already, no need for the callback to call anything else,
- the scan function, [uCellTimeDeepScanAlloc()](https://github.com/u-blox/ubxlib/blob/794a54c59fc754eb1f0a13fc87fad8f9faac0aa1/cell/api/u_cell_time.h#L402C9-L402C31) returns a linked-list rather than an array: this was more memory efficient in terms of implementation since the code doesn't know how big the answer is going to be at the outset; [uCellTimeDeepScanFree()](https://github.com/u-blox/ubxlib/blob/794a54c59fc754eb1f0a13fc87fad8f9faac0aa1/cell/api/u_cell_time.h#L410) is provided for you to free the list.
- all time values are in nanoseconds: seemed easier to stick to one unit throughout, saves having to remember which timescale you're handling,
- most of the `uint32_t`s have become enums, just to make their meanings clearer.

It has all been tested with the exception that I can't test scanning and finding more than one cell since our test system is only attached to a single cell.  I'll rig something up to try that out manually on Monday.

Anyway, take a look when you can and give me your feedback.  FYI, this branch is based on our release 1.3 code, which should get pushed to `master` here on Mondayish I hope.

eeFLis, 2023-08-07:
Hi Rob

Great. Thanks for your work.
I will test it as soon as possible and give you feedback.

RobMeades, 2023-08-07:
Having just tried a deep scan on the live network, manually, I see that each result line "dribbles" out over quite a lot of seconds, eventually returning OK when done, which the code in the `preview` branch probably doesn't cope with well, since it is expecting the results to all come out at once followed by an OK.

Given this, I wonder if it might be worth changing the scan interface, so that it has a call-back for each result, something like:

```
int32_t uCellTimeDeepScan(uDeviceHandle_t cellHandle,
                          (bool) (*pCallback) (uDeviceHandle_t,
                                               uCellTimeCellInfo_t *,
                                               void *),
                          void *pCallbackParameter);
```

...where if `pCallback` returns `true` then the scan continues, else it may return `false` to abort the scan at the point, if it has a useful cell.

What do you think?

eeFLis, 2023-08-07:
Looks good. Perhaps it is sufficient to add `isLast `to the `pCallback` parameter.
If I have seen it correctly the guard timeout is 240 seconds for the deepScan. This would mean that the process can not be canceled before,  if no result was returned. right?

RobMeades, 2023-08-07:
OK, I have pushed a new preview branch (left the old one as it is) to [preview_2_feature_cell_time_rmea](https://github.com/u-blox/ubxlib/tree/preview_2_feature_cell_time_rmea).

In this one `uCellTimeDeepScanAlloc()` becomes [uCellTimeDeepScan()](https://github.com/u-blox/ubxlib/blob/85854f77322235de98c75171deb56eff43ac29bb/cell/api/u_cell_time.h#L413), which takes a callback that will be called either when a cell is found, with the cell information, or periodically (e.g. every second) with `NULL`.  If the function is happy with the cell, or if you just want to give up, it should return `false` to stop the scan, else it should return `true` to continue the scan.

This way you get to find a useful cell with maximum speed and you can still cancel at any time.

I couldn't think of a way of doing the `isLast` thing 'cos at the time the code calls the callback it doesn't know that this is the last one; so I didn't add that in the end.

I've removed my debug prints and this passes all my testing; if you can confirm that you are happy with it I will submit it for internal review and it should end up here tomorrowish.  FYI, the 1.3 release code that it is based on has already been pushed to `master` here.

RobMeades, 2023-08-08:
FYI, after reviewing internally, we will move `uCellTimeDeepScan()` to be `uCellNetDeepScan()` in the `uCellNet` API and remove the `timingAdvance` parameter from there (since it isn't part of the scan) and have it as a separate parameter to `uCellTimeSyncCellEnable()`.  Doesn't change the functionality but makes more logical sense API-wise.  I won't bother pushing that change to the preview here, unless you want me to.

RobMeades, 2023-08-08:
@eeFLis: now ready to be pushed to `master` here if you agree.

eeFLis, 2023-08-08:
Hi Rob
Thank you. It's fine for me. I couldn't test it on the hardware yet because we are still waiting for the new prototypes. These should arrive in the next few days. Since it passed your tests, I am sure it will work. I will certainly give you a feedback.

closed 2023-08-08 by RobMeades

RobMeades, 2023-08-14:
FYI, I realised while looking at cell ID for #124 that I'm decoding the cell ID returned by the `AT+COPS=5` scan as a decimal encoded in a string, whereas I should be decoding it as hex encoded in a string.  A fix for that will be along to `master` here shortly.

RobMeades, 2023-08-14:
See https://github.com/u-blox/ubxlib/commit/429a482781a86457b739f5cab7e370078c459578.

eeFLis, 2023-08-18:
Hi

I am currently testing `uCellNetDeepScan`, which works great, but I very often get the error message `+CME ERROR: Temporary Failure`. I read in your comments that this can happen when the module is busy doing its own search.

But this also happens if we let `uCellNetDeepScan` run in a loop for a long time, where I assume the module has completed its own search.

Do you have any idea what might be the cause?

eeFLis, 2023-08-18:
OK thanks. It does not block us we can ingnore the error for now. Didn't you have the problem with `+CME ERROR: Temporary` in your tests case?

RobMeades, 2023-08-18:
Indeed, we had to [allow retries](https://github.com/u-blox/ubxlib/blob/2d66485aca773f1a80872d4e3add0a1d81f73ead/cell/test/u_cell_time_test.c#L105); the tests are intended to test the `ubxlib` code rather than the module code, which I can do nothing about, so sometimes necessary.

RobMeades, 2023-08-27:
FYI, I've checked through our recent test logs of the [cellTimeBasic()](https://github.com/u-blox/ubxlib/blob/73da678e6293f0eb569f336da7208ea2fe4d146c/cell/test/u_cell_time_test.c#L333) test running on the test system and I can no longer find any cases of `+CME ERROR: Temporary Failure` being returned.

I've talked to the test team here and they say that they have seen this error being returned when the module happens to be in connected mode (when obviously it cannot do a search).

I _think_ I probably saw the `+CME ERROR: Temporary Failure` response when I was testing at my desk on the live network, rather than on the test system (which uses a Nutaq box to provide its own local cellular network); I'll see if I can confirm that the error response does happen when using the live network when I have a moment.